realsenseai/librealsense #9657: realsense using under Windows reports errors

Wenjiahao666, 2021-08-20:
| | |
|---|---|
|**librealsense**|2.48.0 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
Unhandled exception escaping from a worker thread!
Error type: Access Violation!
```

MartyG-RealSense, 2021-08-20:
Hi @Wenjiahao666  May I first ask whether you are using Windows 7 or Windows 10.

Yarong-zhang, 2021-08-24:
Hello, I had the same problem. I am using Windows10

MartyG-RealSense, 2021-08-24:
Thank you @Yarong-zhang - I have replied to the case that you created.

Wenjiahao666, 2021-08-24:
Thanks for your attention. I have solved this problem incidently by just updating RS’s firmware to the newest version,,then it works.
从 Windows 版邮件发送

发件人: MartyG-RealSense
发送时间: 2021年8月20日 14:38
收件人: IntelRealSense/librealsense
抄送: Wenjiahao666; Mention
主题: Re: [IntelRealSense/librealsense] realsense using under Windowsreports errors (#9657)

Hi @Wenjiahao666 May I first ask whether you are using Windows 7 or Windows 10.

MartyG-RealSense, 2021-08-24:
That is excellent news, @Wenjiahao666  - as you have achieved a solution, I will close this case.  Thanks again!

closed 2021-08-24 by MartyG-RealSense