realsenseai/librealsense #9661: Precise timing capture in multi-cam setting

sritee, 2021-08-20:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras). Yes
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)? Yes
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)? Yes
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues? Yesutf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue. - related issue found: https://github.com/IntelRealSense/librealsense/issues/2188

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 }| 
| Firmware Version                   | 01.05.05.00 | 
| Operating System & Version | Ubuntu 18.04 | 
| Kernel Version (Linux Only)    | 5.4.0-80-generic| 
| Platform                                 | PC |
| SDK Version                            || 
| Language                            |  Python| 
| Segment			| Robot| 

### Issue Description
Getting frames which have been captured after a time T, L515 multi-cam setup.

Hi, we are working on a multi-camera setting with the L515, where we have inter-cam sync on. 

The problem of interest is as follows: Capture a frame from each camera set which occur after a specified time T. 
Setup is as follows: We have syncing script running in the background (C++), which turns each of the cameras on-and-off at specified durations (to avoid interference). Similar setup to that as in https://dev.intelrealsense.com/docs/lidar-camera-l515-multi-camera-setup. Looping through N cameras in our case ends up at a 250ms per loop, and successfully avoids interference. 

Since each camera now is triggered at different times (using digital writes), we are interested in getting the precise hardware timestamps. We also want frames which occur at time greater than query time T (this T is gotten using a time.time() python call, which indicates time on the host system). 

After evaluating the several hardware times available for a realsense frames, we currently get the frame time using rs.frame.get_frame_metadata(frameset, rs.frame_metadata_value.time_of_arrival) / 1000. [using time of arrival, TOA]. Referred to https://github.com/IntelRealSense/librealsense/issues/2188 to understand the various timing options available.

Our code loops through the cameras (each has a seperate rs.pipeline) discarding the frame received from a pipeline if it's TOA is less than the query time T. We poll the frames using device.pipeline.poll_for_frames()

Unfortunately, we noticed that this loop now takes 1 second to complete. Ideally since our back-end sync scripts loops through cameras at 4HZ, the expectation was that this condition should be satisified at this frequency, or slightly lesser.

Is there something we're missing? Thanks!

MartyG-RealSense, 2021-08-20:
Hi @sritee   There was a recent L515 multiple camera sync case in https://github.com/IntelRealSense/librealsense/issues/9583 where the RealSense user was experiencing a capture cycle that took approximately 2.8 seconds to complete for all cameras.

They were also checking the time, though their particular method was to store the value of **time.time** in'start' and 'end' variables calculated at different points in the script and then subtract start from end **(end-start**).  A result of 2.8 seconds was clearly wrong, since they expected 640 ms in total to process all of their cameras.

[image]

It sounds as though your hardware sync system's timings are working and the unsatisfactory results are occurring in your timing verification mechanism.  Can you confirm whether that is correct, please?

Case https://github.com/IntelRealSense/librealsense/issues/8112 about apparent frame timestamp discrepancies when using poll_for_frames or wait_for_frames may also be relevant to your problem.

sritee, 2021-08-20:
thanks for the prompt response. the timing is done correctly in our case, and the time.time() calls occur just before we start polling (<few ms gap). 

I should also mention that the polling through all cameras takes less than 10ms (when we don't have condition on their timing), even with inter-cam sync on. This is weird, as we would expect that all 4 cameras get a new frame once every 250ms, and the other cameras are actually turned off at other times. 

When we poll a camera but it's digital state is off, is the behavior just to return an old frame? There is no blocking happening in our case during poll_frames, regardless of any additional conditions we impose.

MartyG-RealSense, 2021-08-20:
The L515 multicam paper states that "data streams do not occur simultaneously" and "L515 halts the stream when it is not projecting", which implies that when a camera is inactive then it should not be returning **any** Frames until it becomes active again.

sritee, 2021-08-20:
Investigated the no frame when camera off case a little more: can confirm poll frames returns the same old frame till the camera is turned on again. Checked both the actual image, as the well as the hardware timestamps. Both are repeats till the next camera period starts.

MartyG-RealSense, 2021-08-21:
If your counter takes the form of a continuously updating printout of a value stored in a variable then it may be that the camera is not actually generating old frames when in the inactive state but that the program is just printing the current stored value of the variable (which is not updated until that camera becomes active again).

In regard to a non-updating image, it is conceivable that the stream could be in a 'paused' state (like when the space bar or pause option is used when live-streaming from the camera in the RealSense Viewer) and un-pauses to resume streaming when the camera becomes active again.  This is speculation on my part though, since the L515 multiple camera paper does not have specific details of the active / inactive mechanism.

sritee, 2021-08-29:
Hi Marty, I do not require further assistance and am closing this issue. For future users of multi-cam, it does look like the camera pipeline returns stale frames even when it's not triggered to be on  (frozen as the last frame when it was on), which can be unintuitive.

closed 2021-08-29 by sritee

MartyG-RealSense, 2021-08-29:
Thanks very much @sritee for the update and for sharing advice for RealSense users with multiple camera projects who are reading this discussion in future!