realsenseai/librealsense #9664: Setting the resolution causes the point cloud display to fail

zma69650, 2021-08-21:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | d455| 
| Firmware Version                   | 05.12.13.50 | 
| Operating System & Version |   win10  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC |
| SDK Version                            |  2.45.0              | 
| Language                            |  {C++        | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
I am using the rs-pointcloud example provided by realsense. When I set the resolution, I added the following code.
`	rs2::config cfg;
	cfg.enable_stream(RS2_STREAM_DEPTH, 640, 480, RS2_FORMAT_Z16, 30);
	cfg.enable_stream(RS2_STREAM_COLOR, 640, 480, RS2_FORMAT_BGR8, 30);
    rs2::pipeline pipe;
    // Start streaming with default recommended configuration
	
	// Use a configuration object to request only depth from the pipeline

	//cfg.enable_all_streams();
	pipe.start(cfg);`
An error occurred during runtime. When I was debugging, I found that the error was in the following code.
`		pc.map_to(color);         // Map the colored depth to the point cloud
		points = pc.calculate(depth); // Generate pointcloud from the depth data
									  /*colorized_depth = color_map.process(f);  */   // Colorize the depth frame with a color map
		
		view.tex.upload(color);   //and upload the texture to the view (without this the view will be B&W)`
The program crashed in this code view.tex.upload(color);

MartyG-RealSense, 2021-08-21:
Hi @zma69650   Are you using the same script from your recent case in https://github.com/IntelRealSense/librealsense/issues/9596 please?  If you are, did you manage to solve the problem that you were having with using **cfg** in that case?

zma69650, 2021-08-21:
Hi@MartyG-RealSense   Thanks for your reply.yes,I didn't solve the problem.I did not find the thread bracket problem, I hope you can help me.

MartyG-RealSense, 2021-08-21:
Comparing your script to **rs-post-processing**, I note that in your script you have not defined the following settings at the head of your script that are referenced within your script:

```
#include <thread>
#include <atomic>
```

https://github.com/IntelRealSense/librealsense/blob/master/examples/post-processing/rs-post-processing.cpp#L9-L10

zma69650, 2021-08-21:
I did not miss these two header files.When I watched the code of'view.tex.upload(color)' run, no error occurred, and it can run normally, but it is all white and white. So I thought the error occurred in'view.tex.upload(color)'  The following link is my code

[https://github.com/zma69650/Arithmetic/blob/master/rs-pointcloud.cpp](url)

MartyG-RealSense, 2021-08-21:
My observation of missing headers was based on your earlier provided script in https://github.com/IntelRealSense/librealsense/issues/9596 that does not list them in the header.

**atomic** is referenced in your script on line 62 but you have commented out the include, preventing it from being accessed.

`//#include <atomic>`

zma69650, 2021-08-21:
Yes, I was negligent.But I still get an error when I add this header file.but can this problem be solved now?

MartyG-RealSense, 2021-08-21:
In your script, you have commented out the colorized_depth reference and use 'color' instead in view.tex.upload(**color**);

But I do not see a definition for 'color' to replace the colorized_depth line that was commented out.

https://github.com/zma69650/Arithmetic/blob/master/rs-pointcloud.cpp#L155

zma69650, 2021-08-21:
I have referred to the practice in rs-pointcloud, and annotated one step of rs-post-processing.
[https://github.com/IntelRealSense/librealsense/blob/master/examples/pointcloud/rs-pointcloud.cpp#L51](url)

zma69650, 2021-08-21:
Because the program runs normally when I don't use cfg, so I didn't care about it.

MartyG-RealSense, 2021-08-21:
Since you are referencing **callbacks** in your script, does it work if your pipe start instruction is like this:

**pipe.start(cfg,callback);**

zma69650, 2021-08-21:
Still not working, I want to try to change the default resolution to what I need, and then don’t use cfg, I don’t know if I can set the default resolution

MartyG-RealSense, 2021-08-21:
The defaults on the D455 camera model will be 848x480 depth and 1280x720 color at 30 FPS.  This is not very different from your cfg resolutions of 640x480 depth and 848x480 color.  So whilst you cannot achieve these resolution settings without cfg (as the pipeline will apply the particular camera model's default configuratio), not using cfg may not make a big difference to the image.

zma69650, 2021-08-21:
Because I am using nite, it needs to pass in 640x480 depth data.

zma69650, 2021-08-21:
Or is there any other way of picture

MartyG-RealSense, 2021-08-21:
How about reducing the depth resolution with a **decimation filter** post-processing filter?

https://github.com/IntelRealSense/librealsense/tree/master/examples/post-processing

zma69650, 2021-08-21:
Thank you for your reply, I will try the method you provided next

closed 2021-08-22 by zma69650