realsenseai/librealsense #9667: L515 using depthframe.get_data() can't have right depth data

Technician13, 2021-08-22:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { L500 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Linux (Ubuntu 18)  | 
| Kernel Version (Linux Only)    |  (e.g. 5.4.0)                                         | 
| Platform                                 | PC |
| SDK Version                            |  { 2.48.0 }                          | 
| Language                            |  {C++ }                          | 
| Segment			|  {Robot }                          | 

### Issue Description
When I use this function, I found that the depth value using this function (get_distance()) is correct, but the depth value of this function(get_data()) is wrong.
The details of the procedure are as follows：

rs2::frameset    frameSet;
rs2::video_frame colorFrame = frameSet.get_color_frame();
rs2::depth_frame alignedDepthFrame = frameSet.get_depth_frame();

depthImage = cv::Mat(cv::Size(640, 480), CV_16UC1, (void*)alignedDepthFrame.get_data(), cv::Mat::AUTO_STEP);
data = (uint16_t*)alignedDepthFrame.get_data();
memcpy(data, (uint16_t*)alignedDepthFrame.get_data(), DEPTH_WIDTH * DEPTH_HEIGHT * 2);
int centerDistance = alignedDepthFrame.get_distance(alignedDepthFrame.get_width()/2, alignedDepthFrame.get_height()/2) ;

The variable (centerDistance) is right ,but after I 'cout' all depth data in variable(data and depthimage), I found depth data was error. I don't kown the reason.

ps: sometimes error depth data is 4 times bigger than the right.

Please help me ! T.T

MartyG-RealSense, 2021-08-22:
Hi @Technician13  It sounds as though you are describing that the distance measurement values at the center of the image are correct but become increasingly inaccurate on the outer areas of the image.  Is this correct, please?

If this is what you are experiencing, it is a known issue with using **frame.get_distance()** with alignment.  Examples of past cases that have been affected by this are provided in https://github.com/IntelRealSense/librealsense/issues/9088#issuecomment-847583702

Is the script that you provided your full script, please?  You are using **alignedDepthFrame** but do not seem to have set up the SDK's **Align processing block** with 'align_to to perform the depth to color alignment.

Technician13, 2021-08-22:
I'm sorry for not describing this problem well. My depth data is read in these two ways. One of the data obtained by reading method is correct, and the other is that all data is wrong. As you see, get_distance() is ok while get_data() is not right

MartyG-RealSense, 2021-08-22:
In the line **rs2::depth_frame alignedDepthFrame = frameSet.get_depth_frame();** you are setting up the contents of the **alignedDepthFrame** variable, but because you have not performed depth to color alignment with align_to before that, the contents of **alignedDepthFrame** will just be the depth frame and not a depth-color aligned image.

ev-mp, 2021-08-23:
@Technician13 , as pointed by @MartyG-RealSense , the issue can be with 
```cpp
data = (uint16_t*)alignedDepthFrame.get_data();
memcpy(data, (uint16_t*)alignedDepthFrame.get_data(), DEPTH_WIDTH * DEPTH_HEIGHT * 2);
```

You need to post the full code segment, including the pipeline configuration and the values assigned to the const vars for review.

Technician13, 2021-08-25:
This is my initialization.
```bash
auto devices = ctx.query_devices();
device_count = devices.size();
while (!device_count)
{
    devices = ctx.query_devices();
    device_count = devices.size();
    return;
}
auto dev = devices[0];
cfg.enable_stream(RS2_STREAM_DEPTH, DEPTH_WIDTH, DEPTH_HEIGHT, RS2_FORMAT_Z16, 30);
cfg.enable_stream(RS2_STREAM_COLOR, IMAGE_WIDTH, IMAGE_HEIGHT, RS2_FORMAT_BGR8, 60);
cout << "enable stream Ok ..\n";
rs2_sensor *depthSensor;
rs2::pipeline_profile selection = pipe.start(cfg);
cout << "start cfg Ok ..\n";

for (int i = 0; i < 5; i++)
{
    frameSet = pipe.wait_for_frames();
}
rs2::video_frame colorFrame = frameSet.get_color_frame();
rs2::depth_frame alignedDepthFrame = frameSet.get_depth_frame();

```
And this is my updating Image function.
```bash
frameSet = pipe.wait_for_frames();
rs2::video_frame colorFrame = frameSet.get_color_frame();
rs2::depth_frame alignedDepthFrame = frameSet.get_depth_frame();
srcImage = cv::Mat(cv::Size(IMAGE_WIDTH, IMAGE_HEIGHT), CV_8UC3, (void*)colorFrame.get_data(), cv::Mat::AUTO_STEP);
depthImage = cv::Mat(cv::Size(DEPTH_WIDTH, DEPTH_HEIGHT), CV_16UC1, (void*)alignedDepthFrame.get_data(), cv::Mat::AUTO_STEP);
memcpy(data, alignedDepthFrame.get_data(), DEPTH_WIDTH * DEPTH_HEIGHT * 2);
```
and I get image by this function
```bash
void updateImage(void)
{   
    thisD435->update();
    srcImage = thisD435->getSrcImage().clone();
    depthImage = thisD435->getDepthImage().clone();
    m_data = thisD435->getData();
    depthData = thisD435->data;
}
```

I DON'T WANT ALIGNED WHOLE DEPTHFRAME TO COLORFRAME, WHICH WASTED MY TIME.
I just get one pixel from colorframe to depthframe and get the depth.
```bash
float depthPixel[2];
rs2_project_color_pixel_to_depth_pixel(depthPixel,depthData,1.0f,0.5,5,&depth_intrin,&color_intrin,&color2depth_extrin,&depth2color_extrin,colorPixel);
int row = depthPixel[1];
int col = depthPixel[0];
float depth = depthData[row * depth_intrin.width + col];
rs2_deproject_pixel_to_point(point , &depth_intrin, depthPixel ,depth);
```
But I can't get the corret data from `point` .
By this way, I needn't to align whole depthframe to colorframe.

ev-mp, 2021-08-25:
@Technician13 , please see the previous comment
```
data = (uint16_t*)alignedDepthFrame.get_data();
memcpy(data, (uint16_t*)alignedDepthFrame.get_data(), DEPTH_WIDTH * DEPTH_HEIGHT * 2);
```

You're copying memory from a region to itself. This is obviously a logical flaw as at best it just wastes CPU clocks. But actually it is worse - according to C standard `memcpy`-ing to overlapping memory regions leads to undefined behavior, so the content of the buffer gets corrupted
https://en.cppreference.com/w/c/string/byte/memcpy

Technician13, 2021-08-25:
But I found using this way on the D435 is ok(two ways are all ok), L515 can't .

Technician13, 2021-08-25:
> @Technician13 , please see the previous comment
>
> ```
> data = (uint16_t*)alignedDepthFrame.get_data();
> memcpy(data, (uint16_t*)alignedDepthFrame.get_data(), DEPTH_WIDTH * DEPTH_HEIGHT * 2);
> ``` [...]

I just used one statement to finish this task, if using only one statement still caused this error?

MartyG-RealSense, 2021-09-02:
Hi @Technician13  Do you require further assistance with this case, please?  Thanks!

Technician13, 2021-09-04:
> Hi @Technician13 Do you require further assistance with this case, please? Thanks!
Yes， I still don't know how to solve this problem. Need I describe this problem again?

MartyG-RealSense, 2021-09-05:
@Technician13   Are you having difficulty with implementing the advice of @ev-mp in the comment https://github.com/IntelRealSense/librealsense/issues/9667#issuecomment-907514050

Technician13, 2021-09-06:
> @Technician13 Are you having difficulty with implementing the advice of @ev-mp in the comment [#9667 (comment)](https://github.com/IntelRealSense/librealsense/issues/9667#issuecomment-907514050)

I don't know the real problem in my program. I try deleting `memcpy` but the depth data from `get_data()` is still error. I put my program on the comment, the data in `depthImage` (I had deleted `memcpy` ) is still error, not as he said

MartyG-RealSense, 2021-09-06:
Hi @ev-mp   Do you have any further advice for @Technician13 please as deleting memcpy did not resolve their problem?

MartyG-RealSense, 2021-09-12:
Hi @Technician13  Have you made any progress in solving your problem or do you still need assistance, please?  Thanks!

Technician13, 2021-09-12:
> Hi @Technician13 Have you made any progress in solving your problem or do you still need assistance, please? Thanks!
NO， I used another way to get my depth data, but I still want to know and solve this problem.

MartyG-RealSense, 2021-09-12:
There is no advice that I can add to the information already provided by @ev-mp  unfortunately.  As you have achieved a solution by another method, I would therefore recommend closing this question, regrettably.

closed 2021-09-18 by MartyG-RealSense