realsenseai/librealsense #9671: Any possibility of open sourcing T265 Odometry Algorithm

zainmehdi, 2021-08-24:
Since it has been announced that Intel will be shutting down its Realsense Division which is really unfortunate. T265 was one of the best readily available odometry solution available in the market. Is there any possibility for this algorithm being open sourced so the community could  build on it and the cycle goes on  ?

boilerbots, 2022-01-07:
I left this comment on the Intel forum:

Since the T265 is being terminated and many people appreciate the high performance and low power of this device, can the source code be shared?

The camera uses a Myriad 2 processor, tightly integrated with the camera and the IMU. I see there are at least two alternatives where the code could be re-used. One could use a Compute Stick or an Oak-D camera, both also contain the Myriad 2 processor.

I think the T265 has a lot of opportunities to grow but it takes time for new sensors to be integrated into new products. Unfortunately Intel didn't give the T265 enough time to grow. Let the technology grow by sharing it.

zainmehdi, 2022-01-10:
So many people were using it and the users were increasing each and every day. Intel just couldnt cope with the added pressure of dealing with the bugs and performance demand.

karrson, 2023-02-22:
Intel needs to respond to this. Why did I even buy the Intel RealSense if I can't even integrate it into my product now? I would at least like permission to reverse engineer and redistribute it. Pretty please? :)

closed 2026-10-23 by MartyG-RealSense