realsenseai/librealsense #9672: Error type: Access Violation!

Yarong-zhang, 2021-08-24:
| | |
|---|---|
|**librealsense**|2.48.0 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
Unhandled exception escaping from a worker thread!
Error type: Access Violation!
```

MartyG-RealSense, 2021-08-24:
Hi @Yarong-zhang  This is a rare error.  The best available reference regarding this problem and possible solutions for it is at https://github.com/IntelRealSense/librealsense/issues/7435 

Before working through the suggestions in that discussion though, you could run the Windows installer file from the 'Assets' file list on the SDK **Releases** page (**Intel.RealSense.SDK-WIN10-2.48.0.3381.exe** for Windows 10 or **Intel.RealSense.SDK-WIN7-2.48.0.3381.exe** for Windows 7).  Even if you have already used this installer, it should overwrite any previous installation of the SDK performed using this method and so may resolve problems introduced during or after the previous installation.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.48.0

Yarong-zhang, 2021-08-24:
Thank you for your reply. I use this installer again, but the problem dosen't be solved.

MartyG-RealSense, 2021-08-24:
Thank you for testing with the installer.

Do you experience the same error in the Viewer if you download and launch the standalone **Intel.RealSenseViewer.exe** version of it from the 'Assets' file list?

You should run the Intel.RealSenseViewer.exe file with a double-leftclick from the folder location that you download it to and not by using the launch shortcut icon on the Windows desktop.

[image]

closed 2021-09-07 by MartyG-RealSense