realsenseai/librealsense #9673: cannot be started after modify the options in the software

Yarong-zhang, 2021-08-24:
| | |
|---|---|
|**librealsense**|2.48.0 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
GL_INVALID_VALUE - C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\common\opengl3.cpp:569

```

MartyG-RealSense, 2021-08-24:
Hi @Yarong-zhang   Could you try launching the **rs-capture** example program using the instructions in https://github.com/IntelRealSense/librealsense/issues/8024#issuecomment-747630656 please.  This will indicate whether or not the problem is with OpenGL graphics support on your computer.  If the rs-capture program does not run too, I recommend updating the graphics drivers on your computer to update the OpenGL support that the RealSense Viewer and rs-capture programs require.

Yarong-zhang, 2021-08-24:
Hello, I tried the way you said that launching the rs-capture example program using the instructions in #8024 (comment). Unfortunately, the rs-capture doesn't work.

In addition, I forget if that's what I changed the GLSL option.  If so, what should I do?

MartyG-RealSense, 2021-08-24:
If rs-capture does not run then that suggests to me that it is an issue with the OpenGL graphics drivers, in which case updating the graphics drivers of your Windows PC may resolve the issue.

If that does not make a difference then you could next try reinstalling the entire SDK using the automated installer file in the 'Assets' file list on the SDK **Releases** page.  If you previously installed the SDK using this method then repeating the process will overwrite the previous SDK installation and so may clear the problem.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.48.0

[image]

closed 2021-09-07 by MartyG-RealSense