realsenseai/librealsense #9679: [unity][d435] getting left IR image and formatting it to work with shadergraph

TheBricktop, 2021-08-24:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { d435 } | 
| Firmware Version                   | (5.12) | 
| Operating System & Version |   {Win 10 |                                        | 
| Platform                                 | PC  |
| SDK Version                            |  { 2 }                          | 
| Language                            |  {C#/unity }                          | 
| Segment			|  {Smartphone/VR/AR/}                          | 

### Issue Description
Im trying to format the left IR camera stream and im having some issues:
1. Im blitting the texture from outtexture of rsstream texture renderer, and while it works with color stream, the y8 formatted ir stream gives me black texture, what should i do to convert it to rgb8 ?
2. how to set the ir stream to filter out the  ir dot pattern in unity?

MartyG-RealSense, 2021-08-24:
Hi @TheBricktop  

1.  The D435 and D435i camera models do not have an RGB8 color stream from the left IR imager, just Y8 and Y16 IR.  This feature is on the D415 and D455 models. 

**D435**
[image]

**D455**
[image]

2.  The visibility of the dot pattern can be reduced by reducing the value of the **Laser Power** setting in the RealSense Unity wrapper's **RealSenseDeviceInspector** component.  This is because the visibility of the pattern is tied to Laser Power, reducing in visibility as Laser Power is reduced or brightening if Laser Power is increased.  You can also turn off the pattern entirely by setting the **EmitterEnabled** option to off.  

[image]

Bear in mind though that unless the camera is being used in a well lit location, reducing laser power or disabling the emitter can darken the infrared image and also reduce depth image quality if you are using the depth stream.

TheBricktop, 2021-08-25:
About RGB8 format im just asking how to convert those values so that they are visible in a render texture because all im seeing now is a black square. What format should the render texture be in?

MartyG-RealSense, 2021-08-25:
Can you get the desired output if you change the Y8 format's definition in RsStramTextureRenderer.cs  from  TextureFormat.Alpha8; to the RGB8 format's **TextureFormat.RGB24;**

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/unity/Assets/RealSenseSDK2.0/Scripts/RsStreamTextureRenderer.cs#L20

TheBricktop, 2021-08-30:
Ive tried the RGB24 format but it was throwing errors about overreading so ive changed to R8 and it seems fine but still I cant get the second IR imager to stream. If I set stream index to anything but 1 on IR source it cant resolve the request.

MartyG-RealSense, 2021-08-30:
Is the camera on a USB 3 connection, please?  Infrared 2 (right-side infrared) is not supported a on USB 2 connection.

TheBricktop, 2021-08-30:
Yup that was the culprit.
Im curious why the engineers chose the ir imager that is closer to color to be supported on 2.1 and not the one further right.

MartyG-RealSense, 2021-08-30:
The center-line of the left IR imager is the 0,0,0 origin of the coordinate system on 400 Series cameras and so it has to always be available no matter what type of USB connection the camera is using.

MartyG-RealSense, 2021-09-05:
Hi @TheBricktop  Do you require further assistance with this case, please?  Thanks!

TheBricktop, 2021-09-06:
Im good for now.

MartyG-RealSense, 2021-09-06:
Thanks very much @TheBricktop for the update.  I will close the issue.  Please feel free to re-open the case or create a new one at a future date if you need to.  Thanks again!

closed 2021-09-06 by MartyG-RealSense