realsenseai/librealsense #9680: realsense viewer launch problem

Yarong-zhang, 2021-08-25:
| | |
|---|---|
|**librealsense**|2.48.0 RELEASE|
|**OS**|Windows10|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
GL_INVALID_VALUE - C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\common\opengl3.cpp:569

```

The way I have tried as follow:
1. Reinstall Intel.RealSense.SDK-WIN10-2.48.0.3381.exe
2. run the Intel.RealSenseViewer.exe file with a double-leftclick from the folder location that I download it to and not by using the launch shortcut icon on the Windows desktop.
3. launching the rs-capture
4. updating the graphics drivers

but the problem doesn't be solved, Could you help me ? please

Yarong-zhang, 2021-08-25:
[image]
Which one should I upgrade?
I've already upgraded Intel UHD.

MartyG-RealSense, 2021-08-25:
Hi @Yarong-zhang   Assuming that the GeForce GTX is used by your computer as its default graphics instead of the Intel UHD, please update the Geforce graphics drivers too.

Yarong-zhang, 2021-08-31:
[image]
I tried all the versions of RealSense,  only the version lower than 2.44 can be opened, but the interface is like this, I installed the SDK of 2.44, still like this, what should I do

MartyG-RealSense, 2021-08-31:
I have only seen a couple of past examples of this visual distortion in the RealSense Viewer.

One cause of it may be insufficient memory for the **texture atlas** - as described in https://github.com/IntelRealSense/librealsense/issues/7533 - though that was on a low-specification Raspberry Pi board rather than a PC.

Another cause may be the **Font Samples** setting of the Viewer, as described in https://github.com/IntelRealSense/librealsense/issues/7479#issuecomment-702880652

Yarong-zhang, 2021-09-01:
Thank you very much! The problem has been solved and can be started normally.

closed 2021-09-01 by Yarong-zhang

MartyG-RealSense, 2021-09-01:
That is excellent news, @Yarong-zhang - thanks very much for the update!