realsenseai/librealsense #9681: D435i USB接口偶尔被识别成USB2.1

meiy-px, 2021-08-25:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435i } | 
| Firmware Version                   | (Newest) | 
| Operating System & Version |   {Ubuntu 20.0.4}  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC& NVIDIA Jetson  |
| SDK Version                            |  { Newest }                          | 
| Language                            |  {C++ }                          | 
| Segment			|  {Robot }                          | 

### Issue Description
#9529 
按照推荐，使用了带螺丝的线束固定USB接口，还是出现了USB接口被识别成USB2.1的情况。

我们尝试将D435i直接连接到PC、通过USB hub连接到PC、直接连接到NVIDIA AGX、通过USB hub连接到NVIDIA AGX、更换摄像头、更换线束、更换NVIDIA AGX、更换USB hub，甚至更换供电电源，都出现概率性地识别成USB2.1的情况。

在机器人运动过程中，还出现了启动时正常识别成USB3.2，运动时突然变成USB2.1的情况。

这个问题比较严重，目前发现约有52%的摄像头曾经出现过识别成USB2.1的问题。

请问D435i是识别到什么问题才判断是USB2.1的？比如供电不稳定？

MartyG-RealSense, 2021-08-25:
Translation:  D435i USB interface is occasionally recognized as USB2.1.  According to the recommendation, a wire harness with screws is used to fix the USB interface, but the USB interface is still recognized as USB2.1.

We tried to connect D435i directly to PC, connect to PC via USB hub, connect directly to NVIDIA AGX, connect to NVIDIA AGX via USB hub, change camera, change wiring harness, change NVIDIA AGX, change USB hub, and even change power supply. 
 Probabilistically recognized as USB2.1.

During the movement of the robot, it was recognized as USB3.2 when it started, and suddenly became USB2.1 when it was moving.  This problem is more serious.  It has been found that about 52% of the cameras have been identified as USB2.1.  What problem did D435i recognize before judging it is USB2.1?  For example, the power supply is unstable?

*****************

Hello @meiy-px  When you say that you are using a wire harness with screws, do you mean that you are using a screw lock USB cable to fix the micro sized end of the USB-C cable into the side of the camera please?

If you are using a screw lock cable, does the full sized USB-A connector at the computer end of the cable also have a screw lock?   If it does not then the connector could still slowly be pulled out of the computer's USB port during motion even though it is fised into the camera's port at the other end.  Once a USB cable is partially withdrawn from a USB3 port by a small amount then the camera can disconnect and then reconnect misdetected as USB2.1.  

**************

你好@meiy-px 當您說您使用的是帶螺絲的線束時，您的意思是您使用螺絲鎖 USB 電纜將 USB-C 電纜的微型端固定到相機的側面嗎？

如果您使用的是螺絲鎖線纜，線纜末端的全尺寸 USB-A 連接器是否也有螺絲鎖？ 如果沒有，則即使連接器已插入另一端的相機端口，在運動過程中仍可能緩慢地將連接器從計算機的 USB 端口中拔出。 一旦 USB 電纜從 USB3 端口部分拔出少量，相機就可以斷開連接，然後重新連接，誤檢測為 USB2.1。

meiy-px, 2021-08-25:
@MartyG-RealSense Yes, the screws are only in the type C side. But when we did the tests, the robot was not moving.

MartyG-RealSense, 2021-08-25:
Okay, thank you for the information.  Which firware driver version is the D435i using please?  If the firmware has not been updated since the camera was purchased then it may have an old firmware that dates back to a time before fixes were added to the firmware to reduce the chance of a USB 2.1 misdetection occurring.

meiy-px, 2021-08-25:
The firmware version is 05.12.15.50. I think this is the latest version.

MartyG-RealSense, 2021-08-25:
When the camera connection becoes USB 2.1 in the Viewer, does it correctly become 3.2 if you go to the **More** option near the top of the Viewer's options side-panel and select the **Hardware Reset** option from its drop-down menu?

[image]

meiy-px, 2021-08-25:
Most of the time it works while applying a hardware reset operation. But the time of reset is too long, about 9 seconds. I think it's not acceptable in robot use case.

MartyG-RealSense, 2021-08-25:
A hardware reset should normally take about 3 seconds to complete, so 9 seconds sounds unusually long.  It sounds like some kind of slowdown may be occurring on your computer.   Could you check your computer's CPU usage with an Ubuntu system monitoring tool such as **htop** when the Viewer is running and see how close to 100% the CPU usage is?

meiy-px, 2021-08-25:
When the raslsense viewer is running, the occupancy rate of CPU is 35% ~ 70%.

MartyG-RealSense, 2021-08-25:
That CPU percentage sounds okay and not excessive.  

How does it behave if you undo the screws at the camera end of the cable and have the connector inserted loose?  I ask this because I recall a case in https://github.com/IntelRealSense/librealsense/issues/8830 (on D455 though, not D435i) where performance was negatively affected when the screws were tightened.

 Also, does the USB 2.1 misdetection occur if you boot the computer without the camera inserted and then insert the camera after boot has completed?

meiy-px, 2021-08-25:
How does it behave if you undo the screws at the camera end of the cable and have the connector inserted loose? 
==>We didn't do this, but tried to insert the USB-A connector loosely, it became worse.

Also, does the USB 2.1 misdetection occur if you boot the computer without the camera inserted and then insert the camera after boot has completed?
==>I have tried to open realsense  viewer and then insert the camera. There was also small probility that the port was  recognized as USB 2.1.

MartyG-RealSense, 2021-08-25:
Altering the USB-A connector would not likely provide an indicator of whether there was a problem with the USB-C screws being wound into the holes in the side of the camera, as it would not account for the possibility that the screws on that particular screw-lock cable might be disrupting the camera internally.

Does the problem occur if you use the official 1 meter cable supplied with the camera?  USB-C cables can have quality issues, so there may be a quality issue in your own-choice USB-C screw-lock cable that affects its performance if the problem does not occur with the official cable.  It should of course also be a USB 3 cable and not a USB 2 one.

meiy-px, 2021-08-25:
Yes, both the official cable and our own-choice cable have been tested. But the results are almost the same.

Do you know what dose the camera find before judging the port is USB2.1? I think it will be very useful for analyzing the problem.

MartyG-RealSense, 2021-08-25:
I believe that if you are correctly using a USB 3 cable (which you definitely are when using the official cable) then the decision whether to identify the connection as USB 2.1 or USB 3.2 will primarily depend on the USB port that the USB-A end of the cable is plugged into and that particular port's **USB controller**.  The USB controller of the computer and the USB controller of the hub may have different manufacturers, so a camera may work better with one manufacturer's controller than another's.  Case https://github.com/IntelRealSense/librealsense/issues/6899 discusses this subject.

Having said that, I note that you are encountering the problem with both direct connection to computer and with USB hub.

If you are using an unpowered hub (one that draws its power from the computer instead of from a mains electricity power point) then a hub's ports may be subject to the same power-supply instabilities that the computer's USB ports may have (as the computer ports draw power from the computer's internal power supply and so may be less stable than the mains socket supplied power of a powered USB3 hub).

I note that you also mentioned earlier that "it was recognized as USB3.2 when it started, and suddenly became USB2.1".   Does this mean that the camera may start with being on USB 3.2 but it switches to USB 2.1 after you have enabled the camera streams?

meiy-px, 2021-08-28:
We will try this

meiy-px, 2021-08-30:
@MartyG-RealSense We have tried to set the 'Emitter Enabled' to 'OFF' , the test result was the same as before.

MartyG-RealSense, 2021-08-30:
Thank you for testing.  Does the Depth Quality Tool program - **rs-depth-quality** - have the same 52% chance of being identified as 2.1 after the program is launched (as this tool's design is structurally similar to the RealSense Viewer).

[image]

MartyG-RealSense, 2021-09-11:
Case closed due to no further comments received.

closed 2021-09-11 by MartyG-RealSense

Zephyr69, 2022-09-13:
I may have a similar issue like this. My D405 is always connected to usb2.1 after booting (although there isn't a physical usb2.1 port). Replugging is strictly required for it to go usb3.

Environment: Ubuntu 18.04, Jetson Xavier NX
SDK&pyrealsense: latest
Cable: the official cable.

Zephyr69, 2022-09-13:
> Does it show USB 3 if you reset the camera in the Viewer using the instructions above at [#9681 (comment)](https://github.com/IntelRealSense/librealsense/issues/9681#issuecomment-905286524) instead of unplugging it and re-inserting it?
> 
> If a camera reset does show USB 3 and you are writing your own pyrealsense2 script then you can place a **hardware_reset()** instruction in the script to automatically reset the camera when the script launches.

Thanks for the quick reply! I'll try it in ~11hours when I get back to work.

MartyG-RealSense, 2022-09-13:
You are very welcome!  If you find that a hardware reset in the Viewer does succeed, there is a simple example below of implementing an automatic reset in a pyrealsense2 script.

```
ctx = rs.context()
devices = ctx.query_devices()
for dev in devices:
dev.hardware_reset()
```

Zephyr69, 2022-09-14:
@MartyG-RealSense Both the manual and script hardware reset works! Thank you!

MartyG-RealSense, 2022-09-14:
You are very welcome, @Zephyr69 - I'm pleased that I could help!