realsenseai/librealsense #9682: D435i对不连续深度的检测问题

meiy-px, 2021-08-25:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435i } | 
| Firmware Version                   | (Newest) | 
| Operating System & Version |   {Ubuntu 20.0.4}  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { Newest }                          | 
| Language                            |  {C++}                          | 
| Segment			|  {Robot }                          | 

### Issue Description
在一面白墙前放置一块棋盘格，深度差约0.5m，D435i检测深度将两者之间没有深度的区域填充，并且使棋盘格的边缘不再锐利。爬楼梯时也出现这种现象，使得楼梯检测不准确。
这种现象是什么原因导致的？是否可以通过调节参数解决？
[image]
[image]

MartyG-RealSense, 2021-08-25:
Translation: D435i's detection problem of discontinuous depth.  Place a checkerboard in front of a white wall with a depth difference of about 0.5m.  D435i detects the depth to fill the area with no depth between the two, and make the edges of the checkerboard no longer sharp.  This phenomenon also occurs when climbing stairs, making the stairs detection inaccurate.
 What causes this phenomenon?  Can it be solved by adjusting the parameters?

******************

Hello @meiy-px  The effect in your image resembles a phenemenon called **ghost noise**.  This noise can take different forms, but the one in your image looks like a type where the points on a cloud can get dragged outward and make the observed object or surface look distorted.  Examples of such distortion are in https://github.com/IntelRealSense/librealsense/issues/4553

The case that I linked to references a point cloud of a banana that would become elongated when the banana was viewed from the side of it, similar to your image of the checkerboard being viewed from a side angle.

[image]

Simply viewing the checkerboard from the front instead of at a side angle might resolve the curving.  If it does not then you could perform a **Dynamic Calibration** on the camera to robustly calibrate both the depth and RGB sensors of the camera.

If you have edges that are not sharp then using the **Medium Density** camera configuration visual preset may help.  This setting provides a good balnace between accuracy and the amount of detail on the depth image.

*************

你好@meiy-px 您圖像中的效果類似於一種稱為鬼噪的現象。這種噪聲可以採用不同的形式，但圖像中的噪聲看起來像是一種類型，其中云上的點可以向外拖動並使觀察到的對像或表面看起來失真。這種失真的例子在 https://github.com/IntelRealSense/librealsense/issues/4553

我鏈接到的案例引用了香蕉的點雲，當從香蕉的側面觀看時，香蕉的點雲會變長，類似於從側面角度觀看棋盤格的圖像。

[image]

簡單地從正面而不是從側面觀察棋盤可能會解決彎曲問題。如果沒有，那麼您可以在相機上執行動態校準，以穩健地校準相機的深度和 RGB 傳感器。

如果您的邊緣不清晰，則使用中密度相機配置視覺預設可能會有所幫助。此設置在精度和深度圖像的細節量之間提供了良好的平衡。

meiy-px, 2021-08-25:
[image]
As the picture above shows, the fault depth will affect image matching because there were no depth in the incorresct area. It seems like effect of filtering. Can we avoid this phenomenon through some way?

MartyG-RealSense, 2021-08-25:
If you need to capture from a side angle then you could next test whether the problem in the image is being caused by confusion produced by that angled wall section by observing from the other side of the board.

[image]

meiy-px, 2021-08-25:
Before the test, we have adjusted the camera and ensure the optical axis of the camera is vertical to the wall. And also we make sure that the checkboard is almost parallel to the wall.

MartyG-RealSense, 2021-08-25:
As the non-depth image of the board is pointing straight towards it, are you capturing a point cloud with the camera in front of the board and then rotating the view of the point cloud (which was what caused the stretching distortion in the banana image)

meiy-px, 2021-08-28:
Rotate the image is not our perpose. The point cloud between  the edge of the checkboard and the wall is not real. And it leads to mistake when matching the point cloud. This is what troubling us.

MartyG-RealSense, 2021-08-28:
If you are not rotating the view of the point cloud then I would still recommend moving the checkerboard along to the center of the wall so that the angled wall section is not in the image, in order to eliminate the possibility that the angled wall is confusing the depth perception.

[image]

closed 2021-09-09 by MartyG-RealSense