realsenseai/librealsense #9684: Pyrealsense2 RuntimeError: No device connected

ydejonghe, 2021-08-25:
| Required Info                         |                                             |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435                           | 
| Firmware Version                   | 05.12.10.00                 | 
| Operating System & Version  |  Ubuntu 20.04.3 LTS    | 
| Kernel Version (Linux Only)    |  5.11.0-27-generic           | 
| Platform                                   | PC                                    |
| SDK Version                            |  unsure                          | 
| Language                                |  python                          | 
| Segment			          |  Robot                          | 

### Issue Description
Hello, I'm facing an issue in opening my realsense camera on linux. The camera can be opened and used be realsense-viewer, but not using pyrealsense2

I installed `pip install pyrealsense2` which gave me version `2.49.0.3474`
Then I installed a prebuilt realsense sdk to get the `realsense-viewer`. I followed [these steps](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md)
It seems the latest ported SDK version is `2.36.0`.

Now that I have the serial number of the realsense device, I can run a python script loading it. But it doesn't seems to find it. (The code sample I use is working the windows machines I have).
```python
  File ".../realsense.py", line 21, in __init__
    self.pipe.start(config)
RuntimeError: No device connected
```

I guess that this could be due to some version conflicts between the installed SDK and pyrealsense2, but there is no more recent prebuilt and pypi doesn't have `pyrealsense2==2.36.0` anymore so I cannot try matching versions.

Can you help me on this ?

MartyG-RealSense, 2021-08-25:
Hi @ydejonghe  Support for Python 3.8 with the **pip install pyrealsense2** method was not added to the RealSense SDK until version **2.44.0**.  Before that, Python 3.7 was the most recent version that a pip install supported and the Python wrapper had to instead be built from source code in order to be used with Python 3.8.

 It is unusual that SDK 2.36.0 was installed when you used the SDK from packages using the **distribution_linux.md** instructions page, as it should have installed a recent version (the latest at the time of writing this is **2.49.0**).  

Which computer or computing device are you using please (PC, Nvidia Jetson, etc)

ydejonghe, 2021-08-25:
I'm using a PC with Ubuntu 20 LTS up to date
also my `/etc/apt/sources.list` mentions:
```shell
# deb http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic main
# deb-src http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic main
# deb http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo focal main
# deb-src http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo focal main
deb https://librealsense.intel.com/Debian/apt-repo focal main
# deb-src https://librealsense.intel.com/Debian/apt-repo focal main
```

MartyG-RealSense, 2021-08-25:
Thank you.  You should be using the https address **https://librealsense.intel.com/Debian/apt-repo focal main** as the other http addresses with the # beside them would no longer apply (Intel switched their URL addresses from http to https).

So can you confirm please that you have followed the steps below:

1.  Registered the server's public key.

2. Used this instruction to add the server to the list of respositories:

`sudo add-apt-repository "deb https://librealsense.intel.com/Debian/apt-repo $(lsb_release -cs) main" -u`

3.  Installed the two compulsory packages below:

```
sudo apt-get install librealsense2-dkms
sudo apt-get install librealsense2-utils
```

And by doing so, SDK 2.36.0 is installed on your computer instead of the latest version.

ydejonghe, 2021-08-25:
I installed all those yes :)

I found the cause of the issue.
There was an old realsense SDK 2.36.0  already installed on my system, but in an unusual way ...
But it was on the `LD_LIBRARY_PATH` so it was conflicting with the freshly installed realsense SDK 2.44.0 (the repository you mentionned did work in fact) and so python was loading the old libraries instead of the new installed ones.

The issue was all due to me, so I apologize for wasting your time.
Have a pleasant day !

closed 2021-08-25 by ydejonghe

MartyG-RealSense, 2021-08-25:
No need at all for apologies - it's great that you found the solution.  Thanks very much for the update!