realsenseai/librealsense #9687: Prebuilt pyrealsense2 for macOS released

cansik, 2021-08-25:
For many projects I (and others https://github.com/IntelRealSense/librealsense/issues/6402, https://github.com/IntelRealSense/librealsense/issues/5275, https://github.com/IntelRealSense/librealsense/issues/3864) would like to use python and a realsense camera. Of course it is possible to build the librealsense with python bindings yourself, but an installation with pip would be so much easier.

Because of that, I created the repository [cansik/pyrealsense2-macosx](https://github.com/cansik/pyrealsense2-macosx) which contains a powershell script to automate the build process. For **Big Sur** and **Catalina** a Github Action creates pre-built wheel files and provides them in the [releases](https://github.com/cansik/pyrealsense2-macosx/releases/tag/v2.48.0).

Now the installation on macOS is as simple as that:

```bash
brew install librealsense
pip install pyrealsense2 -f https://github.com/cansik/pyrealsense2-macosx/releases/tag/v2.48.0
```

*Currently Supported Versions*

- OS: macOS Catalina (`10.15`), macOS Big Sur (`11.0`)
- Python: `3.6`, `3.7`, `3.8`, `3.9`

I hope this helps others too and hopefully sooner or later Intel will include the macOS packages in their official [pyrealsense2](https://pypi.org/project/pyrealsense2/) PyPI repo.

MartyG-RealSense, 2021-08-25:
Thanks so much @cansik for creating your brew-based prebuilt MacOs installation solution for pyrealsense2 and sharing it with other Mac users in the RealSense community!

MartyG-RealSense, 2021-09-01:
Thanks very much again @cansik for your installation guide.  I will close this case, as the guide is accessible through web searching.

HiWiQiao, 2023-04-28:
sorry, i can not find requirements.txt

HiWiQiao, 2023-04-28:
it is doesn't work in Ventura

PaulRuan2023, 2023-06-18:
how to solve the problem no device connected

MartyG-RealSense, 2023-06-18:
Hi @PaulRuan2023  Are you using MacOS Ventura, please?

PaulRuan2023, 2023-06-18:
Yes

MartyG-RealSense, 2023-06-18:
There have been problems reported with using RealSense cameras with the latest Ventura OS.  There were also problems with the preceding Monterey version but solutions were eventually found by RealSense Mac users.  There is not a clear solution for Ventura problems yet, unfortunately.

The best place to discuss this issue with other members of the RealSense Mac is likely to be at https://github.com/IntelRealSense/librealsense/issues/11815