realsenseai/librealsense #9698: ERROR: Unhandled exception at realsense2.dll

anhTuan0712, 2021-08-28:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentation](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is your question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  L515  | 
| Firmware Version                   | 1.5.8.1 | 
| Operating System & Version |  Win 10 Pro (version 2004) | 
| Kernel Version (Linux Only)    |                      | 
| Platform                                 | PC|
| SDK Version                            | 2.47 and 2.49                         | 
| Language                            |  c#                        | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
I am facing this error: "Unhandled exception at (realsense2.dll") when using version SDK: 2.36, 2.42, 2.47 and 2.49
The error images I obtained: 
[image]
[image]
### Source code: 
https://drive.google.com/drive/folders/1BQSOBW8dOWEfjp_gpFwCeOeK8HP3NRL7?usp=sharing
### Memory dump file: 
https://drive.google.com/drive/folders/1BQSOBW8dOWEfjp_gpFwCeOeK8HP3NRL7?usp=sharing
### Measures I have tried to catch the error but with no success:
1) AccessViolationException (Link: https://docs.microsoft.com/en-us/dotnet/api/system.accessviolationexception?view=net-5.0)
2) AppDomain.UnhandledException (Link: https://docs.microsoft.com/enus/dotnet/api/system.appdomain.unhandledexception?view=net-5.0)
3) Application.DoEvents (Link: https://docs.microsoft.com/en-us/dotnet/api/system.windows.forms.application.doevents?view=net-5.0)
### Add information:
Also, this source code when I run it on Win 10 version 1909, this problem doesn't happen.
### I rate rake to anyone who has a solution to this problem
<Describe your issue / question / feature request / etc..>

anhTuan0712, 2021-08-28:
Hi @MartyG-RealSense , Hi @RealSenseCustomerSupport , Hi @RealSenseSupport 
Please help me to solve this problem.

MartyG-RealSense, 2021-08-28:
Hi @anhTuan0712   The key information in this case is that your program works on Windows 10 but not on Windows 10 Pro indicating that the code itself is okay), and that the error is occurring at the **devices = CTX.QueryDevices()** line.  As this line queries the camera, this suggests to me that the best place to start investigating is to check whether there is a block to accessing the camera on Windows 10 Pro.

1.  First, please use the instructions in the link below to check whether access to the camera hardware and the ability of applications to access the camera are both set to On in the Windows privacy settings.

https://github.com/IntelRealSense/librealsense/issues/3381#issuecomment-503424789

2.  If they are both set to On, you could then explore whether there is a **group policy** set on your Windows 10 Pro computer that may be blocking access to the camera.  Google for the term **windows 10 pro group policy camera** for more information about this.

anhTuan0712, 2021-08-28:
Hi @MartyG-RealSense, Thanks for your quick reply
But sorry I gave the wrong information, I am using win 10 Pro, only versions 1909 and 2004 are different (error only occurs on version 2004)
* This error occurs randomly, easy to happen if you disconnect the camera L515 from the PC while the program (my source code above) is running.

MartyG-RealSense, 2021-08-28:
No rpoblem at all.  A RealSense C# user in https://github.com/IntelRealSense/librealsense/issues/6166#issuecomment-607190650 took the approach of setting up an exception catch for QueryDevices() to handle a disconnection event.

anhTuan0712, 2021-08-28:
Thank @MartyG-RealSense, I will try this code and let you know as soon as the results are available

anhTuan0712, 2021-08-29:
Hi @MartyG-RealSense 
It still fails, the time when I get the exception "Frame didn't arrive within 5000." and somewhere undefined in realsense2.dll.
[image]
[image]
### Here is my new source code: 
 https://drive.google.com/drive/folders/1BQSOBW8dOWEfjp_gpFwCeOeK8HP3NRL7?usp=sharing (Programnew.cs file)
### I can't find the realsense.pbd file, I hope you can provide it to me so I can provide more detailed error information.

MartyG-RealSense, 2021-08-29:
The main reason that I can think of why a **pipe.Stop()** instruction would generate an error is if the pipeline was already inactive (i.e it hadn't been started already, or was stopped before the pipe.Stop() instruction was called).  This is because you cannot stop the pipeline before it has been started.   I note that you received errors that stated 'The video recording device is no longer present' and 'Frame didn't arrive within 5000', suggesting that the camera may not currently be accessible.

My understanding from reading the Visual Studio documentation link below is that **.pdb** debug files are located in your project folder, inside a sub-directory called **/debug**

https://docs.microsoft.com/en-us/visualstudio/debugger/specify-symbol-dot-pdb-and-source-files-in-the-visual-studio-debugger?view=vs-2019

Providing the .pdb file to me in this discussion may not be useful though, as C# programming is not one of my specialist areas of RealSense knowledge.

As the problem in your project occurs on Windows 10 Pro build 2004 and not 1909, it may not be a problem with the RealSense SDK but with a change that Microsoft has made in Windows10 Pro that causes detection of the camera to be lost.  This has sometimes happened in the past with RealSense cameras when Microsoft have released a new Windows build or a smaller update.  In those situations, the only remedy may be to wait and see if the problem is rectified in a future build as a positive side-effect of a change made elsewhere in the build.

anhTuan0712, 2021-08-30:
There is a difference between version 2004 and version 1909 described here
[image]
https://docs.microsoft.com/en-us/windows/win32/api/timeapi/nf-timeapi-timebeginperiod
There was 1 other library that also had the error and I solved it as described above. I hope this can help you.

MartyG-RealSense, 2021-08-30:
Thanks very much @anhTuan0712  for the information.  Does your solution for the other library work with this case too?

anhTuan0712, 2021-09-01:
No, this case cannot be resolved. There are 2 main reasons:
1) maybe I didn't fully wrap the Intel.RealSense.
2) It is possible that this workaround is not suitable for the current error.

MartyG-RealSense, 2021-09-01:
Does it run if you untick the **Break when this exception type is thrown** box?

[image]

It is not a 'clean' programming technique to disable exception throwing though, admittedly.

closed 2021-09-13 by MartyG-RealSense