realsenseai/librealsense #9700: rs-save-to-disk: error - no device connected

vuhuulan, 2021-08-30:
I tried running rs-save-to-disk but didn't succeed. At first, it seems like it's waiting for around 15s (cursor blinking), then the following message appears: 
`RealSense error calling rs2_pipeline_start(pipe:0x5646aa0aa760):
    No device connected`
I checked the camera with rs-hello-realsense and rs-capture and it seems to be working normally. Do you have any ideas to fix this?

_Originally posted by @vuhuulan in https://github.com/IntelRealSense/librealsense/issues/1485#issuecomment-906934884_

MartyG-RealSense, 2021-08-30:
Hi @vuhuulan  Can you confirm please that you did not have any other RealSense programs running (other examples, the RealSense Viewer tool, etc) when you tried to launch rs-save-to-disk.

vuhuulan, 2021-08-31:
Hi, thank you for your reply. Yes, I could confirm that there is no other program running in background.

MartyG-RealSense, 2021-08-31:
rs-save-to-disk is apparently meant to save the PNGs to your 'current working directory', as described in https://github.com/IntelRealSense/librealsense/issues/2918 

I tested the program thoroughly with D435i but was not able to get it to generate PNGs unfortunately.  So I would recommend skipping this example program, or building a similar C++ single-image capture program created by a RealSense user in https://github.com/IntelRealSense/librealsense/issues/2219#issuecomment-412899468 if that feature is important to your project.

Please note that this user-created script specifically looks for a camera with the name **D415** so you should edit this reference to **D435i** to use it with your particular camera model if you decide to use the script.

[image]

vuhuulan, 2021-09-01:
Thank you so much! I will try it out and let ask if there is problem. I will let this be open for a few days in wait of such case.

MartyG-RealSense, 2021-09-01:
Thanks very much @vuhuulan - good luck!

vuhuulan, 2021-09-10:
Hello, I don't where to put this file, as well as how to write a CMakeFiles in order to compile this. Currently, I made a new folder at `~/librealsense/examples/capture01` and set the file here. Could you help me with this?

MartyG-RealSense, 2021-09-11:
The best way to learn how to build a project may be to work through a couple pf the example starter projects provided by the RealSense SDK's documentation.

I would recommend doing the **Getting Started** project first for creating a simple first application.

https://github.com/zivsha/librealsense/tree/getting_started_example/examples/getting-started

And then secondly you could look at the tutorial for creating a basic application with CMake:

https://github.com/IntelRealSense/librealsense/tree/master/examples/cmake

In regard to **stb_image_write.h**, in https://github.com/IntelRealSense/librealsense/issues/1485#issuecomment-378990445 a RealSense user provides information about where to find it and where to put it.

MartyG-RealSense, 2021-09-17:
Hi @vuhuulan  Do you require further assistance with this case, please?  Thanks!

vuhuulan, 2021-09-22:
Hi!
I'm sorry for the late reply. I was totally doing something else in the past days and haven't been continuing with this. I'm checking out those links you recommended, and will let you know if there is any question soon.

MartyG-RealSense, 2021-09-22:
No problem at all, @vuhuulan - thanks very much for the update!

vuhuulan, 2021-09-23:
I have succeeded compiling the file, however, I couldn't get saved image.
After compile there are two executable files `path/to/project/build/save_image` and `path/to/project/a.out; when executed the cursor just blink until I keyboard interrupted.

MartyG-RealSense, 2021-09-23:
I note your mention that "I have succeeded compiling the file".  If you are referring to rs-save-to-disk, a pre-built executable version of this example program should already have been placed in the **usr/local/bin** folder of Ubuntu when the librealsense SDK was installed, along with pre-built versions of other examples and tools.  Do you have this file in usr/local/bin, and are you able to run it if you do?

vuhuulan, 2021-09-26:
Ah, that was the program written in [#2219 (comment)](https://github.com/IntelRealSense/librealsense/issues/2219#issuecomment-412899468) (I called it save_image). About the pre-built program of rs-save-disk, I was able to run it, but it runs into same problem as save_image: a cursor blinking, and only stop when I keyboard interrupted; there is no files is saved.

MartyG-RealSense, 2021-09-27:
I was able to replicate the blinking cursor behavior in **rs-save-to-disk** if another RealSense program had already enabled the streams before launching rs-save-to-disk (for example, if I enabled depth and RGB in the RealSense Viewer and then launched rs-save-to-disk whilst the Viewer streaming was still running).  You confirmed earlier that you had no other RealSense programs running at the time of launching rs-save-to-disk though.

Could you try launching the pre-built version of rs-save-to-disk as the first thing that you do after the computer has booted please to make sure that the camera is not being accessed by anything else?

vuhuulan, 2021-09-28:
Yes, but this time it gives the same error as initial post: blinking for a while, then this message 
`RealSense error calling rs2_pipeline_start(pipe:0x55dac0d7f6c0):
    No device connected`

MartyG-RealSense, 2021-09-28:
Testing **rs-save-to-disk** again, it also produced the blinking cursor outcome if launched when no camera is attached.  So that could indicate that the problem is communication with the camera and not the camera being used by another program.  However, at the start of this case you stated that the camera performs normally with the rs-hello-realsense and rs-capture example programs, which would seem to show that in general the camera can be communicated with - just not with rs-save-to-disk specifically.

 If the camera is working normally other than this specific example then at this point it may be best to stop trying to get it working and instead use a different image-saving program if that function is what you require, such as the script at https://github.com/IntelRealSense/librealsense/issues/2219#issuecomment-412899468

vuhuulan, 2021-09-29:
I have compiled the script from the comment you suggested. When executed, it does nothing (or maybe something, but I can't see any results) and jump to a new command line.

MartyG-RealSense, 2021-09-29:
That particular script has a mechanism in an If statement that checks the camera model and does not proceed if the model is not a D415.  so you either need to edit the check to look for the D435i model or remove the If statement so that it does not check the camera model.

[image]

vuhuulan, 2021-09-30:
Hi, 
I have succeeded overcome the `if` statement (I used `D435i` instead of `D435I` :disappointed:).
After that, a new problem arose though:
```
RealSense error calling rs2_load_json(dev:0x55d8e37146e0, json_content:0x7ffd023c1050, content_size:0):
    parse error - unexpected end of input
sh: 1: pause: not found
```

I have asked for some helps from labmate, and it seems the problem lies in this paragraph:
```cpp 
             // Select the custom configuration file
             std::ifstream t(json_file_name);
             std::string preset_json((std::istreambuf_iterator<char>(t)), std::istreambuf_iterator<char>());
             advanced_mode_dev.load_json(preset_json);
             cfg.enable_device(serial);
```
and he specified that I don't have this `json_file_name` (which is `9.json` in the OG script), as well as `preset_json`. How should I fix this?

MartyG-RealSense, 2021-09-30:
It looks as though **9.json** is a custom camera configuration file created by the RealSense user who created the script.  You could remove from your script the lines that refer to the json loading that are highlighted below so that the camera uses its default configuration instead.

[image]

vuhuulan, 2021-09-30:
Thank you for the quick reply! 
I just did that, and got the following:
```
Starting pipeline for current camera 036522072607

Starting capture with the camera : 036522072607
terminate called after throwing an instance of 'rs2::invalid_value_error'
  what():  object doesn't support option #35
Aborted (core dumped)
```

MartyG-RealSense, 2021-09-30:
Next, please try removing **cfg.enable_device(serial);**

vuhuulan, 2021-09-30:
I did that, and have the same result.

MartyG-RealSense, 2021-09-30:
Could you paste your full C++ script in a comment please?

vuhuulan, 2021-10-03:
```cpp
#include <librealsense2/rs.hpp>
#include <librealsense2/rs_advanced_mode.hpp>
#include <iostream>
#include <fstream>
#include <map>
#include <thread>
#include <string>
#include <vector>

// 3rd party header for writing png files
#define STB_IMAGE_WRITE_IMPLEMENTATION
#include "stb_image_write.h"

using namespace rs2;
using namespace std;

//---------------------------------------------------------------------------------------------------------------------------------------------------------------------
// Capture & save function
//---------------------------------------------------------------------------------------------------------------------------------------------------------------------

void capture(rs2::pipeline p, string serial)
{
    cout << "\nStarting capture with the camera : " << serial << endl;
    frameset fs = p.wait_for_frames();
    for (auto j = 0; j < 30; ++j) fs = p.wait_for_frames(); // Wait a bit for auto exposure to settle
    rs2::pointcloud pc; 
    auto color = fs.get_color_frame();
    pc.map_to(color);
    std::stringstream png_file;
    png_file << "Image_" << serial << ".png";
    stbi_write_png(png_file.str().c_str(), color.get_width(), color.get_height(), color.get_bytes_per_pixel(), color.get_data(), color.get_stride_in_bytes());
    cout << "Saved " << png_file.str() << endl;
}

//---------------------------------------------------------------------------------------------------------------------------------------------------------------------
// This function will go through every connected device (as long as it is a D435I), apply a custom ".json", and then capture & save a ".png" for each device simultaneously.
//---------------------------------------------------------------------------------------------------------------------------------------------------------------------

int main() try 
{
    context ctx;
    vector<thread> threads;
    map<string, pipeline> pipes;

    for (auto dev : ctx.query_devices()) // For each device do :
    {   
        cout << dev.get_info(RS2_CAMERA_INFO_NAME);
        if (strcmp(dev.get_info(RS2_CAMERA_INFO_NAME), "Intel RealSense D435I") == 0) // Check for compatibility
        {
            pipeline p;
            config cfg;
            string serial = dev.get_info(RS2_CAMERA_INFO_SERIAL_NUMBER);
            //string json_file_name = "9.json";
            cout << "Configuring camera : " << serial << endl;

            // Add desired stream to configuration
            cfg.enable_stream(RS2_STREAM_COLOR, 1920, 1080, RS2_FORMAT_RGB8, 30);

            cout << "1" << endl;

            auto advanced_mode_dev = dev.as<rs400::advanced_mode>();

            // Check if advanced-mode is enabled to pass the custom config
            if (!advanced_mode_dev.is_enabled())
            {
                // If not, enable advanced-mode
                advanced_mode_dev.toggle_advanced_mode(true);
                cout << "Advanced mode enabled. " << endl;
            }

            cout << "2" << endl;

            // Select the custom configuration file
            //std::ifstream t(json_file_name);
            //std::string preset_json((std::istreambuf_iterator<char>(t)), std::istreambuf_iterator<char>());
            //advanced_mode_dev.load_json(preset_json);
            //cfg.enable_device(serial);

            cout << "3" << endl;

            // Start the pipeline
            p.start(cfg);
            cout << "Starting pipeline for current camera " << serial << endl;
            pipes[serial] = p;

        }
    }

    // Call the Capture & save function to start simultaneous threads
    for (map<string, pipeline>::iterator it = pipes.begin(); it != pipes.end(); ++it)
    {
        threads.push_back(std::thread(capture, it->second, it->first));
    }

    // Joining the threads
    for (auto& t : threads)
    {
        t.join();
    }

    return EXIT_SUCCESS;
}

catch (const rs2::error & e)
{
    std::cerr << "RealSense error calling " << e.get_failed_function() << "(" << e.get_failed_args() << "):\n    " << e.what() << std::endl;
    system("pause");
    return EXIT_FAILURE;
}

catch (const std::exception& e)
{
    std::cerr << e.what() << std::endl;
    system("pause");
    return EXIT_FAILURE;
}
```

MartyG-RealSense, 2021-10-03:
Thanks very much for the code.  I note that the json code is now commented out so that the program ignores it.

I compared the code of your script line by line with the original simultaneous capture script and the only significant difference that I could find was the path provided for **rs_advanced_mode.hpp**

`#include <C:/Program Files (x86)/Intel RealSense SDK 2.0/include/librealsense2/rs_advanced_mode.hpp>`

Since the json code has been commented out, we may not need to check for Advanced Mode (which json accesses).  So you could try commenting out the above include line for advanced_mode.hpp and also comment out the section that checks for whether Advanced Mode is supported, and see whether that makes the error disappear.

vuhuulan, 2021-10-04:
Um it still gives the same error.
I just tried some random things and noticed whenever **save_file** is executed, it disconnects the camera from the computer (sorry i don't know is it the correct way to say it). Proof is, after save_file, I open realsense-viewer or open **rs-hello-realsense** it shows no camera connected or shows the same cursor blink error respectively, and could be fixed by re-plugging the USB. I hope this helps somehow.

MartyG-RealSense, 2021-10-04:
You could test further for disconnection by adding the code below from the SDK's error handling documentation to your list of error catches.

```
catch (const rs2::camera_disconnected_error& e)
{
cerr << "Camera was disconnected! Please connect it back" << endl;
// wait for connect event
}
```

MartyG-RealSense, 2021-10-10:
I went through the case from the beginning.  The **rs-save-to-disk** program, when run unchanged, causes a blinking cursor which may be caused by the camera not being detected.  Other example programs such as **rs-capture** run fine though.

Your custom script is based on rs-save-to-disk and so you inherit the blinking cursor problem from it.  Additionally, with your added code in your custom script you get a **'terminate called after throwing an instance of 'rs2::invalid_value_error'** message.

It may therefore be the case that this error is occurring because the blinking cursor problem prevents an instruction that you added to the script from running correctly - for example, it needs to access the camera but the camera cannot be detected.

MartyG-RealSense, 2021-10-10:
If there is a problem with a librealsense project on Ubuntu that does not have a clearly identifiable cause then reinstalling librealsense is a good strategy, yes.  

If you installed librealsense from packages then you can uninstall all librealsense related packages by using the command below in the Ubuntu terminal.

`dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge`

If you installed librealsense from source code with CMake, you can go to the 'build' directory of the librealsense folder and run this command:

`sudo make uninstall && make clean`