realsenseai/librealsense #9701: How can l get a 16UC1  depth picture from D435i?

veryneo, 2021-08-30:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |       D435i                                                     | 
| Firmware Version                   |       2.48.0 | 
| Operating System & Version |      Win 10  | 
| Kernel Version (Linux Only)    |                                          | 
| Platform                                 |     PC  |
| SDK Version                            |        2.48.0                         | 
| Language                            |        C++                           | 
| Segment			|                       | 

### Issue Description

Hello, I get a trouble in the sample "imshow".
I use opencv to get the depth picture from the depth_frame, as the following pic show:

[image]

When l use format **8UC3**, no problem.
You can see the pic is ok.

[image]

The format i want is 16UC1, so l change the format form 8UC3 to 16UC1.
But  the picture doesn't show as well as before.

[image]

I search the issue ,but can not find the solution, can you give me a help?

MartyG-RealSense, 2021-08-30:
Hi @veryneo   A modified version of the **rs-imshow** example provided by a RealSense team member in https://github.com/IntelRealSense/librealsense/issues/2634#issuecomment-433904079 that uses CV_16U1 may be a useful reference for you.

https://github.com/IntelRealSense/librealsense/issues/2634#issuecomment-433904079

[image]

MartyG-RealSense, 2021-09-06:
Hi @veryneo  Do you require further assistance with this case, please? Thanks!

veryneo, 2021-09-06:
> Hi @veryneo Do you require further assistance with this case, please? Thanks!

I still cannot solve this problem.
So I use ros to get the depth data in my project.
You can close the issue, thank you!!!

MartyG-RealSense, 2021-09-06:
Thanks very much @veryneo for the update!  As suggested,  I will close the issue.  Thanks again!

closed 2021-09-06 by MartyG-RealSense