realsenseai/librealsense #9702: D435i not getting detected after firmware upgrade to 5.12.13.50

chiragandhi93, 2021-08-30:
| | |
|---|---|
|**librealsense**|2.49.0 RELEASE|
|**OS**|Linux|

Please provide a description of the problem

I have D435i which was working just fine till before firmware upgrade to 5.12.14.50. I updated firmware from 5.12.13.50 using realsense-viewer. I still can add the camera manually from realsense-viewer using "Add Source" button so the hardware is fine, but it is only adding RGB stream while the other streams are not visible. Even the firmware update utility "rs-fw-update" is not detecting the camera.
I even tried over Windows, but it says "USB not recognized"

MartyG-RealSense, 2021-08-30:
Hi @chiragandhi93   If you are able to add the camera in the Viewer with **Add Source**, does the 'More' option offer the **Update Firmware** option in its drop-down menu?

[image]

If it does, could you next please go to the firmware releases page for the 400 Series cameras at the link below and download the file **Signed_Image_UVC_5_12_13_50.bin**

https://dev.intelrealsense.com/docs/firmware-releases

Once the file is downloaded, use the **Update Firmware** option in the Viewer and select the firmware .bin file that you downloaded.  The firmware should then begin installing.

************

If this update procedure does not work or the Update Firmware option is not available in the 'More' drop-down menu, please try a **recovery** update in the rs-fw-update program instead using the .bin file that you downloaded.

**rs-fw-update -r -f Signed_Image_UVC_5_12_13_50.bin**

chiragandhi93, 2021-09-01:
Hi @MartyG-RealSense 

Sorry for the delayed response.
I have tried both the methods but with NO luck.

'More' drop-down menu does not show 'Update Firmware' option.
'rs-fw-update -r -f Signed_Image_UVC_5_12_13_50.bin' does not give any output on the terminal. but the firmware image does not get updated. Also 'rs-fw-update -l' does not list any devices.

MartyG-RealSense, 2021-09-01:
I will seek advice from Intel about your case.  Thanks very much for your patience.

MartyG-RealSense, 2021-09-01:
Hi again @chiragandhi93 - I have received feedback from Intel. 

1.  Is the problem still present if you reboot the computer?

2.  Do you have another computer available that you can test the camera on?

chiragandhi93, 2021-09-02:
Hi @MartyG-RealSense 
Thanks for prompt response.

1. Yes, The problem persists across reboots.

2. I tried on the other PC having Windows 10. The camera is not getting even on that. When I connect USB cable to PC, it shows a pop-up that "USB not recognized". Before firmware upgrade, it used to show new USB device detected and Windows would successfully recognize the D435i

MartyG-RealSense, 2021-09-02:
Intel have requested that you return your camera to the supplier that you purchased it from using the **RMA (Return Merchandise Authorization)** process of that supplier.

Please also email me the serial number of your camera.  If the camera cannot be detected, the serial number can be found on a sticker on the base of the camera.  You can send it to the address below:

martyx.grover@intel.com

MartyG-RealSense, 2021-09-05:
I can confirm that I have received your serial number and responded by email.  Thanks again!

closed 2021-10-01 by RealSenseCustomerSupport