realsenseai/librealsense #9705: Correctness of default intrinsic parameters

robopassio, 2021-08-30:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 01.05.08.01| 
| Operating System & Version |   ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  ubuntu 18.04                                        | 
| Platform                                 | PC|
| SDK Version                            |  latest  | 
| Language                            | python }                          | 
| Segment			| robot | 

### Issue Description

I tried to project 3D points to 2D image by using `rs.rs2_project_point_to_pixel`. However, this function is based on the intrinsic model. After rechecking, I found that my projection is mismatched between 3D and 2D scenes (yields different scales, the one in the 2D image is smaller than in 3D scene). Thus, I'm doubting the correctness of the factory intrinsic parameters. 
1. Should I absolutely trust that and check others? 
2. Do I need to recalibrate those parameters?
3. Is there a case of wrong factory intrinsic parameters?

MartyG-RealSense, 2021-08-30:
Hi @robopassio  I hope that the information in https://github.com/IntelRealSense/librealsense/issues/4315 will be useful in answering your questions.  In summary:

1.  As mentioned in https://github.com/IntelRealSense/librealsense/issues/4315 use of the rs2_project_point_to_pixel instruction may result in discrepancies. 

2. and 3.  Whilst you are free to perform a re-calibration at any time if you believe it to be necessary, a mis-calibration or an issue with the original factory calibration is not likely the cause of the mismatches that you experienced but rather the expected behavior of rs2_deproject_pixel_to_point.

robopassio, 2021-08-31:
Hi @MartyG-RealSense,
I think that thread is different from my purpose. Here, I'm trying to project 3D points to 2D image. I first define four points on an object (has rectangular shape) then project these four points into the image plane and draw it. I have checked and see that 
For the depth data: the point coordinates reading from the point PCL data is larger than the actual one.
For the color data: I use the mentioned function `rs.rs2_project_point_to_pixel` to project into the image frame and it is smaller than the actual one.
The object is located in the center view of the camera. The errors are around 4mm for each height and width. The depth error is around 15mm (measure from the surface of the camera to the object).  are that normal errors in this product?

MartyG-RealSense, 2021-08-31:
Advice in https://github.com/IntelRealSense/librealsense/issues/4644#issuecomment-522754917 that is related to the information in https://github.com/IntelRealSense/librealsense/issues/4315 states that if distortions are introduced by creating a 3D point cloud with deprojection, then measurable and visible artifacts will also be present when projecting that 3D point cloud back to 2D pixels.

In general, error in depth measurements increases linearly as distance of the observed object / surface from the camera increases.  This phenomenon is called **RMS error**.  The scale of the error can depend on the particular camera model being used.  For example, on the D435 / D435i the drift in accuracy may become noticable at 3 meters and beyond from the camera.  The D455 model though has 2x the depth accuracy over distance of the D435 / D435i, so it has the same accuracy at 6 meters from the camera that the D435 / D435i do at 3 meters.

More information about RMS error can be found in point 5 of the section of Intel's camera tuning guide linked to below.

https://dev.intelrealsense.com/docs/tuning-depth-cameras-for-best-performance#section-verify-performance-regularly-on-a-flat-wall-or-target

robopassio, 2021-08-31:
My measurement distance is within 20-50cm, however, the depth error from reading data and the physical measurement is about 1-1.5cm and width/height errors are 0.5cm, which is quite large. Any tutorial/guideline in detail to recalibrate/fix this issue? 

It looks that both RGB and depth data are incorrect and I cannot use the factory intrinsic parameters.

MartyG-RealSense, 2021-08-31:
My apologies, I remind myself that you have an L515 lidar depth camera instead of a 400 Series stereo depth camera.  Please disregard the information about RMS error then.

The L515 camera model does not have a depth and RGB calibration tool as it is a different camera technology to the 400 Series cameras.  I am also not aware of a factory reset mechanism for the L515 model.

It may be worth checking whether there is an environmental factor in the location that the camera is being used that is negatively affecting depth sensing results (such as infrared sources / sunlight), or a type of material that the L515 has difficulty with depth sensing, such as transparencies such as see-through plastic bottles.

robopassio, 2021-08-31:
@MartyG-RealSense . Thanks for the information. From this diagram, 
1. Is the origin of the RGB camera located at the front cover glass? or any offset?
2. Is the origin of the Depth camera located at the IR receiver?
[image]

MartyG-RealSense, 2021-08-31:
The top of page 12 of the L515 data sheet document (where the above image is sourced from) states that "the depth origin X-Y coordinates is the X-Y center of the IR Transmitter".

In https://github.com/IntelRealSense/librealsense/issues/7507#issuecomment-755768421 a RealSense team member provides information about the position of the RGB camera relative to the IR transmitter.

MartyG-RealSense, 2021-09-06:
Hi @robopassio  Do you require further assistance with this case, please? Thanks!

robopassio, 2021-09-06:
From the above comment, the answer for my #1 question is the origin camera is not located at the glass surface but is inside with 6.5 mm offset right?

MartyG-RealSense, 2021-09-06:
Pages 10 and 11 of the data sheet state that the '0' depth origin is at the front glass, and that the 'ground truth' of depth measurements can be found by adding **-4.5** mm to depth measurements.

[image]

[image]

robopassio, 2021-09-06:
Thanks. But my first question is about the origin position of the RGB sensor.

MartyG-RealSense, 2021-09-06:
In RealSense depth cameras, the 0,0,0 'world origin' of the coordinate system is the **depth** system (the IR Transmitter in the L515 camera model), not the RGB.  Positional differences between sensors are described with **extrinsics**, not with a separate RGB origin coordinate.

robopassio, 2021-09-07:
The extrinsic parameter of the RGB camera represents the rotation and translation from the origin of RGB to the object. After obtaining the extrinsic parameters (by calibrating), I realized there exists an offset along the z-axis of the RGB. I'm not sure about 
> not with a separate RGB origin coordinate 

Can you explain that? The extrinsic parameters cannot be used for RGB only?

MartyG-RealSense, 2021-09-07:
You can obtain the extrinsics for the **Color** stream from your L515 by using the SDK's **rs-enumerate-devices** tool by launching it in calibration information mode with the command below:

**rs-enumerate-devices -c**

Extrinsics describe a relationship between the positions of one type of sensor and another type of sensor.  The image below of L515 extrisics was obtained by using rs-enumerate-devices -c with an L515.

[image]

***********

You may also find it useful to look at an L515 script that makes use of **rs2_project_point_to_pixel** that is provided in https://github.com/IntelRealSense/librealsense/issues/8655

robopassio, 2021-09-13:
Thanks, that's clear.

closed 2021-09-13 by robopassio