realsenseai/librealsense #9708: About start RealSense.

Rachel1111, 2021-08-31:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |                        D400                               | 
| Firmware Version                   |                        2.48                                 | 
| Operating System & Version |                        Ubuntu 18                       | 
| Kernel Version (Linux Only)    |                        4.4.194                            | 
| Platform                                 |                        Arm 板                            |
| SDK Version                            |                       12.40                               | 
| Language                            |                           c++                                 | 
| Segment			|                                                                        | 

### Issue Description
Hi, a strange thing happened to me using RealSense. When I start RealSense by using pipe_.start(cfg_), it may cannot start successfully, and when I stop using pipe.stop(), it also raise a segmentfault. 
I really don't know why these happen, and hope you can help me to solve.

MartyG-RealSense, 2021-08-31:
Hi @Rachel1111   If you used the term **cfg_** in your pipe_.start(cfg_) instruction then the RealSense SDK would not be able to understand it unless you had named your custom configuration term as 'cfg_' in your **rs2::config** definition.

The term 'cfg' is more commonly used in scripts that use a custom configuration.  

Also, in your pipe start instruction the **pipe_.start(cfg_)** term would not have a '_' after pipe unless your had set up the **rs2::pipeline** definition with the name 'pipe_

The example script below shows a typical C++ setup of the pipeline.

```
rs2::config cfg;
cfg.enable_stream(RS2_STREAM_INFRARED, 1);
cfg.enable_stream(RS2_STREAM_INFRARED, 2);
rs2::pipeline pipe;
pipe.start(cfg);
```

Rachel1111, 2021-08-31:
Hi， @MartyG-RealSense I have these definition in my RealSense.h like:rs2::config  cfg_;  rs2::pipeline pipe_;

MartyG-RealSense, 2021-08-31:
What error message do you receive when the pipe start fails?  For example, do you receive the message **Runtime error: Couldn't resolve requests**

Rachel1111, 2021-09-01:
My code is as follow, and the RealSense cannot be started, it just cannot start RealSense stream.
[image]

Rachel1111, 2021-09-01:
And if I use RealSense-Viewer, It may have the issue " USB SCP overflow", and If I opened the RealSense, the issue "incompleted video frame detected " happened.
[image]

Rachel1111, 2021-09-01:
I used the same version of SDK in another Linux system(both Linux version are the same ), and that didn't happened these issues. So, I cannot understand why the same version would cause these issues?

Rachel1111, 2021-09-01:
And when I use pipe.stop(), it also released some issues. I had catched these issues.

> ========== signal 11 ============
./myprogram.out(_Z13signalHandleri+0x68) [0x558192c918]
linux-vdso.so.1(__kernel_rt_sigreturn+0) [0x7f8505f6c0]
librealsense2.so.2.48(+0x66ea5c) [0x7f807c1a5c]
librealsense2.so.2.48(+0x66ee30) [0x7f807c1e30]
librealsense2.so.2.48(_ZN12librealsense17composite_matcher4syncENS_12frame_holderERKNS_26syncronization_environmentE+0x1030) [0x7f807c7e60]
librealsense2.so.2.48(+0x66fbf8) [0x7f807c2bf8]
librealsense2.so.2.48(_ZN12librealsense7matcher4syncENS_12frame_holderERKNS_26syncronization_environmentE+0x64) [0x7f807c246c]
librealsense2.so.2.48(_ZN12librealsense16identity_matcher8dispatchENS_12frame_holderERKNS_26syncronization_environmentE+0x40) [0x7f807c1198]
librealsense2.so.2.48(_ZN12librealsense17composite_matcher8dispatchENS_12frame_holderERKNS_26syncronization_environmentE+0x9c) [0x7f807c5c4c]
librealsense2.so.2.48(_ZN12librealsense17composite_matcher8dispatchENS_12frame_holderERKNS_26syncronization_environmentE+0x9c) [0x7f807c5c4c]
librealsense2.so.2.48(+0x4ff718) [0x7f80652718]
librealsense2.so.2.48(+0x4ffe48) [0x7f80652e48]
librealsense2.so.2.48(_ZN12librealsense16processing_block6invokeENS_12frame_holderE+0x5c) [0x7f8064a91c]
librealsense2.so.2.48(+0x563f0c) [0x7f806b6f0c]
librealsense2.so.2.48(+0x650848) [0x7f807a3848]
librealsense2.so.2.48(_ZNK12librealsense12frame_source15invoke_callbackENS_12frame_holderE+0xd0) [0x7f807be950]
librealsense2.so.2.48(_ZN12librealsense16synthetic_source11frame_readyENS_12frame_holderE+0x3c) [0x7f80649df4]
librealsense2.so.2.48(rs2_synthetic_frame_ready+0x5c) [0x7f8076b904]
librealsense2.so.2.48(+0x4fb41c) [0x7f8064e41c]
librealsense2.so.2.48(_ZN12librealsense16processing_block6invokeENS_12frame_holderE+0x5c) [0x7f8064a91c]
librealsense2.so.2.48(+0x65aea8) [0x7f807adea8]
librealsense2.so.2.48(_ZNK12librealsense12frame_source15invoke_callbackENS_12frame_holderE+0xd0) [0x7f807be950]
librealsense2.so.2.48(+0x65237c) [0x7f807a537c]
librealsense2.so.2.48(+0x652bf8) [0x7f807a5bf8]
librealsense2.so.2.48(_ZN12librealsense8platform14v4l_uvc_device4pollEv+0xe98) [0x7f806f03f0]
librealsense2.so.2.48(_ZN12librealsense8platform14v4l_uvc_device12capture_loopEv+0x38) [0x7f806f0a60]
/usr/lib/aarch64-linux-gnu/libstdc++.so.6(+0xbbe94) [0x7f7d483e94]
/lib/aarch64-linux-gnu/libpthread.so.0(+0x7088) [0x7f7d563088]
======================

MartyG-RealSense, 2021-09-01:
I will link you to a list of resources about the **Incomplete video frame detected!** error so that you can see if any of the situations listed might apply to you.

https://github.com/IntelRealSense/realsense-ros/issues/1899#issuecomment-849837579

Rachel1111, 2021-09-06:
Hi, @MartyG-RealSense I also met another issue, when I used pipe.stop(), the whole program would hang all the time if I didn't terminate it. 
And I used gdb to print the possible issue like below:

> (gdb) backtrace
#0  0x0000007f7b312690 in __lll_lock_wait () from /lib/aarch64-linux-gnu/libpthread.so.0
#1  0x0000007f7b30b7c8 in pthread_mutex_lock () from /lib/aarch64-linux-gnu/libpthread.so.0
Backtrace stopped: previous frame identical to this frame (corrupt stack?)

> #0  0x0000007f7b312690 in __lll_lock_wait () from /lib/aarch64-linux-gnu/libpthread.so.0
#1  0x0000007f7b30b7c8 in pthread_mutex_lock () from /lib/aarch64-linux-gnu/libpthread.so.0

I didn't know why displayed these errors.

MartyG-RealSense, 2021-09-06:
Does the pipe.Stop() issue occur on the other Linux system that you said your program worked on in https://github.com/IntelRealSense/librealsense/issues/9708#issuecomment-909812306

Rachel1111, 2021-09-07:
No, It only happened in this Linux system.

MartyG-RealSense, 2021-09-07:
It's a difficult problem to diagnose.  I recall a case where a program didn't work on a RealSense user's computer but worked fine on their colleague's computer which was the same model.  It was found that the colleague had updated their computer to a newer configuration.  Once the RealSense user updated their computer to match the configuration of their colleague's computer that the program worked on then the problem was resolved.

Rachel1111, 2021-09-07:
thanks for your reply, @MartyG-RealSense would you please tell me what version of ubuntu your colleague use?

MartyG-RealSense, 2021-09-07:
The mentioned case didn't involve me, it was the work colleague of a RealSense user.  And the configuration that they used may not be relevant to your particular project.  I was making the point that a difference in configuration between computers of the same manufacturer / model may account for why a project works on one computer but not the other.

Rachel1111, 2021-09-08:
Hi @MartyG-RealSense I updated my ubuntu but that didn't help, the problem is still exiting.

MartyG-RealSense, 2021-09-08:
I have reviewed this case again carefully.  Recalling that you are using **pipe_.start(cfg)** in your script but state that you are stopping with **pipe.stop()**, then that could cause an error.  This is because if you defined the pipe with the name **pipe_** with an underslash then the same underslash would be used to stop the pipeline too with **pipe_.stop()**

Are you using **pipe_.stop()** with the underslash included, please?

Rachel1111, 2021-09-09:
Of course, I used pipe_.stop(), the same pipe_ as pipe_.start(cfg_)

MartyG-RealSense, 2021-09-09:
Are you still experiencing **Incomplete video frame detected!** in the RealSense Viewer, as mentioned in https://github.com/IntelRealSense/librealsense/issues/9708#issuecomment-909784249 and https://github.com/IntelRealSense/librealsense/issues/9708#issuecomment-910233867 of this discussion?  If so, that may be an indicator of a problem somewhere in your SDK installation on the affected computer that could be affecting the running of your correct code.

Rachel1111, 2021-09-13:
Yeah, I am still experiencing this issue.

MartyG-RealSense, 2021-09-13:
A list of possible causes of **Incomplete video frame detected!** was given in https://github.com/IntelRealSense/realsense-ros/issues/1899#issuecomment-849837579 - have you been through this list to see if any of the listed possibilities might apply to you on the computer that has this problem, please?

Rachel1111, 2021-09-16:
Actually, these listed possibilities don't apply to me on the computer that has this problem.

MartyG-RealSense, 2021-09-16:
The problem occurs on one Linux computer but not another.  You updated the computer that has the problem to match the configuration of the computer that the program works on, but it did not work.  If it works on one machine then the code is likely okay.  The main possibility that remains is a hardware problem with the computer itself, such as a glitch in its USB system (which could account for the **USB SCP Overflow** and the **Incomplete video frame detected!** error messages experienced in the RealSense Viewer.

closed 2021-09-28 by MartyG-RealSense