realsenseai/librealsense #9709: Build will not complete on macbook m1

gschian0, 2021-08-31:
On M1 Macbook pro when attempting to build librealsense for mac I get this error... 

`The system is: Darwin - 20.6.0 - arm64
Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" succeeded.
Compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ 
Build flags: 
Id flags: -c 

The output was:
0

Compilation of the CXX compiler identification source "CMakeCXXCompilerId.cpp" produced "CMakeCXXCompilerId.o"

The CXX compiler identification is AppleClang, found in "/Users/gennaroschiano/Documents/GitHub/librealsense/build/CMakeFiles/3.21.2/CompilerIdCXX/CMakeCXXCompilerId.o"

Compiling the C compiler identification source file "CMakeCCompilerId.c" succeeded.
Compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc 
Build flags: 
Id flags: -c 

The output was:
0

Compilation of the C compiler identification source "CMakeCCompilerId.c" produced "CMakeCCompilerId.o"

The C compiler identification is AppleClang, found in "/Users/gennaroschiano/Documents/GitHub/librealsense/build/CMakeFiles/3.21.2/CompilerIdC/CMakeCCompilerId.o"

Detecting CXX compiler ABI info compiled with the following output:
Change Dir: /Users/gennaroschiano/Documents/GitHub/librealsense/build/CMakeFiles/CMakeTmp

Run Build Command(s):/usr/bin/make -f Makefile cmTC_fb786/fast && /Applications/Xcode.app/Contents/Developer/usr/bin/make  -f CMakeFiles/cmTC_fb786.dir/build.make CMakeFiles/cmTC_fb786.dir/build
Building CXX object CMakeFiles/cmTC_fb786.dir/CMakeCXXCompilerABI.cpp.o
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++   -arch arm64 -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX11.3.sdk   -v -Wl,-v -MD -MT CMakeFiles/cmTC_fb786.dir/CMakeCXXCompilerABI.cpp.o -MF CMakeFiles/cmTC_fb786.dir/CMakeCXXCompilerABI.cpp.o.d -o CMakeFiles/cmTC_fb786.dir/CMakeCXXCompilerABI.cpp.o -c /opt/homebrew/Cellar/cmake/3.21.2/share/cmake/Modules/CMakeCXXCompilerABI.cpp
Apple clang version 12.0.5 (clang-1205.0.22.11)
Target: aarch64-apple-darwin20.6.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
clang: warning: -Wl,-v: 'linker' input unused [-Wunused-command-line-argument]
 "/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang" -cc1 -triple arm64-apple-macosx11.0.0 -Wdeprecated-objc-isa-usage -Werror=deprecated-objc-isa-usage -Werror=implicit-function-declaration -emit-obj -mrelax-all -disable-free -disable-llvm-verifier -discard-value-names -main-file-name CMakeCXXCompilerABI.cpp -mrelocation-model pic -pic-level 2 -mframe-pointer=non-leaf -fno-strict-return -fno-rounding-math -munwind-tables -target-sdk-version=11.3 -fvisibility-inlines-hidden-static-local-var -target-cpu apple-a12 -target-feature +v8.3a -target-feature +fp-armv8 -target-feature +neon -target-feature +crc -target-feature +crypto -target-feature +fullfp16 -target-feature +ras -target-feature +lse -target-feature +rdm -target-feature +rcpc -target-feature +zcm -target-feature +zcz -target-feature +sha2 -target-feature +aes -target-abi darwinpcs -fallow-half-arguments-and-returns -debugger-tuning=lldb -target-linker-version 650.9 -v -resource-dir /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/clang/12.0.5 -dependency-file CMakeFiles/cmTC_fb786.dir/CMakeCXXCompilerABI.cpp.o.d -skip-unused-modulemap-deps -MT CMakeFiles/cmTC_fb786.dir/CMakeCXXCompilerABI.cpp.o -sys-header-deps -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX11.3.sdk -stdlib=libc++ -internal-isystem /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX11.3.sdk/usr/include/c++/v1 -internal-isystem /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX11.3.sdk/usr/local/include -internal-isystem /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/clang/12.0.5/include -internal-externc-isystem /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX11.3.sdk/usr/include -internal-externc-isystem /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include -Wno-reorder-init-list -Wno-implicit-int-float-conversion -Wno-c99-designator -Wno-final-dtor-non-final-class -Wno-extra-semi-stmt -Wno-misleading-indentation -Wno-quoted-include-in-framework-header -Wno-implicit-fallthrough -Wno-enum-enum-conversion -Wno-enum-float-conversion -Wno-elaborated-enum-base -fdeprecated-macro -fdebug-compilation-dir /Users/gennaroschiano/Documents/GitHub/librealsense/build/CMakeFiles/CMakeTmp -ferror-limit 19 -stack-protector 1 -fstack-check -mdarwin-stkchk-strong-link -fblocks -fencode-extended-block-signature -fregister-global-dtors-with-atexit -fgnuc-version=4.2.1 -fcxx-exceptions -fexceptions -fmax-type-align=16 -fcommon -clang-vendor-feature=+disableNonDependentMemberExprInCurrentInstantiation -fno-odr-hash-protocols -mllvm -disable-aligned-alloc-awareness=1 -o CMakeFiles/cmTC_fb786.dir/CMakeCXXCompilerABI.cpp.o -x c++ /opt/homebrew/Cellar/cmake/3.21.2/share/cmake/Modules/CMakeCXXCompilerABI.cpp
clang -cc1 version 12.0.5 (clang-1205.0.22.11) default target arm64-apple-darwin20.6.0
ignoring nonexistent directory "/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX11.3.sdk/usr/local/include"
ignoring nonexistent directory "/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX11.3.sdk/Library/Frameworks"
#include "..." search starts here:
#include <...> search starts here:
 /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX11.3.sdk/usr/include/c++/v1
 /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/clang/12.0.5/include
 /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX11.3.sdk/usr/include
 /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include
 /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX11.3.sdk/System/Library/Frameworks (framework directory)
End of search list.
Linking CXX executable cmTC_fb786
/opt/homebrew/Cellar/cmake/3.21.2/bin/cmake -E cmake_link_script CMakeFiles/cmTC_fb786.dir/link.txt --verbose=1
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++  -arch arm64 -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX11.3.sdk -Wl,-search_paths_first -Wl,-headerpad_max_install_names  -v -Wl,-v CMakeFiles/cmTC_fb786.dir/CMakeCXXCompilerABI.cpp.o -o cmTC_fb786 
Apple clang version 12.0.5 (clang-1205.0.22.11)
Target: aarch64-apple-darwin20.6.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
 "/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ld" -demangle -lto_library /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/libLTO.dylib -dynamic -arch arm64 -platform_version macos 11.0.0 11.3 -syslibroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX11.3.sdk -o cmTC_fb786 -search_paths_first -headerpad_max_install_names -v CMakeFiles/cmTC_fb786.dir/CMakeCXXCompilerABI.cpp.o -lc++ -lSystem /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/clang/12.0.5/lib/darwin/libclang_rt.osx.a
@(#)PROGRAM:ld  PROJECT:ld64-650.9
BUILD 13:09:13 May 28 2021
configured to support archs: armv6 armv7 armv7s arm64 arm64e arm64_32 i386 x86_64 x86_64h armv6m armv7k armv7m armv7em
Library search paths:
	/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX11.3.sdk/usr/lib
Framework search paths:
	/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX11.3.sdk/System/Library/Frameworks/

Parsed CXX implicit include dir info from above output: rv=done
  found start of include info
  found start of implicit include info
    add: [/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX11.3.sdk/usr/include/c++/v1]
[... 474 lines cut ...]

[... cut ...]
/opt/homebrew/Cellar/cmake/3.21.2/bin/cmake -E cmake_link_script CMakeFiles/cmTC_c0085.dir/link.txt --verbose=1
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++  -arch arm64 -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX11.3.sdk -Wl,-search_paths_first -Wl,-headerpad_max_install_names CMakeFiles/cmTC_c0085.dir/src.cxx.o -o cmTC_c0085 

Source file was:
int main() { return 0; }
`

thanks for any help !

MartyG-RealSense, 2021-08-31:
Hi @gschian0  You could first try installing librealsense on Mac with the brew command **brew install librealsense** (though it has only been tried on older Intel-based Macs as far as I know at the time of writing this).

If this brew command does not work then it is possible to install librealsense on an M1 Mac but it can be complicated.  I have also not seen before the error **The output was: 0** during an M1 installation case.  could you describe please the installation steps that you took when building on your M1 Mac?  Thanks!

gschian0, 2021-08-31:
I can't find the librealsense formula on brew... it keeps returning an error for me as I try to install and suggest trying building from source.  I try that and get a similar error when I try to build from source based off the OSX instructions in this repository.  I've tried a couple things as well like setting the python executable manually seeing that was the issue for other people.  I get a shorter error now though ... 
`CMake Error at /opt/homebrew/Cellar/cmake/3.21.2/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find Threads (missing: Threads_FOUND)
Call Stack (most recent call first):
  /opt/homebrew/Cellar/cmake/3.21.2/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:594 (_FPHSA_FAILURE_MESSAGE)
  /opt/homebrew/Cellar/cmake/3.21.2/share/cmake/Modules/FindThreads.cmake:238 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  tools/convert/CMakeLists.txt:10 (find_package)`

MartyG-RealSense, 2021-08-31:
There is a brew formula entry in the link below for librealsense

https://formulae.brew.sh/formula/librealsense

The main forum help location for problems with installing librealsense on Mac M1 is at https://github.com/IntelRealSense/librealsense/issues/8090 and there are further reading links at the bottom of that discussion.  

For example, a RealSense user with an M1 Mac in https://github.com/IntelRealSense/librealsense/issues/8250#issuecomment-768309524 also experienced the **Could NOT find Threads (missing: Threads_FOUND)** error and shared the method that they used to resolve it.

gschian0, 2021-08-31:
when I run that command I get this ... "`Error: librealsense: no bottle available!
You can try to install from source with:
  brew install --build-from-source librealsense
Please note building from source is unsupported. You will encounter build
failures with some formulae. If you experience any issues please create pull
requests instead of asking for help on Homebrew's GitHub, Twitter or any other
official channels.`

gschian0, 2021-08-31:
I've attempted to add this to my CMakeLists.txt 
`set(CMAKE_THREAD_LIBS_INIT "-lpthread")
set(CMAKE_HAVE_THREADS_LIBRARY 1)
set(CMAKE_USE_WIN32_THREADS_INIT 0)
set(CMAKE_USE_PTHREADS_INIT 1)
set(THREADS_PREFER_PTHREAD_FLAG ON)`

the project was able to build with this command 

`cmake .. -DBUILD_EXAMPLES=false -DBUILD_GRAPHICAL_EXAMPLES=false -DBUILD_WITH_OPENMP=false -DHWM_OVER_XU=false -DBUILD_PYTHON_BINDINGS=false -G Unix\ Makefiles -DPYTHON_EXECUTABLE=/usr/bin/python3`

and I'm using this command to run the install... `make -j2`

at 50% build I get this error ... 
`[ 55%] Building CXX object CMakeFiles/realsense2.dir/src/proc/depth-formats-converter.cpp.o
In file included from /Users/gennaroschiano/Documents/GitHub/librealsense/src/proc/color-formats-converter.cpp:31:
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/clang/12.0.5/include/cpuid.h:11:2: error: this header is for x86 only
#error this header is for x86 only
 ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/clang/12.0.5/include/cpuid.h:280:5: error: invalid output constraint '=a' in asm
    __cpuid(__leaf, __eax, __ebx, __ecx, __edx);
    ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/clang/12.0.5/include/cpuid.h:245:11: note: expanded from macro '__cpuid'
        : "=a"(__eax), "=r" (__ebx), "=c"(__ecx), "=d"(__edx) \
          ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/clang/12.0.5/include/cpuid.h:295:5: error: invalid output constraint '=a' in asm
    __cpuid(__leaf, *__eax, *__ebx, *__ecx, *__edx);
    ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/clang/12.0.5/include/cpuid.h:245:11: note: expanded from macro '__cpuid'
        : "=a"(__eax), "=r" (__ebx), "=c"(__ecx), "=d"(__edx) \
          ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/clang/12.0.5/include/cpuid.h:309:5: error: invalid output constraint '=a' in asm
    __cpuid_count(__leaf, __subleaf, *__eax, *__ebx, *__ecx, *__edx);
    ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/clang/12.0.5/include/cpuid.h:252:11: note: expanded from macro '__cpuid_count'
        : "=a"(__eax), "=r" (__ebx), "=c"(__ecx), "=d"(__edx) \
          ^
/Users/gennaroschiano/Documents/GitHub/librealsense/src/proc/color-formats-converter.cpp:33:5: error: invalid output constraint '=a' in asm
    __cpuid_count(info_type, 0, info[0], info[1], info[2], info[3]);
    ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/clang/12.0.5/include/cpuid.h:252:11: note: expanded from macro '__cpuid_count'
        : "=a"(__eax), "=r" (__ebx), "=c"(__ecx), "=d"(__edx) \
          ^
5 errors generated.
make[2]: *** [CMakeFiles/realsense2.dir/src/proc/color-formats-converter.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
clang: warning: -latomic: 'linker' input unused [-Wunused-command-line-argument]
clang: warning: argument unused during compilation: '-mfpu=neon' [-Wunused-command-line-argument]
clang: warning: argument unused during compilation: '-mfloat-abi=hard' [-Wunused-command-line-argument]
make[1]: *** [CMakeFiles/realsense2.dir/all] Error 2
make: *** [all] Error 2`

gschian0, 2021-08-31:
If I run the basic command described in in the guide on this repository I get this error ...
`  8%] Building CXX object CMakeFiles/realsense2.dir/src/proc/depth-decompress.cpp.o
clang: warning: -latomic: 'linker' input unused [-Wunused-command-line-argument]
clang: warning: argument unused during compilation: '-mfpu=neon' [-Wunused-command-line-argument]
clang: warning: argument unused during compilation: '-mfloat-abi=hard' [-Wunused-command-line-argument]
In file included from /Users/gennaroschiano/Documents/GitHub/librealsense/src/proc/color-formats-converter.cpp:31:
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/clang/12.0.5/include/cpuid.h:11:2: error: this header is for x86 only
#error this header is for x86 only
 ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/clang/12.0.5/include/cpuid.h:280:5: error: invalid output constraint '=a' in asm
    __cpuid(__leaf, __eax, __ebx, __ecx, __edx);
    ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/clang/12.0.5/include/cpuid.h:245:11: note: expanded from macro '__cpuid'
        : "=a"(__eax), "=r" (__ebx), "=c"(__ecx), "=d"(__edx) \
          ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/clang/12.0.5/include/cpuid.h:295:5: error: invalid output constraint '=a' in asm
    __cpuid(__leaf, *__eax, *__ebx, *__ecx, *__edx);
    ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/clang/12.0.5/include/cpuid.h:245:11: note: expanded from macro '__cpuid'
        : "=a"(__eax), "=r" (__ebx), "=c"(__ecx), "=d"(__edx) \
          ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/clang/12.0.5/include/cpuid.h:309:5: error: invalid output constraint '=a' in asm
    __cpuid_count(__leaf, __subleaf, *__eax, *__ebx, *__ecx, *__edx);
    ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/clang/12.0.5/include/cpuid.h:252:11: note: expanded from macro '__cpuid_count'
        : "=a"(__eax), "=r" (__ebx), "=c"(__ecx), "=d"(__edx) \
          ^
/Users/gennaroschiano/Documents/GitHub/librealsense/src/proc/color-formats-converter.cpp:33:5: error: invalid output constraint '=a' in asm
    __cpuid_count(info_type, 0, info[0], info[1], info[2], info[3]);
    ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/clang/12.0.5/include/cpuid.h:252:11: note: expanded from macro '__cpuid_count'
        : "=a"(__eax), "=r" (__ebx), "=c"(__ecx), "=d"(__edx) \
          ^
5 errors generated.
make[2]: *** [CMakeFiles/realsense2.dir/src/proc/color-formats-converter.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
clang: warning: -latomic: 'linker' input unused [-Wunused-command-line-argument]
clang: warning: argument unused during compilation: '-mfpu=neon' [-Wunused-command-line-argument]
clang: warning: argument unused during compilation: '-mfloat-abi=hard' [-Wunused-command-line-argument]
make[1]: *** [CMakeFiles/realsense2.dir/all] Error 2
make: *** [all] Error 2`

MartyG-RealSense, 2021-09-01:
As I do not have a Mac to test with, I recommend posting a link to this case on https://github.com/IntelRealSense/librealsense/issues/8090 to seek the advice of other Mac M1 users.  I do apologize.

MartyG-RealSense, 2021-09-07:
I'm pleased to see that you have made progress with your problem in the discussion at https://github.com/IntelRealSense/librealsense/issues/8090

MartyG-RealSense, 2021-09-18:
Hi @gschian0   Can you confirm whether your M1 Mac installation problem was solved, please?  Thanks!

gschian0, 2021-09-24:
Installation problem was solved,  stuck on node bindings at the moment.

MartyG-RealSense, 2021-09-24:
What binding are you attempting to install, please?

JoeFirmament, 2021-09-25:
[Check it ](https://github.com/JoeFirmament/RealSenseStuffForSiliconM1)

MartyG-RealSense, 2021-09-25:
Thanks so much @JoeFirmament for sharing your RealSense pyrealsense2 and OpenCV installation system for Apple Silicon M1 in the above link   :)

MartyG-RealSense, 2021-10-08:
Case closed due to no further comments received.

closed 2021-10-08 by MartyG-RealSense

gschian0, 2021-10-12:
sorry, was away from the computer for a bit, thanks again for the help.