realsenseai/librealsense #9712: Data is lost or incomplete during data collection.

nevynwong, 2021-09-01:
When collecting data, it happens that the data is not captured or not captured comprehensively, what is the reason? The camera temperature is too high caused? Or is it not reading the json file accurately? We are using c# to start the camera and record the video.

[image]

We set the parameters and saved them to json, but after resetting the camera hardware, reading the set json file doesn't seem to work, it still captures the background in.

[image]

MartyG-RealSense, 2021-09-01:
Hi @nevynwong   It looks as though you are using the RealSense Viewer tool.  It does not look normal to me though that the OpenVINO face detection box is overlaid on the depth stream as it should normally only be overlaid on the RGB image.

[image]

Do you experience the same effect if you start the Viewer's depth and RGB streams the normal way with its side-panel controls instead of using C# to start the camera?

nevynwong, 2021-09-01:
This is the result of starting the camera in c#, recording the video file, and playing it back in viewer. Depth information is not captured.

MartyG-RealSense, 2021-09-01:
It looks as though the image with background data but incorrect head rendering is being displayed in the built-in **3D Viewer** app of Windows.  This suggests to me that the file being rendered in the 3D Viewer app is a **ply**.

If it is a ply, could you describe to me please the process that you used to create the ply?

nevynwong, 2021-09-06:
I recorded the video in c# and played it back in the viewer, and used rs-convert.exe to extract the ply file, and found the data missing.
And I find that using c# to call the camera and record video sometimes does not match the set parameters. What is the reason?

MartyG-RealSense, 2021-09-06:
When using rs-convert to extract the ply, are you including the **-c** condition to inform the tool that you are only converting color frames (since depth was not captured in the bag file)?

[image]

nevynwong, 2021-09-07:
You may not understand me, but I'd like to restate my question. I was able to capture color and depth maps, but the depth maps were mostly missing. In the c# program, I set the parameters, but the result is sometimes like this. When I reset the hardware in the Viewer, the background is all included, and the background removal parameter set does not take effect. The second recording will be fine. What's the reason?

nevynwong, 2021-09-07:
rs-convert.exe decompression out of the file name is too long, including the time stamp, the previous decompression out of the name is relatively short, this how to modify?

MartyG-RealSense, 2021-09-07:
If you believe that performing a hardware reset in your C# application might improve your results like a reset in the Viewer does, https://github.com/IntelRealSense/librealsense/issues/2878#issue-389794690 contains a C# script for the SDK's **hardware_reset()** function.

Regarding rs-convert filenames: if you mean that including the timestamp in the filename makes them too long, a change of file naming from frame number to timestamp number took place from SDK **2.35.0** onwards, according to the SDK release notes.

[image]

The Pull Request page https://github.com/IntelRealSense/librealsense/pull/6308 shows the changes made in the converter's files to change from frame number to timestamp naming (old version in red, new version in green).  So you could maybe reverse these changes in the files to make the tool use frame timestamps in naming again like the pre-2.35.0 version.

nevynwong, 2021-09-13:
I still have a couple of problems. 
First, when I set the reset hardware, I can't call the json file and can't set other parameters.
Second, can I call viewer.exe directly in c#, and then set the parameters through the json file
Third, can I call viewer in c# and then set the parameters through the program

MartyG-RealSense, 2021-09-13:
1.  If you need to change the camera configuration (load a different json or change a cfg stream configuration option) then you need to stop the pipeline, apply the change and then start the pipeline again.

2.  My understanding is that in .NET in general an .exe can be launched with the **Process.Start()** instruction.  so conceivably the Viewer .exe could be launched from an application such as a C# WPF form in this way, though I do not know of a RealSense-specific example of a C# script for doing so.

https://docs.microsoft.com/en-us/dotnet/api/system.diagnostics.process.start?view=net-5.0

3.  If you mean changing the Viewer settings with C# instructions sent from your own script then accessing the camera's Advanced Mode settings more directly though **NativeMethods.cs** may work better.

https://support.intelrealsense.com/hc/en-us/community/posts/360033065534/comments/360007889814

closed 2021-09-24 by MartyG-RealSense