realsenseai/librealsense #9713: How to  recover colorized depth images in python

gachiemchiep, 2021-09-01:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 05.10.06.00 | 
| Operating System & Version | Ubuntu 16.04  | 
| Kernel Version (Linux Only)    |  4.4.0-210-generic                                       | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  Python                         | 
| Segment			|  others                           | 

### Issue Description

Helo @RealSenseCustomerSupport
How can we recover colorized depth images in python?

We follows the document at [depth-image-compression-by-colorization-for-intel-realsense-depth-cameras](https://dev.intelrealsense.com/docs/depth-image-compression-by-colorization-for-intel-realsense-depth-cameras?_ga=2.144484341.1344785264.1630475756-741361283.1627625618) and [this issue](https://github.com/IntelRealSense/librealsense/issues/7930), but we couldn't calculate correctly. 

```bash
# Few values of depth_image_colorizer 
# depth_image_colorizer[0][0:5]
array([[  0, 131, 255],
       [  0, 131, 255],
       [  0, 133, 255],
       [  0, 133, 255],
       [  0, 137, 255]], dtype=uint8)

# The corresponding depth values
# depth_image[0][0:5]
array([ 987,  987,  993,  993, 1000], dtype=uint16)

# Our calculated values
# depth_image_recover[0][0:5]
array([[0, 0, 0],
       [0, 0, 0],
       [0, 0, 0],
       [0, 0, 0],
       [0, 0, 0]], dtype=uint16)
```
The code we used to recover colorized depth images

```bash
def RGBtoD(r, g, b):
    if r + g + b < 255:
        return 0
    elif r >= g and r >= b:
        if g >= b:
            return g - b
        else:
            return (g - b) + 1529
    elif g >= r and g >= b:
        return b - r + 510
    elif b >= g and b >= r:
        return r - g + 1020
    
def ColorToD(frame, is_disparity=False):
    min_depth = 0.29
    max_depth = 10.0
    height = frame.shape[0]
    width = frame.shape[1]
    
    depth_frame = np.zeros(frame.shape, dtype=np.ushort)
    
    for i in range(height):
        for j in range(width):
            r, g, b = frame[i][j]
            hue_value = RGBtoD(r, g, b)
             
            z_value = min_depth + (div * hue_value) / 1529.0 + 0.5
            depth_frame[i][j] = z_value # into 
            
    return depth_frame
```
We used the C++ code as in the document but the result didn't make sense. For  [  0, 131, 255] input, the result should be 987 but the C++ code return 6

[image]

MartyG-RealSense, 2021-09-01:
Hi @gachiemchiep  Use of this technique to recover a colorized depth image may not be ideal because of the difficulties involved in recovering the image outside of the RealSense SDK and of converting the recovery logic to a programming language other than the C++ scripting provided in the paper.  As such, I am not aware of any references about converting the technique for Python other than the discussion at https://github.com/IntelRealSense/librealsense/issues/7930 that you quoted above.

There is a detailed Python discussion about image compression alternatives to the recovery method at https://github.com/IntelRealSense/librealsense/issues/8117

You could also look at recording a compressed bag file instead of image files by using the SDK's **Z16H** format in conjunction with the **enable_record_to_file** instruction.

https://github.com/IntelRealSense/librealsense/issues/9295#issuecomment-874045003

MartyG-RealSense, 2021-09-02:
The SDK's **software_device()** feature can be used to import a **PNG** image file and generate a synthetic depth frame from it using its **create_synthetic_texture** function, like in the C++ example program in the link below.  I am not aware of an ability to generate a depth frame from a **jpg** file with software_device() though.

https://github.com/IntelRealSense/librealsense/tree/master/examples/software-device

Would it be possible to adapt your project to save depth to file in PNG format instead?

Using software_device on Python does not currently work well in its default state, though a RealSense user provided a 'patch' in the link below that they created to try to fix some of the issues.

https://support.intelrealsense.com/hc/en-us/community/posts/1500000934242-exploring-equivalent-API-in-python-for-rs2-software-device-create-matcher-RS2-MATCHER-DLR-C-defined-in-c-

MartyG-RealSense, 2021-09-14:
Case closed due to no further comments received.

closed 2021-09-14 by MartyG-RealSense

n1ckfg, 2022-07-24:
@MartyG-RealSense Have you got a reverse example of your Python code, to encode depth values to color? We're making a system with a non-realtime Python encoder, and a realtime GLSL decoder.

MartyG-RealSense, 2022-07-24:
Hi @n1ckfg  As you mention a 'reverse' example of encoding depth to color, are you aiming to convert a color image into RealSense depth data, please?  Thanks!

n1ckfg, 2022-07-24:
Ah, good point. I mean starting from a grayscale depth image, converting it into the RealSense color format, streaming it as an mp4, and decoding back to the original grayscale value on the other side.

MartyG-RealSense, 2022-07-24:
The main problem will be that once the depth data is converted then it will be very difficult to convert it back to accurate depth values on the other side due to loss of most of the depth information during the conversion.

It is certainly possible to live-stream RealSense data as a video feed using tools such as the RealSense GStreamer plugin at the link below.  

https://github.com/WKDSMRT/realsense-gstreamer

Once the original depth data has been converted though then you could likely not go back to those original values.

It may be worth considering using RealSense's **networking** system so that you can transfer the depth data from one computer with the camera attached to another and then access the data in real-time on the destination computer as though it was directly accessing the camera.

https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras

There is also an older RealSense networking tool called **EtherSense** that is for Python specifically.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python/examples/ethernet_client_server

https://dev.intelrealsense.com/docs/depth-camera-over-ethernet-whitepaper

AlexanderKhazatsky, 2022-12-08:
+1 For a formula for DtoRGB. Even without compression, the formula presented in the paper fails for very simple examples. There must be an issue somewhere.

AlexanderKhazatsky, 2022-12-08:
If anyone has written python code for the encoding and decoding scheme it would be extremely appreciated for you to share it :)

n1ckfg, 2022-12-08:
Yeah, you'll need to extract it from the other stuff but this is what I came up with:
https://github.com/n1ckfg/latk-video-001

AlexanderKhazatsky, 2022-12-08:
Agh!! You are a life savor, thanks so much!! What has your experience been like with this depth compression scheme?

AlexanderKhazatsky, 2022-12-08:
I'm considering this colorization scheme, [this one](https://reality.cs.ucl.ac.uk/projects/depth-streaming/depth-streaming.pdf), or some kind of lossless compression. However, I will be generating huge amounts of depth data and the last option is probably not possible.