realsenseai/librealsense #9717: Save a complete scan in to single PLY file or OBJ file?

chain-tech, 2021-09-01:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
I'm using Node JS wrapper and working on with exporting to PLY output.
Even if I put the capturing in a loop, looks like the final export_to_ply is writing only the last scanned frame in to the PLY file.

Other option I found is to keep saving several PLY files in the loop with different names.

Is there a way I can keep adding the frames in to the same PLY file until the camera scan is over?
Also, is there a way I can save the scans in OBJ format?

MartyG-RealSense, 2021-09-01:
Hi @chain-tech  The closest solution that I can think of to what you have in mind is the OpenCV RealSense application **rs-kinfu**.  A point cloud is built up in real-time, fusing new frames as the camera is moved around, and then the image can be exported to ply using export_to_ply.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/opencv/kinfu

OpenCV provides documentation about how to use OpenCV in a Node.js application.

https://docs.opencv.org/4.5.2/dc/de6/tutorial_js_nodejs.html

I did not find a good Node.js solution for .obj file export though.

chain-tech, 2021-09-06:
Hi @MartyG-RealSense 

Thanks for your suggestions above.

Please suggest/clarify for below concerns:

1. About **rs-kinfu**
I have tried using the rs-kinfu, was able to launch after several attempts and successfully generated PLY files. However, I've noticed that even this feature resets saving one ply file and creates an another. So for a duration of 2 to 3 minutes of scan, it creates several PLY files. I was able to achieve this currently in a loop using the original Node js wrapper as well.
But we're looking for a better option where we could just create one single PLY file for entire scan that can last for few minutes. Please suggest if there are any better options to achieve this.

2. About **rs-convert:**: I've stumbled across this rs-convert feature which claims of able to converting bag to several formats.
I've tried using it as below to convert to png/ ply formats:

**rs-convert -i myrecord.bag -p ./png_out/png 
rs-convert -i myrecord.bag -p ./png_out/
rs-convert -i myrecord.bag -p . (dot - for current directory)**

In all these attempts, the output is as below in the command prompt:
_100%
PNG converter
899 Depth frame(s) processed_

But in reality, there are no output files generated from any of the above commands, inspite of various trials regarding the output path and file name prefix opions. The same happens for -l option for PLY files. I've gone through several issues on rs-convert usage but nothing helped. Please confirm how to generate the converted output files properly. I wanted to give this an attempt to see if it satisfies my requirements as mentioned in point 1. 

Is rs-convert worth trying more per my needs? Is it capable of generating one big PLY out of a large BAG file? I am not able to confirm this as the ouput ply files are not created at my end and there are no errors. Log file shows empty as well.

MartyG-RealSense, 2021-09-06:
1.  I do not know of a similar method to rs-kinfu for progressively building up a point cloud.  You could though try increasing the **buffer size** in the rs-kinfu script from its default value of '50' to see whether this extends recording duration though.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/opencv/kinfu/rs-kinfu.cpp#L104

2.  For some RealSense users, rs-convert unfortunately does not output .ply files.  I have experienced this myself.  I would therefore not recommend using it any further for ply export if this mode is not working for you as there is not an available solution to get it working.

MartyG-RealSense, 2021-09-12:
Hi @chain-tech  Do you require further assistance with this case, please?  Thanks!

chain-tech, 2021-09-15:
Hi @MartyG-RealSense 

Thanks a lot for your help so far. Unfortunately we couldn't find or get it working to achieve what we are exactly looking for.
As of now exploring few other different options. So this can be closed for now.

closed 2021-09-15 by chain-tech

MartyG-RealSense, 2021-09-15:
Thanks very much for the update, @chain-tech - good luck with your work!