realsenseai/librealsense #9718: D455: failed to set power state on macOS

fdb, 2021-09-01:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | 05.12.15.50 | 
| Operating System & Version |    MacOS  | 
| Platform                                 | MacBook Pro 16"  |
| SDK Version                            |  2.49.0 (from Git)                          | 
| Language                            |  C++                          | 
| Segment			|  VR                          | 

### Issue Description

I can't open the device on macOS. Every tool/example I try gives some variation of "failed to set power state", e.g. `rs-enumerate-devices`:

```bash
$ ./rs-enumerate-devices 
 01/09 17:26:59,894 ERROR [0x700003c09000] (handle-libusb.h:95) failed to claim usb interface: 0, error: RS2_USB_STATUS_ACCESS
Could not create device - failed to set power state . Check SDK logs for details
 01/09 17:26:59,894 ERROR [0x11a0a4600] (sensor.cpp:537) acquire_power failed: failed to set power state
```

### Notes

- On Windows, using the 2.49.0 release, I can open the viewer and read the depth/rgb image without issue.
- On macOS I need to use a USB hub (Satechi) to connect the D455. I tried using this hub (instead of connecting directly) on Windows and that works.

[image]

MartyG-RealSense, 2021-09-01:
Hi @fdb  Up til this date, the most recent MacOS version that RealSense cameras have been successfully used with is **Big Sur**, the predecessor of Monterey Beta.  So there are no references regarding using the cameras with Monterey at the time of writing this. 

What was the method that you used to install the RealSense SDK on your Mac, please?  And is your Mac an **Apple Silicon M1** model or an older Intel-based model?

fdb, 2021-09-01:
Hi,

> What was the method that you used to install the RealSense SDK on your Mac, please? And is your Mac an Apple Silicon M1 model or an older Intel-based model?

I also have a MacBook Air with Big Sur with a M1 processor, but couldn't get the SDK to build there.

So I'm currently using a MacBook Pro 16" with an Intel processor. I compiled this from git using the instructions in the README. I had some issues with the Homebrew-installed openssl, and had to include ***@***.***/1.1.1k` with the cmake command.

I'll try downgrading to Big Sur and see if that helps.

MartyG-RealSense, 2021-09-01:
Thanks very much, @fdb - I look forward to your next update.

You may also like to explore the option of installing librealsense via brew formula on your Intel-based Mac with the command **brew install librealsense**

https://formulae.brew.sh/formula/librealsense

MartyG-RealSense, 2021-09-13:
Hi @fdb  Do you require further assistance with this case, please?  Thanks!

fdb, 2021-09-13:
Hi @MartyG-RealSense , I tried running this on my MacBook Pro, and with the openssl link it seems to work (I now notice the exact dependency has been censored by GitHub because they think it's an email address 😄).

MartyG-RealSense, 2021-09-13:
That's excellent news that you got it working on your MacBook, @fdb - thanks very much for the update!

closed 2021-09-24 by MartyG-RealSense