realsenseai/librealsense #9724: macOS SDK build instructions are not correct (OpenSSL)

fdb, 2021-09-02:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | Irrelevant (this is about building the SDK) | 
| Operating System & Version |   MacOS Big Sur 10.15.7  | 
| Platform                                 | PC |
| SDK Version                            |  2.49.0 (but building from Git)                          | 
| Language                            |  C++                          | 

### Issue Description

I'm trying to build the SDK on macOS by following the instructions to the letter. I just did a clean install of Big Sur, no other apps are installed on the machine (except for Google Chrome).

I followed the [macOS Installation procedure](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_osx.md). When I get to the point of running `cmake`, I get the following error:

```bash
fdb@fdbs-MacBook-Pro build % cmake .. -DBUILD_EXAMPLES=true -DBUILD_WITH_OPENMP=false -DHWM_OVER_XU=false

-- The CXX compiler identification is AppleClang 12.0.0.12000032
-- The C compiler identification is AppleClang 12.0.0.12000032
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Library/Developer/CommandLineTools/usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Checking internet connection...
-- Internet connection identified
-- Info: REALSENSE_VERSION_STRING=2.49.0
-- Performing Test COMPILER_SUPPORTS_CXX11
-- Performing Test COMPILER_SUPPORTS_CXX11 - Success
-- Performing Test COMPILER_SUPPORTS_CXX0X
-- Performing Test COMPILER_SUPPORTS_CXX0X - Success
-- Setting Unix configurations
-- Building libcurl enabled
CMake Error at /usr/local/Cellar/cmake/3.21.2/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find OpenSSL, try to set the path to OpenSSL root folder in the
  system variable OPENSSL_ROOT_DIR (missing: OPENSSL_INCLUDE_DIR)
Call Stack (most recent call first):
  /usr/local/Cellar/cmake/3.21.2/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:594 (_FPHSA_FAILURE_MESSAGE)
  /usr/local/Cellar/cmake/3.21.2/share/cmake/Modules/FindOpenSSL.cmake:574 (find_package_handle_standard_args)
  CMake/external_libcurl.cmake:56 (find_package)
  CMake/global_config.cmake:93 (include)
  CMakeLists.txt:44 (global_set_flags)
```

Here's the full procedure:
1. Start from a clean install
2. Install Homebrew
3. Install dependencies:
```
brew install cmake libusb pkg-config
brew install apenngrace/vulkan/vulkan-sdk --cask
```
4. Clone librealsense:
```
mkdir Source
cd Source
git clone https://github.com/IntelRealSense/librealsense.git
cd librealsense
```
5. Setup build directory:
```
mkdir build && cd build
sudo xcode-select --reset
```
6. Run cmake:
```
cmake .. -DBUILD_EXAMPLES=true -DBUILD_WITH_OPENMP=false -DHWM_OVER_XU=false
```

MartyG-RealSense, 2021-09-02:
Hi @fdb   A RealSense user on Mac in https://github.com/IntelRealSense/librealsense/issues/8090#issuecomment-835797389 experienced this OpenSSL error.  Another RealSense Mac user offered a solution in the following comment at https://github.com/IntelRealSense/librealsense/issues/8090#issuecomment-836892782

fdb, 2021-09-02:
@MartyG-RealSense I tried that, it doesn't work since `/opt/homebrew/opt/openssl` doesn't exist (it only works on [Apple Silicon](https://docs.brew.sh/FAQ#why-is-the-default-installation-prefix-opthomebrew-on-apple-silicon)).

I got a bit further, though. 

First I installed openssl as well:

```brew install openssl```

Then I set the OpenSSL root dir:

```
cmake .. -DBUILD_EXAMPLES=true -DBUILD_WITH_OPENMP=false -DHWM_OVER_XU=false -DOPENSSL_ROOT_DIR=/usr/local/opt/openssl
[configures without error]
make -j4
```

This still gives an error where the system indicates it can't find the OPENSSL_ROOT_DIR:
(I've shown output after the second run of make to avoid seeing all the irrelevant output):
```
fdb@fdbs-MacBook-Pro build % make -j4                                                                                                              

Consolidate compiler generated dependencies of target fw
[  0%] Performing configure step for 'libcurl'
[  1%] Built target fw
Consolidate compiler generated dependencies of target realsense-file
Consolidate compiler generated dependencies of target glfw
-- curl version=[7.75.0-DEV]
[  7%] Built target realsense-file
Consolidate compiler generated dependencies of target realsense2
[ 12%] Built target glfw
CMake Error at /usr/local/Cellar/cmake/3.21.2/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find OpenSSL, try to set the path to OpenSSL root folder in the
  system variable OPENSSL_ROOT_DIR (missing: OPENSSL_INCLUDE_DIR)
Call Stack (most recent call first):
  /usr/local/Cellar/cmake/3.21.2/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:594 (_FPHSA_FAILURE_MESSAGE)
  /usr/local/Cellar/cmake/3.21.2/share/cmake/Modules/FindOpenSSL.cmake:574 (find_package_handle_standard_args)
  CMakeLists.txt:387 (find_package)

-- Configuring incomplete, errors occurred!
See also "/Users/fdb/Source/librealsense/build/libcurl/src/libcurl-build/CMakeFiles/CMakeOutput.log".
See also "/Users/fdb/Source/librealsense/build/libcurl/src/libcurl-build/CMakeFiles/CMakeError.log".
make[2]: *** [libcurl/src/libcurl-stamp/libcurl-configure] Error 1
make[1]: *** [CMakeFiles/libcurl.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 49%] Built target realsense2
make: *** [all] Error 2
```

fdb, 2021-09-02:
Okay I tried [the mentioned fix](https://github.com/IntelRealSense/librealsense/issues/8090#issuecomment-836892782) by editing `CMake/external_libcurl.cmake` and adding the following on line 35:

```
                    -DANDROID_STL=${ANDROID_STL} ${CURL_FLAGS}
                    -DOPENSSL_ROOT_DIR=/usr/local/opt/openssl
```

That compiles libcurl.

Now I get an error while linking:

```
[100%] Linking CXX executable realsense-viewer
ld: cannot link directly with dylib/framework, your binary is not an allowed client of /usr/lib/libcrypto.dylib for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [tools/realsense-viewer/realsense-viewer] Error 1
```

MartyG-RealSense, 2021-09-02:
Whilst the link below that relates to this error is not a librealsense case, the solutions suggested in it may be useful.

https://github.com/ton-blockchain/ton/issues/119

fdb, 2021-09-02:
Seems like I need to update the external libcurl CMake file to use a specific version of libusb, and then it works.

For reference, the things I changed on top of the regular procedure (this is for an Intel Mac, Apple Silicon has different filepaths):

1. Install openssl:
```brew install openssl```

2. In `CMake/external_libcurl.cmake` add the following after line 35:

```
                    -DOPENSSL_ROOT_DIR=/usr/local/opt/openssl@1.1
```

Then run the build process as documented:

```
mkdir build && cd build
cmake .. -DBUILD_EXAMPLES=true -DBUILD_WITH_OPENMP=false -DHWM_OVER_XU=false
make -j4
```

@MartyG-RealSense thanks for all the help!

MartyG-RealSense, 2021-09-02:
You're very welcome, @fdb - great to hear that you were successful  :)

closed 2021-09-03 by fdb