realsenseai/librealsense #9726: error on Depth.Quality.Tool 2.49 and Intel.RealSense.Viewer 2.49 but works fine on version 2.40

ngassend, 2021-09-03:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435i                                                     | 
| Firmware Version                   | D400_Series_FW_5_12_15_50                 | 
| Operating System & Version | Windows 10                                           | 
| Kernel Version (Linux Only)    | NA                                                          | 
| Platform                                 | PC                                                           |
| SDK Version                            |  2.49                                                       | 
| Language                                |  labview                                                     | 
| Segment			         |               | 

### Issue Description
I have error when I start demos
Depth.Quality.Tool  "device is not functional or busy"
Intel.RealSense.Viewer  " Program terminated due to an unrecoverable SEH exception: Access Violation!"

I tried version 2.4 and it seems to works fine (like people with old and slow computer)

But in my case I think that the problem is not exactli the same !

my configuartion :
CPU : Intel(R) Core(TM) i9-10900K 
RAM :32,0 Go 

HAve you an idea and could I expect that is jus an issue in 2.49 version and that will be OK on future versions ?

MartyG-RealSense, 2021-09-03:
Hi @ngassend  I will direct you first to a recent discussion about this error at https://github.com/IntelRealSense/librealsense/issues/9672

If the advice in that discussion does not resolve the problem for you, please let me know and i will be pleased to continue helping you.

ngassend, 2021-09-03:
The problem is only on version 2.49 ! I tested 2.48 and it seems to works like 2.40 !

MartyG-RealSense, 2021-09-03:
Thank you for the information.  If you are installing the SDK with the automated installer file on the SDK **Releases** page, you could either run the 2.49.0 installer again (which would overwrite the previous installation) or use 2.48.0 for now and wait for the next SDK version.

closed 2021-09-03 by ngassend