realsenseai/librealsense #9728: Jetson Nano Installation and General Issue

pratikhrohane, 2021-09-04:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |   D455                                                     |
| Firmware Version                   |  2.49                                                        |
| Operating System & Version |   Ubuntu 18 .04                                       |
| Kernel Version (Linux Only)    |  4.9.140-tegra                                         | 
| Platform                                 |  Jetson nano                                           |
| SDK Version                           |                                                                | 
| Language                               |  Python3,  ROS                                       | 
| Segment			        |  Robot                                                     | 

### Issue Description

- When I open the realsense-viewer, I got some error and warning in terminal 

[image]

- When I launch the launch file `roslaunch realsense2_camera rs_camera.launch`, getting error even I connected with USB3 port

[image]

- I ignored all the errors and warnings and moving towards RTABMAP

Using the below two commands
1.  `roslaunch realsense2_camera rs_camera.launch align_depth:=true`
2.  ```roslaunch rtabmap_ros rtabmap.launch \
    rtabmap_args:="--delete_db_on_start" \
    depth_topic:=/camera/aligned_depth_to_color/image_raw \
    rgb_topic:=/camera/color/image_raw \
    camera_info_topic:=/camera/color/camera_info \
    approx_sync:=false```

I am unable to get the 3D view in RTABMAP window
[image]

But I am getting something like this on RVIZ
[image]

- last issue, when I try to open the ROS Network or VNC on my laptop while connecting the D455 camera with jetson, I am unable to get the connection with that, (getting time out), as soon I disconnect the camera connection establish with ROS network and also with VNC. 

with the solution for these issues I also wanted to know, intel RealSense D455 camera minimal hardware configuration requirement, for my project( camera will use this to create 3D map of the environment and Visual SLAM)

Thank You :)

MartyG-RealSense, 2021-09-04:
Hi @pratikhrohane   As you are able to vizualize the camera data in RViz despite the **control_transfer returned** warnings, this suggests that your launch is performing okay.

It looks as though you are using mostly the same RTABMAP arg values as those in the RealSense ROS wrapper's tracking launch file **opensource_tracking.launch**.  That launch file takes the approach though of activating **rtabmap.launch** through an include instruction and **then** referencing the rtabmap args.

https://github.com/IntelRealSense/realsense-ros/blob/development/realsense2_camera/launch/opensource_tracking.launch#L17

Your Jetson Nano should be adequate for this project, as although the Nano is small, Jetsons are very capable boards.

In regard to very slow / failed communications, the recent discussions in the links below provide some suggestions for actions to try that include both ssh and non-ssh connections.

https://github.com/IntelRealSense/realsense-ros/issues/2054
https://github.com/IntelRealSense/realsense-ros/issues/2052

pratikhrohane, 2021-09-06:
I launch the file opensource_tracking.launch, it's working able to creat the map but unable to load few things like as below image show  ColorOccupancyGrid here
[image]

And getting warnings after around 35 to 40 minutes "Hot Surface Do not Touch" as shown on the image, Jetson nano heat up to much, 
After heating I am unable to create 3d map also, camera data is coming but nothing happen on RVIZ

MartyG-RealSense, 2021-09-06:
Are you able to get a color image if you use the **Add** button in the RViz side-panel to add the RealSense camera topic **/camera/color/image_raw**

pratikhrohane, 2021-09-07:
> /camera/color/image_raw

Yes If I change to this topic I am getting the color image data

MartyG-RealSense, 2021-09-07:
Thanks very much @pratikhrohane for the confirmation of receiving the color image.  Do you require further assistance, please?

pratikhrohane, 2021-09-07:
> Hot Surface Do not Touch

Yes, but this issue, after heating too much MapCloud doesn't work, it's showing the point cloud data on RVIZ but not creating Map

MartyG-RealSense, 2021-09-07:
Could you test please whether the overheating problem goes away if you use the command below in Ubuntu to set the Jetson's fan to run at full speed?

**sudo sh -c 'echo 255 > /sys/devices/pwm-fan/target_pwm'**

pratikhrohane, 2021-09-08:
But I do not have fan in my Jetson Nano.

MartyG-RealSense, 2021-09-08:
Ah, thank you.  Fans can be fitted to the Nano model of Jetson but they do not come fitted with one on that board model by default.  Thanks for the clarification.

Is the casing of the camera hot to the touch?  If the camera casing was hot after only a few minutes of operation then this could suggest a USB problem causing overheating, such as a bad USB cable or a glitch on the USB port of the computing device.

pratikhrohane, 2021-09-10:
Camera not heating much its too mild, But Jetson Nano heat too much. My cable also new which i get with camera. 

What if I go for High Performance Jetson, like TX2 or NX..

Did this camera need any system required if I working on Heavy project?

MartyG-RealSense, 2021-09-10:
The 400 Series cameras contain **Vision Processor D4** hardware that performs some of the camera data processing that a computer would otherwise have to do, enabling the cameras to work well on low-specification hardware.  There does come a point though where a strong computer specification is recommended if a project involves functions that place a processing burden on the CPU that cannot be offloaded to the Jetson's graphics GPU using the RealSense SDK's **CUDA** support.

The operations that the Nvidia GPU on the Jetson can handle instead of the GPU include depth-color alignment, color conversion and pointcloud generation.

Which method did you use to build the librealsense SDK and the RealSense ROS wrapper on your Jetson Nano, please?

closed 2021-09-22 by MartyG-RealSense