realsenseai/librealsense #9729: bag file fails to load if path contains German umlauts

lisa-rosenberg, 2021-09-04:
Hi I am getting an error when trying to open a bag file (both in Intel RealSense Viewer 2.48.0 and pyrealsense2) when it is inside a folder with a German umlaut (äöüß). When I rename the folder everything is fine again luckily. I wanted to report this nonetheless, because I think that at least the error message could be better? I almost deleted these files because I thought they were corrupt or something and that renaming idea was really only a hunch.

Failed to load file [...]. Reason: Io in
rs2_context_add_device(ctx:0000028DF8277DA0, file:C):
Failed to create ros reader: Error opening file:
[...]

MartyG-RealSense, 2021-09-04:
Hi @lisa-rosenberg   Thanks very much for your report.  I will highlight it to Intel to seek their advice.

MartyG-RealSense, 2021-09-05:
I received a response from Intel.  They do not plan to add the suggested feature to recognise umlaut in an error message but said that as the RealSense SDK is open-source you could modify the code yourself to include it.  Thanks again for the request!

lisa-rosenberg, 2021-09-05:
Hi Marty! Thanks for your and Intel's quick reply! Alright, no problem and thanks for the info about modifying the code myself.

You can close this issue. Have a nice day 🌻

MartyG-RealSense, 2021-09-05:
You are very welcome, @lisa-rosenberg  - as suggested, I will close the issue.  Thanks again!

closed 2021-09-05 by MartyG-RealSense