realsenseai/librealsense #9730: Compilation invalidating Docker layer cache

dgorissen, 2021-09-05:
Im debugging a weird docker caching issue which occurs when trying to build librealsense in a container.

Relevant `Dockerfile` section:

```
# ... many previous steps

RUN git clone --depth 1 --branch v2.49.0 https://github.com/IntelRealSense/librealsense.git && \
  cd librealsense && \
  mkdir build  && cd build && \
  cmake .. \
  -DBUILD_UNIT_TESTS=OFF \
  -DBUILD_EXAMPLES=OFF \
  -DCMAKE_BUILD_TYPE=Release \
  -DFORCE_RSUSB_BACKEND=ON \
  -DCMAKE_PREFIX_PATH=~/apriltag

RUN cd librealsense/build && \
  make -j2.  

#...many other steps...
```

The second RUN command will never get cached, even though it (and following commands) complete successfully. When I restart the build it always restarts from the second RUN, discarding the layer cache for it for some reason.

```
...
#28 [25/33] RUN git clone --depth 1 --branch v2.49.0 https://github.com/IntelRealSense/librealsense.git &&   cd librealsense &&   mkdir build  && cd build &&   cmake ..   -DBUILD_UNIT_TESTS=OFF   -DBUILD_EXAMPLES=OFF   -DCMAKE_BUILD_TYPE=Release   -DFORCE_RSUSB_BACKEND=ON   -DCMAKE_PREFIX_PATH=~/apriltag
#28 sha256:ebfb42ae7cd3c464d6abc58ef92432f7de136010fff58ae1d90024e2045d273b
#28 CACHED

#29 [26/33] RUN cd librealsense/build &&   make -j2
#29 sha256:3e8f407631684baeb4c059af3aa4f8ddcfa613600e554fff4270d4b66562e2d7.
#29 0.620 [  0%] Building C object common/fw/CMakeFiles/fw.dir/empty.c.o.  -> why not cached
...
```

My docker command (im cross compiling for arm / Raspberry Pi on OSX 10.15.7, Docker version `20.10.7`, build `f0df350`)

```
	docker buildx build -t dgorissen/clare \
	--build-arg vinodist=${vinodist} \
	--platform linux/arm/v7 \
	--progress=plain \
	--push .
```

Of course I can just let it re-run but it increases build times significantly. It doesnt feel like this should be related to realsense but I havent seen this issue anywhere else, with any other steps. Any thoughts?

MartyG-RealSense, 2021-09-05:
Hi @dgorissen   I have not personally seen a Docker case on the librealsense forums involving this issue.  I did research cache invalidation in general though.  A guide in the link below suggests that the cache may be invalidated and force the build if a **build argument** is included in the Docker instructions.

https://blog.no42.org/code/docker-build-cache-invalidation/

dgorissen, 2021-09-05:
Thanks @MartyG-RealSense, I had come across that two but then I would have expected other steps to show the same problem (I had one usage of ARG).

 However to be sure I removed all usage of ARG & ENV and I wasnt using ADD & COPY to begin with.
I tried re-ordering some steps, did a clean build, waited to completion, then immediately ran the build command again. All steps were cached except the same compilation step. Really odd. The only thing I can think of that is different for that step is that it takes a long time to complete. But that should not matter.

I'll try to find some time to see if I can make a minimal example.

dgorissen, 2021-09-06:
Tried one more thing. Removed any existing build cache, active buildx builders, and turned on buildkit (which was off). Then a fresh build + re-build works as expected.  Odd, so must have gotten into some weird state. Thanks for considering, closing issue.

closed 2021-09-06 by dgorissen

MartyG-RealSense, 2021-09-06:
Great to hear that you found a solution, @dgorissen  - thanks very much for the update!