realsenseai/librealsense #9734: Are camera intrinsics (ppx, ppy, fx, fy) stored in rosbag file? MATLAB

sessiveENGGstu, 2021-09-07:
| Camera Model                       | L515 | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win 10  | 
| Platform                                 | PC|
| SDK Version                            |  { legacy 2}                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                           
| Segment			        |  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description

Are camera intrinsics stored in any of the ros Topics? I've looked through a lot of them and couldnt find it...
How do I get camera intrinsic (ppx, ppy, fx, fy)  information if its not in the topics? 

[image]

sessiveENGGstu, 2021-09-07:
bSel2 = select(bagselect,'Topic','device_0/sensor_0/Depth_0/info/camera_info');
msgStructs2 = readMessages(bSel2,'DataFormat','struct');
intrinsics = msgStructs2{length(msgStructs2)}.K;

closed 2021-09-07 by sessiveENGGstu

MartyG-RealSense, 2021-09-07:
Hi @sessiveENGGstu  I was just looking at your question - it's great to hear that you found a solution.  Thanks very much for sharing your code with the RealSense community!