realsenseai/librealsense #9735: the repository cannot be signed

outbreak-sen, 2021-09-07:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

MartyG-RealSense, 2021-09-07:
Hi @outbreak-sen  Intel changed their URL addresses from http to the more secure https in March 2021.  Intel updated their official installation documentation pages, but non-official instructions or installation scripts that still use the old http addresses may fail.

https://github.com/IntelRealSense/librealsense/issues/9349#issuecomment-882042628

outbreak-sen, 2021-09-07:
> Hi @outbreak-sen Intel changed their URL addresses from http to the more secure https in March 2021. Intel updated their official installation documentation pages, but non-official instructions or installation scripts that still use the old http addresses may fail.
> 
> [#9349 (comment)](https://github.com/IntelRealSense/librealsense/issues/9349#issuecomment-882042628)

sorry ，I havn‘t got your point. I just follow the github 's instruction. And I wonder if you could give me some detail instruction about my problem . thanks

outbreak-sen, 2021-09-07:
And when i run "sudo add-apt-repository "deb https://librealsense.intel.com/Debian/apt-repo $(lsb_release -cs) main" -u
" ,the process showed that 
"Err:10 http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic InRelease
  403  Forbidden [IP: 52.218.1.154 80]
"
i think that this is the main reason ,and i have no idea what to do

MartyG-RealSense, 2021-09-07:
**vcpkg** is maintained by Microsoft and not Intel.  

https://github.com/IntelRealSense/librealsense#building-librealsense---using-vcpkg

Whilst some RealSense users prefer vcpkg to installing librealsense from packages or source code using Intel's official instruction pages, it is not an installation method that I would personally recommend if you have the option of installing librealsense by another method.  

I note though that you said after you had built the SDK successfully using vcpkg that "this can not be used".  Do you mean that your vcpkg installation did not work and that you then tried building librealsense from packages with the official Intel instructions for doing so on the **distribution_linux.md** page?

https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md

If you are using Intel's official package instructions then I will next refer you to the discussion in https://github.com/IntelRealSense/librealsense/issues/9105 about the errors that you experienced.

ttencate, 2021-09-08:
@outbreak-sen 

> Err:10 http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic InRelease
> 403 Forbidden [IP: 52.218.1.154 80]

The `realsense-hw-public.s3.amazonaws.com` is the **old** repository, and you shouldn't be using it anymore. Probably this was left over from a previous installation, even if you added the **new** `librealsense.intel.com` repository today.

You should look at the file `/etc/apt/sources.list` and all files in `/etc/apt/sources.list.d`, and remove any mentions of the old repository there (or comment them out by prefixing with `#`). Then run `sudo apt-get update` again and the error should be gone.

outbreak-sen, 2021-09-09:
@ttencate
my problem had been solved with your method. It's pretty strange to fix my trouble like this
thank you

closed 2021-09-09 by outbreak-sen

sub400, 2021-11-30:
Hello,

I'm running into a similar problem when trying  to install librealsense on Debian (tested Bullseye and Buster). I followed the instructions and successfully registered the servers public key.

But the next step fails:

```
sudo add-apt-repository "deb https://librealsense.intel.com/Debian/apt-repo $(lsb_release -cs) main" -u

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/apt/cache.py", line 570, in update
    res = self._cache.update(fetch_progress, slist,
apt_pkg.Error: E:Failed to fetch https://librealsense.intel.com/Debian/apt-repo/dists/bullseye/InRelease  403  Forbidden [IP: 99.86.241.53 443], E:The repository 'https://librealsense.intel.com/Debian/apt-repo bullseye InRelease' is not signed.
```

Any suggestions would be very helpful! Thanks in advance.

ttencate, 2021-12-01:
@sub400 That's a different issue. I don't think Debian is supported at all, only Ubuntu. You may need to switch distros or build from source.

sub400, 2021-12-01:
@ttencate Thanks for your reply, but I'm pretty sure it worked some time ago on Debian.

MartyG-RealSense, 2021-12-01:
Hi @sub400   You mention that you tested the package installation on 'Buster and Bullseye'.  Those are versions of Raspbian OS, suggesting that you are using a Raspberry Pi.  Can you confirm whether that is the case, please?  If you are using a Pi and Raspbian then the packages on the **distribution_linux_md** instruction page would not work as they are designed for x86/AMD64 processors (such as those in desktop and laptop computers), whilst Raspberry Pi has an Arm chip architecture.  The **Acrobotic** guide in the link below would be an appropriate method for Raspian OS on a Pi.

https://github.com/acrobotic/Ai_Demos_RPi/wiki/Raspberry-Pi-4-and-Intel-RealSense-D435

sub400, 2021-12-01:
Hi @MartyG-RealSense, no I'm not talking abauot Raspberry OS, we are using regular Debian on x86/AMD64 architecture.

MartyG-RealSense, 2021-12-01:
@sub400  I see, Raspian OS is based on Debian and has a Bullseye version, but there is also **Debian 11 Bullseye**, and you are attempting to install librealsense using the Debian packages on the **distribution_linux.md** page.  Thanks very much for the clarification.

I have seen Debian package builds based on the Xenial (16.04), Bionic (18.04) and Focal (20.04) branches of Ubuntu, but not Debian Buster or Bullseye.

sub400, 2021-12-01:
So I have built the libraries form source on Debian abd everithing is working now. The manual for building from source is not 100% accurate when doing this on Debian as some prerequisites are missing on a fresh Debian installation that might be present on Ubuntu.

Thanks a lot for your assistance.

MartyG-RealSense, 2021-12-01:
It's great to hear that you succeeded, @sub400 - thanks very much for the update and sharing your experience of a Debian build with the RealSense community!