realsenseai/librealsense #9736: Intel RealSense D415 SEH exception on Windows 10

priamai, 2021-09-07:
| | |
|---|---|
|**librealsense**|2.49.0 RELEASE|
|**OS**|Windows 10 64bit |

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
Program terminated due to an unrecoverable SEH exception:
Access Violation!
```

```
connected devices:
1) Name: Intel RealSense D415, serial number: xxxxxxxxxx, update serial number: xxxxxxxx, firmware version: 05.12.15.50, USB type: 3.2
```

priamai, 2021-09-07:
Interesting error message during firmware update:

```
PS D:\Users\pippo\Downloads> .\Intel.RealSense.FW.Update.exe -s xxxxxxxx -f .\Signed_Image_UVC_5_12_15_50.bin

search for device with serial number: xxxx

updating device:
Name: Intel RealSense D415, serial number: xxxx, update serial number: xxxx, firmware version: 05.11.01.100, USB type: 3.2
 07/09 16:35:43,557 ERROR [30100] (../types.h:310) hr returned: HResult 0x8007001f: "A device attached to the system is not functioning."
```

priamai, 2021-09-07:
[image]

MartyG-RealSense, 2021-09-07:
Hi @priamai  I will direct you first to a recent discussion about this error at https://github.com/IntelRealSense/librealsense/issues/9672

If the advice in that discussion does not resolve the problem for you, please let me know and I will be pleased to continue helping you.

priamai, 2021-09-07:
Hi @MartyG-RealSense I am doing exactly that just running the Intel.RealSense.Viewer.exe from the download folder after downloading from the github link.
I have also tried to run as admin but same error.

MartyG-RealSense, 2021-09-07:
Thanks very much for the information.  Are you able to launch the **rs-capture** example program in Windows using the instructions in https://github.com/IntelRealSense/librealsense/issues/8024#issuecomment-747630656

priamai, 2021-09-07:
Okay give me some time.

MartyG-RealSense, 2021-09-07:
No problem at all, @priamai

priamai, 2021-09-07:
What components should I install to have the rs-capture binary?

[image]

MartyG-RealSense, 2021-09-07:
Usually you do not need to change any options when using the SDK installer file.  You can just keep clicking **Next** until the installation has completed.

MartyG-RealSense, 2021-09-13:
Hi @priamai  Do you require further assistance with this case, please?  Thanks!

priamai, 2021-09-13:
Hello, yes I do, the installer does not let me choose the installation folder, I run a standard SSD+HDD setup on my Win10 box and I am unable to install the entire SDK on the SSD which is just for the OS.
This is why I was asking whether I could just choose a subset of those tools to minimize the space constraint.
Let me know if there is a solution.

MartyG-RealSense, 2021-09-13:
If you only need to use the RealSense Viewer tool, there is a standalone versipn of the Viewer that is around 38 mb and can be downloaded to a computer location of your choice, as it is just a single .exe file.  I just installed it on a Windows computer without the full RealSense SDK installed on it and confirmed with a D455 camera that it works.

The process to set it up is as follows.

1.  Go to the same **Assets** list on the SDK 'Releases' page that the full SDK installer file is obtained from.  Instead of downloading that though, right-click on the link in the file list titled **Intel.RealSense.Viewer.exe** and select the browser menu option to save a link.  It might be called Save Link As, or Save Target As, or something similar.

[image]

2.  Choose from the pop-up file selector box the drive location that you wish to download the **Intel.RealSense.exe** file to.

3.  Once download to the chosen location is completed, attach your RealSense camera to the computer if it is not attached already and then double-leftclick on **Intel.RealSense.exe** to launch the Viewer.  Your camera should then appear in the options side-panel of the Viewer.

You could likely perform the same procedure with the **Depth.Quality.Tool.exe** file that is also in the Assets list.

robomotic, 2021-09-13:
Yes I forgot to say I tried that already and it says the camera device has a problem. I will send a screenshot too later.

priamai, 2021-09-13:
Here we go, I can see the device

[image]

but when I click on it, I get the same error as in the other tool

[image]

Also since you mentioned an Intel D455 camera but as stated in my first post I have a D415 camera, could it be that the SDK and tools are not tested with the D415 version?

MartyG-RealSense, 2021-09-14:
The RealSense SDK is fully compatible with all 400 Series camera models, including the D415.

If the camera is being detected correctly then its details should be displayed beneath the Add Source button and not require clicking the button to find it.

[image]

Can you confirm please that the Windows camera privacy settings granting permission for access to the camera hardware and for applications to access the camera are both set to On, as described in the instructions in https://github.com/IntelRealSense/librealsense/issues/3381#issuecomment-503424789

priamai, 2021-09-14:
Great to know that!
Yes all options were already set to On as described in that link.
I have tried also just to be sure to disable and re-enable, restart Windows but still same issue.

MartyG-RealSense, 2021-09-14:
Has the camera worked at any time since you purchased it, or did it work for a time and then was no longer detectable?

Also, have you stored the camera unused in a cool dry place such as a box or a drawer for a time (e.g weeks or months) immediately before you started having this issue with the camera when using it?

priamai, 2021-09-14:
Hi there,
I ordered it from Mouser UK, it was dispatched next business day from a Texas warehouse.
I literally just opened it and plugged it.
The packaged looked in a good state.
I don't believe is faulty, I tested it on an Ubuntu and I was able to see the data frames.
It would be nice to work out why it doesn't work on Windows 10.
Cheers.

MartyG-RealSense, 2021-09-15:
Is your Windows 10 computer a laptop?  If so, I wonder whether a built-in webcam is causing a detection conflict.  If you do have a laptop with a built-in webcam, could you try temporarily disabling it in the Windows Device Manager settings using the instructions in https://github.com/IntelRealSense/librealsense/issues/9638#issuecomment-902120601

robomotic, 2021-09-15:
Oh yes that could be the case i do have a built in camera. Will report soon.

MartyG-RealSense, 2021-09-21:
Hi @priamai  / @robomotic   Do you have an update about this case that you can provide, please?  Thanks!

closed 2021-09-27 by MartyG-RealSense