realsenseai/librealsense #9737: How can I remove error depth information?

lemon5254, 2021-09-08:
[image]
Hi
How can I remove error depth information?
I think there is a reflective zone
Some time this zone get some depth information like this! But this data is error (depth information is bad)
[image]
Please help me~~

MartyG-RealSense, 2021-09-08:
Hi @lemon5254  Depth cameras in general (not just RealSense) have difficulty reading depth from black or dark grey objects because physics dictates that these colors absorb light.  So black / dark grey objects may appear to be rendered on the image but are actually empty areas without depth detail that are shaped like the object.

For example, scanning a black cable may result in a shape on the image that looks like a cable but is actually a cable-shaped area of empty detail with no texture.

Projecting a strong light source onto surfaces that have such colors can help to bring out depth detail.  This may be what is accounting for the variable results on your dark ceiling area if the lighting conditions are varying at different times.

You may be able to improve the depth image by applying a post-processing filter called a **hole-filling filter**.  You can do this in the RealSense Viewer tool by going to the **Post-Processing** section of the Viewer's options side-panel, expanding open the list of filters and enabling **Hole Filling Filter** by left-clicking on the red icon beside it (indicating an Off state) to turn it to blue (On).

[image]

MartyG-RealSense, 2021-09-15:
Hi @lemon5254  Do you require further assistance with this case, please?  Thanks!

lemon5254, 2021-09-15:
Hi @MartyG-RealSense thank your reply! 
I have same problem like this! **It still have error depth information !** 
[image]
And i had turned on hole filling filter like this!
[image]
 Can you help me to solve it ~?

MartyG-RealSense, 2021-09-15:
Does the incorrect strip of depth still occur if you change the **depth unit scale** of the camera from its default value of 0.001 (meters) to **0.0001**

You can adjust the scale in the RealSense Viewer by going to the **Depth Units** option under the 'Controls' section of the Viewer side-panel and then left-clicking on the pencil icon beside it to manually enter the value '0.0001' with the keyboard.

[image]

lemon5254, 2021-09-17:
Hi, I try to modify this parameter but i still got bad information. How can i improve it? thank your suggest !!
[image]
And I had modified depth units like the picture
[image]

MartyG-RealSense, 2021-09-17:
Have you got depth Auto-Exposure enabled, please?  If you have not then the **33000** exposure value in the above image could be considered high and may result in more holes, as the default exposure value for the 400 Series cameras is **8500**

lemon5254, 2021-09-22:
yes, I have turned on  Auto-Exposure enabled. Do I need to change to manual adjustment?

MartyG-RealSense, 2021-09-22:
If you have Auto-Exposure enabled then you can leave that enabled as it should cause the 33000 manual value to be ignored.

closed 2021-10-04 by MartyG-RealSense