realsenseai/librealsense #9738: restart pipe mGLSurfaceView becomes three areas

dota2mhxy, 2021-09-08:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435 | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win 10  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | Android. |
| SDK Version                           | 2                       | 
| Language                              |  java                            | 
| Segment			        |  Android | 

Hello, when I restart the pipe, the first few times were normal, but later mGLSurfaceView becomes three areas. One of them should be the previous image. I don't know what to do
this  is  nomal

[image]

this  is  three areas

[image]

this  is  my  code
[image]

[image]

[image]

MartyG-RealSense, 2021-09-08:
Hi @dota2mhxy   Could you provide more information about what you mean when you say there are three areas.  Do you mean that there should be two panels (the RGB and the depth) but then it becomes three panels (RGB and two depth).  Thanks!

dota2mhxy, 2021-09-08:
yes，It used to be two panels, color and depth. After I restarted many times, it became three, two depths and one color. One depth seems to be an image left over from the past, not the current image information

dota2mhxy, 2021-09-08:
@MartyG-RealSense

MartyG-RealSense, 2021-09-08:
I compared your code to the Capture example.  The code of that project and your script looks similar, with the main difference being that some of the code blocks that are inside functions that are jumped to when called are moved so that a jump is no longer necessary. 

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/android/examples/capture/src/main/java/com/intel/realsense/capture/MainActivity.java

This may introduce unforeseen errors though.  It may be better to use the Capture example's code and find a way to insert your own **mStart RecordFab** block of listener code into it.

Before doing that, you could try removing the index numbers from the cfg instructions.

**cfg.enableStream(StreamType.DEPTH, width: 1280, height: 720, StreamFormat.ANY, framerate: 6);
cfg.enableStream(StreamType.COLOR, width: 1280, height: 720, StreamFormat.ANY, framerate: 6);**

dota2mhxy, 2021-09-10:
hello @MartyG-RealSense  .removing the index numbers from the cfg instructions   Is not feasible。this  is code  from 
com.intel.realsense:librealsense:2.4.you can see  if you  not give  index，-1 is the default
[image]

Then, in this code, I actually made some modifications to the capture example, made some custom calculations for the functions of mstartrecordfab and mstopprecordfab buttons, and the displayed interface has always been mglsurfaceview. To prevent out of frames, I have to stop(), start()
The previous restarts were normal. About the third and fourth time, two depth maps appeared. I tried to move the camera and found that the color map and one depth map will change with the scenery, and the other depth map will not change. It may be a certain image. it seems to be the first copy of the depth map。You can see from the following two pictures that the top depth map is fixed, which is the copy of the  first depth map. The bottom depth map corresponds to the color map。
[image]

[image]

and  if  i  Unplug the USB interface, it will restart, and the new view will become normal

MartyG-RealSense, 2021-09-10:
If the problem only occurs after the third / fourth pipeline start and an unplug-replug of the USB cable corrects it, a solution might be to perform a camera on each start.  In https://github.com/IntelRealSense/librealsense/issues/4091#issuecomment-561935062 an Android script that makes use of reset is provided.  Left-click on the **Streamer.java.txt** link in the comment to open the script.

closed 2021-09-16 by dota2mhxy