realsenseai/librealsense #9739: L515 depth bin twice the size as expected && get real depth from depth matrix

tonyzhang2035, 2021-09-08:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 01.05.08.01 | 
| Operating System & Version | Linux (Ubuntu 16.04) | 
| Kernel Version (Linux Only)    |  4.15.0                                         | 
| Platform                                 | PC |
| SDK Version                            |  v2.49.0                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
I recorded a .bag file with L515 and used the [rs-convert](https://github.com/IntelRealSense/librealsense/tree/master/tools/convert) tool for data extraction. The depth visualizations look fine but the extracted .bin files (depth matrix) seems to have a strange size (1228800 Bytes or 614400 pixels). I'm using 640x480 and for uint16 format I should be expecting a file size of 640 x 480 x 2 = 614400 Bytes, whereas I'm getting twice the size. Why is this happening? Also how do I convert the depth data to the real, physical distance? I assume the formula is something like: depth = scale * data + offset? The metadata doesn't contain relevant info. 

Command used for data extraction:
`./rs-convert -i <intput.bag> -b <output_dir> -d`

Wanted to upload a sample .bin file but it's not supported. I'll still put its metadata here.
[_Depth_metadata_112353.81099999998696.txt](https://github.com/IntelRealSense/librealsense/files/7125833/_Depth_metadata_112353.81099999998696.txt)

MartyG-RealSense, 2021-09-08:
Hi @tonyzhang2035  In general I would not recommend placing faith in simple formulas, even if they seem logical (though they can be useful for providing an initial rough estimate to set your expectations).  

For example, in theory multiplying the number of frames recorded by the FPS speed may give an estimate for the number of frames that should have been created during a recording session, but in practice the actual value rarely matches the theoretical value because the process has more factors that can influence the outcome than the formula takes account of.

Real physical distance in meters is calculated by multipying the pixel depth value (uint16_t) by the **depth unit scale** of the particular RealSense camera model being used.  The camera's depth unit scale can be retrieved from the camera in real-time with program scripting.  But as the depth unit scale does not change unless deliberately altered, it can be easier to simply multiply uint16_t by the known default depth scale of the camera model.  For L515, that default scale value is **0.000250**

tonyzhang2035, 2021-09-08:
Hey @MartyG-RealSense thanks for explaining the distance calculation and I understand the formula is for theoretical calculation only and will be affected by many other factors. Is there any good calibration method that you would suggest?

And how about the file size mismatch issue? I'm getting twice the expected size (1228800bytes vs. 480x640x2=614400) from the .bin file extracted using rs-convert.
[image]

MartyG-RealSense, 2021-09-08:
The L515 lidar depth camera model does not have a calibration tool like the 400 Series depth cameras do, as its lidar technology is different from the stereo-based technology that the 400 Series models use.

I would say though that when recording camera data to file, other factors such as the stability of the USB system and processing performance of the computer that the camera is being used with will be a more significant influence on the results though.  If the speed of the drive being used to wrote data to is slow then that could also create a bottleneck during recording.

As far as I am aware the subject of the byte size of the .bin files obtained from rs-convert has never been raised before, so there is no reference about what a 'correct' file size of a .bin export from rs-convert should look like, unfortunately.

tonyzhang2035, 2021-09-08:
> As far as I am aware the subject of the byte size of the .bin files obtained from rs-convert has never been raised before, so there is no reference about what a 'correct' file size of a .bin export from rs-convert should look like, unfortunately.

It's not about referencing to a 'correct' file size and what matters here is to make the exported .bin usable, i.e. to convert 640x480x4 byte data into a 640x480 uint16_t depth map. Is the uint16_t data stored in a form of uint32_t, where only the last 16 bits contain the actual value? Or the .bin file contains more data along with the depth map? Some thing is still missing here.

Also the raw frame size given in metadata shows 614400 (640x480x2) whereas the depth output should be 640x480. I guess some extra processing is required to bridge this gap.

MartyG-RealSense, 2021-09-09:
There is very little information available unfortunately on reading a saved binary file into the SDK to turn it back into usable depth frame data.  The best reference on the subject is https://github.com/IntelRealSense/librealsense/issues/2462

tonyzhang2035, 2021-09-11:
I couldn't figure out how to process the raw file converted via rs-convet but got the desired depth map (640x480) following the reference provided. Thank you!

closed 2021-09-11 by tonyzhang2035

MartyG-RealSense, 2021-09-11:
That's great news that you were successful, @tonyzhang2035 - thanks very much for the update   :)