realsenseai/librealsense #9740: RGB-Sensor "No Frames Received"

YBachmann, 2021-09-08:
| | |
|---|---|
|**librealsense**|2.47.0 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense L515|
|**Serial Number**|f1150136|
|**Firmware Version**|01.05.08.01|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|L500|
|**Asic Serial Number**|0003b9f8cfc5|
|**Firmware Update Id**|0003b9f8cfc5|

Suddenly the L515 Camera is not able to deliver an RGB image. 
When enabling RGB Camera in the Intel RealSense Viewer I get a white image with a red text `No Frames Received!` in the center of it. 
When trying to get frames via Python I get `Frame didn't arrive within 5000`. 
It worked before and both the Intel RealSense Viewer and my Python program work with my other camera (d455). 

**Error messages:**
[image]
The message that is cut off in the screenshot says: 
`types.h:310 - get_camera_control()->GetRange(CameraControl_Exposure, &minVal, &maxVal, &steppingDelta, &defVal, &capsFlag) returned: HResult 0x80070490: "Element nicht gefunden."`

I tried reconnecting the camera and changing the USB cable. 
The camera is plugged directly into my laptop. 
I also tried using the RealSense Viewer on my Jetson Nano (Ubuntu 18.04.5 LTS) and got the following Error: 
`http-downloader.cpp:143 - Download error from URL: https://librealsense.intel.com/Releases/rs_versions_db.json, error info: Couldn't resolve host name`

The depth sensor still works fine.

MartyG-RealSense, 2021-09-08:
Hi @YBachmann   There was a past L515 case with similar RGB errors but depth working normally in https://github.com/IntelRealSense/librealsense/issues/8302 

Could you tell me the method that you used to install the RealSense SDK on Windows, please.  Did you use the automated installer file from the SDK's **Releases** page, or install it from source code using CMake?

YBachmann, 2021-09-08:
Hi @MartyG-RealSense 
I just downloaded the Intel.RealSense.Viewer.exe from the GitHub Release Page. 
I tried different versions (including the newest one) but the problem was always the same.

MartyG-RealSense, 2021-09-08:
Have you also tried installing the full RealSense SDK for Windows from the Assets list with the **Intel.RealSense.SDK-Win10** installer file and then launching the Viewer using the launch shortcut icon that the SDK installer places on the Windows desktop, please?

[image]

YBachmann, 2021-09-09:
I just did that but sadly the error still remains.

YBachmann, 2021-09-10:
I just get two black windows when the L515 is connected. 
I cross tested with my D455 connected and that worked.

MartyG-RealSense, 2021-09-10:
And the same problem occurs if you unplug the USB cable from the side of the camera, turn it around the other way and re-insert it? (as USB-C cables are two-way insertion at the micro-sized end of the cable)

YBachmann, 2021-09-10:
Yes, rotating the USB-C Connector by 180° makes no difference. 
I also tried different USB ports on my laptop and on a different laptop.

MartyG-RealSense, 2021-09-10:
Is RGB restored if you downgrade the version from 1.5.8.1 to the previous version **1.5.5.0**

https://dev.intelrealsense.com/docs/firmware-releases-l500#section-l500-series-firmware-downloads

YBachmann, 2021-09-13:
Thank you very much for your help but unfortunately the error still persists: 
[image]

MartyG-RealSense, 2021-09-13:
Thanks very much.  I have referred your case to Intel to seek advice.  Thanks very much for your patience!

MartyG-RealSense, 2021-09-13:
I have received advice from Intel.  They request that you return your camera to the supplier that you purchased it from using that supplier's **Return Materials Authorization (RMA)** returns process.

Could you also email the serial number of your L515 camera to me at the email address below, please.  Thanks!

martyx.grover@intel.com

MartyG-RealSense, 2021-09-16:
Thank you very much, I have received the serial number of your camera by email and will pass it to Intel.