realsenseai/librealsense #9741: Error while opening "Intel.RealSense.Viewer.exe"

MuhammadBilal1, 2021-09-08:
| | |
|---|---|
|**librealsense**|2.45.0 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
Program terminated due to an unrecoverable SEH exception:
Access Violation!
```

MartyG-RealSense, 2021-09-08:
Hi @MuhammadBilal1  I will direct you first to a recent discussion about this error at https://github.com/IntelRealSense/librealsense/issues/9672

If the advice in that discussion does not resolve the problem for you, please let me know and I will be pleased to continue helping you.

MuhammadBilal1, 2021-09-08:
> Hi @MuhammadBilal1 I will direct you first to a recent discussion about this error at #9672
> 
> If the advice in that discussion does not resolve the problem for you, please let me know and I will be pleased to continue helping you.

Thank you for a quick response.

I tried to do the same as instructed on the #9672  , but still no progress. Getting the same issue.
The attached image shows all the details which I have.

[image]

MartyG-RealSense, 2021-09-08:
Are you able to successfully use the **rs-capture** pre-built example program using the instructions in https://github.com/IntelRealSense/librealsense/issues/8024#issuecomment-747630656

MuhammadBilal1, 2021-09-09:
No, I am not able to run the **rs-capture** pre-built example program. The attached screenshot shows the results.
I performed the same test on **Intel.RealSense.SDK-WIN10-2.47.0.3309** & **Intel.RealSense.SDK-WIN10-2.49.0.3474**, but the result is same from both.

[image]

MartyG-RealSense, 2021-09-09:
Are you using the official 1 meter USB cable supplied with the camera or a longer cable of your own choice, please?

Also, do you know which firmware driver version your camera is using?

MuhammadBilal1, 2021-09-09:
**Cable:** Yes, I am using the original cable that came in the camera box.

Now I just installed the [Linux Distribution](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages) of librealsense on a Ubuntu machine.
It was installed successfully on Ubuntu and Intel RealSense Viewer opened. Upon first time opening it gave me the option to update the firmware, I updated the firmware to the latest available firmware.

After updating the firmware on the Ubuntu machine, I again connected the camera to the Windows machine. This time the [rs-capture.exe](https://github.com/IntelRealSense/librealsense/issues/8024#issuecomment-747630656) worked for me on windows.

**But still the Intel RealSense Viewer isn't working on Windows machine.**

[image]

MartyG-RealSense, 2021-09-09:
Thanks very much @MuhammadBilal1 for the update of your positive progress.  As rs-capture works on that PC, it suggests that there is not a problem with the **OpenGL** graphics drivers on your computer, as both rs-capture and the Viewer make use of OpenGL. 

Your situation is like the case https://github.com/IntelRealSense/librealsense/issues/7185#issuecomment-688457080 in which a RealSense user also had an access violation on Windows.  In the end they mostly managed to resolve the problem but still could not get the Viewer to work.  In their particular case the Viewer was not important to their work though.  Their solution was to remove hidden camera driver entries in the Device Manager of Windows, as described in https://github.com/IntelRealSense/librealsense/issues/7185#issuecomment-688837669

May I also ask whether you are able to launch the **Depth Quality Tool** program, which has a similar design to the Viewer?

MartyG-RealSense, 2021-09-15:
Hi @MuhammadBilal1  Do you require further assistance with this case, please?  Thanks!

MuhammadBilal1, 2021-09-15:
> Hi @MuhammadBilal1 Do you require further assistance with this case, please? Thanks!

I didn't get the time to check your latest hunch, but hopefully tomorrow I will try that.
I will update you after checking https://github.com/IntelRealSense/librealsense/issues/7185#issuecomment-688837669.
Thanks for asking.

MartyG-RealSense, 2021-09-15:
Thanks very much @MuhammadBilal1  :)

illmade-knight, 2021-09-16:
I am having a similar issue.
rs-capture works but not rs-depth-quality or realsense-viewer

realsense-viewer worked yesterday but not today...

|---|---|
|**librealsense**|2.48.0 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
Unhandled exception escaping from a worker thread!
Error type: Access Violation!
```

Device info:
    Name                          :     Intel RealSense D435I
    Serial Number                 :     048122071624
    Firmware Version              :     05.12.15.50
    Recommended Firmware Version  :     05.12.14.50
    Physical Port                 :     \\?\usb#vid_8086&pid_0b3a&mi_00#6&2efa4cdc&0&0000#{e5323777-f976-4f5b-9b55-b94699c46e44}\global
    Debug Op Code                 :     15
    Advanced Mode                 :     YES
    Product Id                    :     0B3A
    Camera Locked                 :     YES
    Usb Type Descriptor           :     3.2
    Product Line                  :     D400
    Asic Serial Number            :     046523052244
    Firmware Update Id            :     046523052244

illmade-knight, 2021-09-16:
This resolved when I downloaded an older version of the viewer and ran it. The newer version then started working...

MartyG-RealSense, 2021-09-17:
Thanks very much @illmade-knight for the update about your particular problem - it's great to hear that you found a solution!

@MuhammadBilal1  I look forward to your next update.  Thanks!

MartyG-RealSense, 2021-09-23:
Hi @MuhammadBilal1  Do you have an update about your case that you can provide, please, as you said in https://github.com/IntelRealSense/librealsense/issues/9741#issuecomment-919759178 that you were going to check https://github.com/IntelRealSense/librealsense/issues/7185#issuecomment-688837669

MuhammadBilal1, 2021-09-28:
Sorry, I don't have the time to check the said instructions for now. You can close the ticket. I will get back to you if I need any assistance.
Thank you for your support.

MartyG-RealSense, 2021-09-28:
You are very welcome, @MuhammadBilal1 - thanks very much for the update.  As you suggested, I will close the ticket.  Thanks again!

closed 2021-09-28 by MartyG-RealSense