realsenseai/librealsense #9743: Jeson Not reading realsense after firmware upgrade

master-25, 2021-09-08:
| | |
|---|---|
|**librealsense**|2.49.0 RELEASE|
|**OS**|Linux|
|**Name**|Intel RealSense D4XX Recovery|
|**Product Line**|D400|
|**Firmware Update Id**|919423050830|

Please provide a description of the problem

MartyG-RealSense, 2021-09-09:
Hi @master-25  Let's first try using the RealSense SDK's **rs-fw-update** firmware updating tool in its **recovery** mode.

1.  Visit the firmware releases page for the 400 Series cameras and download the firmware file **Signed_Image_UVC_5_12_15_50.bin** (which is the recommended firmware for SDK version 2.49.0).

https://dev.intelrealsense.com/docs/firmware-releases#section-d400-series-firmware-downloads

2.  Place the firmware bin file in the same folder location that the rs-fw-update tool is located in.  In Linux, this would be **usr/local/bin**

3.  Launch rs-fw-update in recovery mode using the instruction below:

**rs-fw-update -r -f Signed_Image_UVC_5_12_15_50.bin**

MartyG-RealSense, 2021-09-15:
Hi @master-25  Do you require further assistance with this case, please?  Thanks!

master-25, 2021-09-15:
Sorry for the late reply, no I haven't had the time to work with it due to
my thesis sorry, I'll confirm by Monday

MartyG-RealSense, 2021-09-15:
That's no problem at all, @master-25 - thanks very much for the update, and good luck with your thesis!

master-25, 2021-09-15:
Its working but after I moving the .bin file to /usr/local/bin, the command still didn't work. Though the realsense-viewer worked perfectly. Thanks for your help, and thank I need all the luck I can get for my thesis.

MartyG-RealSense, 2021-09-16:
Great to hear that the camera is working for you in the RealSense Viewer now - thanks for the update!

master-25, 2021-09-23:
No, it's all good and working fine, thank you for your support

MartyG-RealSense, 2021-09-23:
Thanks very much @master-25 for the update!  As you do not require further assistance, I will close the case.  Thanks again!

closed 2021-09-23 by MartyG-RealSense