realsenseai/librealsense #9744: Is arm Mali T860 supported?

liu-jiaming, 2021-09-09:
Is CUDA acceleration only supported here? Is arm Mali T860 supported? I have a board with a GPU of Mali T860. I want to know if I can use this GPU to speed up.

MartyG-RealSense, 2021-09-09:
Hi @liu-jiaming  The librealsense SDK supports an alternative method to CUDA called **GLSL** for offloading CPU work for certain functions onto the GPU.  Unlike CUDA's requirement of an Nvidia GPU, GLSL is 'vendor-neutral' and should work with any brand of GPU.

https://github.com/IntelRealSense/librealsense/pull/3654 has a very good pros and cons analysis of GLSL processing blocks and when it can be used.

There is also an example **rs-gl** program in C++ language:

https://github.com/IntelRealSense/librealsense/tree/master/examples/gl

liu-jiaming, 2021-09-09:
> 嗨@liu- jiaming librealsense SDK 支持一种称为**GLSL 的**CUDA 替代方法，用于将某些功能的 CPU 工作卸载到 GPU 上。与 CUDA 对 Nvidia GPU 的要求不同，GLSL 是“供应商中立的”，应该适用于任何品牌的 GPU。
> 
> #3654对 GLSL 处理块以及何时可以使用有非常好的利弊分析。
> 
> 还有一个C++ 语言的示例**rs-gl**程序：
> 
> https://github.com/IntelRealSense/librealsense/tree/master/examples/gl

When compiling the source code, I can use it without additional parameters, right? The following is my cmake
cmake .. -DBUILD_EXAMPLES=false -DCMAKE_BUILD_TYPE=release -DBUILD_PYTHON_BINDINGS=true

MartyG-RealSense, 2021-09-09:
You do not need to add terms to your CMake build instruction to enable GLSL.  For the functions that support GLSL, it is enabled by adding **::gl** to rs2:: instructions, as described in the **How to use it?** section of the GLSL guide that I linked to.  For example, **rs2::pointcloud** becomes **rs2::gl::pointcloud**

https://github.com/IntelRealSense/librealsense/blob/master/examples/gl/rs-gl.cpp#L54

closed 2021-09-16 by liu-jiaming