realsenseai/librealsense #9747: Setup for NodeJS at Apple Silicon

shadowgroundz, 2021-09-09:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 01.05.05.00 | 
| Operating System & Version |  MacOS Apple Silicon | 
| SDK Version                            |2| 
| Language                            |  nodejs | 

### Issue Description

Hi, I feel confused when setting up a new project with NodeJS language. Step I do :
1. Install `XCode` from the apps store 
2. Already installed `homebrew`
3. Install the following packages via brew:
`brew install cmake libusb pkg-config`
`brew install arch -arm64 brew install vulkan-sdk` (adjusted for Apple M1)
4. Set env variable
```
export OPENSSL_ROOT_DIR=/opt/homebrew/Cellar/openssl@1.1/1.1.1l
export OPENSSL_LIBRARIES=/opt/homebrew/Cellar/openssl@1.1/1.1.1l/lib
export OPENSSL_CRYPTO_LIBRARY=/opt/homebrew/Cellar/openssl@1.1/1.1.1l/lib/libcrypto.a
export OPENSSL_INCLUDE_DIR=/opt/homebrew/Cellar/openssl@1.1/1.1.1l/include
```
5. `git clone https://github.com/IntelRealSense/librealsense`
6. `cd librealsense`
7. Edit file `src/proc/color-formats-converter.cpp` and change `#if defined (ANDROID) || (defined (__linux__) && !defined (__x86_64__))` to `#if defined (ANDROID) || (defined (__linux__) && !defined (__x86_64__)) || defined(__aarch64__)` https://github.com/IntelRealSense/librealsense/issues/8090#issuecomment-826164584
8. `mkdir build && cd build`
9. `cmake .. -DCMAKE_BUILD_TYPE=Release -DBUILD_EXAMPLES=true -DBUILD_NODEJS_BINDINGS=1 -DCMAKE_APPLE_SILICON_PROCESSOR=arm64 -DCMAKE_BUILD_TYPE=release`
10. `make -j2`

But from all step reproduce 2 error 
1. 
`[ 99%] Linking CXX executable realsense-viewer`
`ld: warning: ignoring file ../../libcurl/libcurl_install/lib/libcurl.a, building for macOS-arm64 but attempting to link with file built for macOS-x86_64`
2.
`[100%] Linking CXX executable rs-depth-quality`
`ld: warning: ignoring file ../../libcurl/libcurl_install/lib/libcurl.a, building for macOS-arm64 but attempting to link with file built for macOS-x86_64`

I'm frustrated LOL. Anyone can help me? I know you may be referred to this #8090 , but I don't know what step I wrong. Thanks

gschian0, 2021-09-09:
I am still having errors over here building on M1.  Let's keep trying!! Success is on the way!

gschian0, 2021-09-09:
**I get stuck here...** 

the error is in the `/src/proc/color-formats.cpp` file

**setting up the project with**
`cmake .. -DCMAKE_BUILD_TYPE=Release -DBUILD_EXAMPLES=true -DBUILD_NODEJS_BINDINGS=1 -DCMAKE_APPLE_SILICON_PROCESSOR=arm64 -DCMAKE_BUILD_TYPE=release`

**when running** `make -j4`

`1 warning generated.
[ 29%] Building CXX object CMakeFiles/realsense2.dir/src/proc/depth-formats-converter.cpp.o
/Users/gennaroschiano/Documents/GitHub/librealsense/src/proc/color-formats-converter.cpp:21:73: error: token is not a valid binary operator in a preprocessor subexpression
#if defined (ANDROID) || (defined (__linux__) && !defined (__x86_64__)) to #if defined (ANDROID) || (defined (__linux__) && !defined (__x86_64__)) || defined(__aarch64__)
                         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ ^
In file included from /Users/gennaroschiano/Documents/GitHub/librealsense/src/proc/color-formats-converter.cpp:31:
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/clang/12.0.5/include/cpuid.h:11:2: error: this header is for x86 only
#error this header is for x86 only
 ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/clang/12.0.5/include/cpuid.h:280:5: error: invalid output constraint '=a' in asm
    __cpuid(__leaf, __eax, __ebx, __ecx, __edx);
    ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/clang/12.0.5/include/cpuid.h:245:11: note: expanded from macro '__cpuid'
        : "=a"(__eax), "=r" (__ebx), "=c"(__ecx), "=d"(__edx) \
          ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/clang/12.0.5/include/cpuid.h:295:5: error: invalid output constraint '=a' in asm
    __cpuid(__leaf, *__eax, *__ebx, *__ecx, *__edx);
    ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/clang/12.0.5/include/cpuid.h:245:11: note: expanded from macro '__cpuid'
        : "=a"(__eax), "=r" (__ebx), "=c"(__ecx), "=d"(__edx) \
          ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/clang/12.0.5/include/cpuid.h:309:5: error: invalid output constraint '=a' in asm
    __cpuid_count(__leaf, __subleaf, *__eax, *__ebx, *__ecx, *__edx);
    ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/clang/12.0.5/include/cpuid.h:252:11: note: expanded from macro '__cpuid_count'
        : "=a"(__eax), "=r" (__ebx), "=c"(__ecx), "=d"(__edx) \
          ^
/Users/gennaroschiano/Documents/GitHub/librealsense/src/proc/color-formats-converter.cpp:33:5: error: invalid output constraint '=a' in asm
    __cpuid_count(info_type, 0, info[0], info[1], info[2], info[3]);
    ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/clang/12.0.5/include/cpuid.h:252:11: note: expanded from macro '__cpuid_count'
        : "=a"(__eax), "=r" (__ebx), "=c"(__ecx), "=d"(__edx) \
          ^
6 errors generated.
make[2]: *** [CMakeFiles/realsense2.dir/src/proc/color-formats-converter.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
clang: warning: -latomic: 'linker' input unused [-Wunused-command-line-argument]
clang: warning: argument unused during compilation: '-mfpu=neon' [-Wunused-command-line-argument]
clang: warning: argument unused during compilation: '-mfloat-abi=hard' [-Wunused-command-line-argument]
make[1]: *** [CMakeFiles/realsense2.dir/all] Error 2
make: *** [all] Error 2`

**on other build settings ... I get this error** `cpuid.h` **is causing it**

`4 warnings generated.
[ 30%] Building CXX object CMakeFiles/realsense2.dir/src/proc/auto-exposure-processor.cpp.o
/Users/gennaroschiano/Documents/GitHub/librealsense/src/proc/color-formats-converter.cpp:21:73: error: token is not a valid binary operator in a preprocessor subexpression
#if defined (ANDROID) || (defined (__linux__) && !defined (__x86_64__)) to #if defined (ANDROID) || (defined (__linux__) && !defined (__x86_64__)) || defined(__aarch64__)
                         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ ^
In file included from /Users/gennaroschiano/Documents/GitHub/librealsense/src/proc/color-formats-converter.cpp:31:
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/clang/12.0.5/include/cpuid.h:11:2: error: this header is for x86 only
#error this header is for x86 only
 ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/clang/12.0.5/include/cpuid.h:280:5: error: invalid output constraint '=a' in asm
    __cpuid(__leaf, __eax, __ebx, __ecx, __edx);
    ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/clang/12.0.5/include/cpuid.h:245:11: note: expanded from macro '__cpuid'
        : "=a"(__eax), "=r" (__ebx), "=c"(__ecx), "=d"(__edx) \
          ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/clang/12.0.5/include/cpuid.h:295:5: error: invalid output constraint '=a' in asm
    __cpuid(__leaf, *__eax, *__ebx, *__ecx, *__edx);
    ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/clang/12.0.5/include/cpuid.h:245:11: note: expanded from macro '__cpuid'
        : "=a"(__eax), "=r" (__ebx), "=c"(__ecx), "=d"(__edx) \
          ^
[... cut ...]
        : "=a"(__eax), "=r" (__ebx), "=c"(__ecx), "=d"(__edx) \
          ^
/Users/gennaroschiano/Documents/GitHub/librealsense/src/proc/color-formats-converter.cpp:33:5: error: invalid output constraint '=a' in asm
    __cpuid_count(info_type, 0, info[0], info[1], info[2], info[3]);
    ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/clang/12.0.5/include/cpuid.h:252:11: note: expanded from macro '__cpuid_count'
        : "=a"(__eax), "=r" (__ebx), "=c"(__ecx), "=d"(__edx) \`

shadowgroundz, 2021-09-09:
> I am still having errors over here building on M1.  Let's keep trying!! Success is on the way!

Yes!! I hope Intel developer can make better documentation and proper step to build on M1

shadowgroundz, 2021-09-09:
> **I get stuck here...**
>
> **the error is in the** `/src/proc/color-formats.cpp` **file**
>
> **setting up the project with**
> `cmake .. -DCMAKE_BUILD_TYPE=Release -DBUILD_EXAMPLES=true [...]

I think you edit wrong file bro. See my step on number 7. Make sure after edit you delete all things on build folder and do `cmake .......` again

gschian0, 2021-09-09:
**I've also tried to compile with xcode using this command**

`cmake .. -DCMAKE_INSTALL_PREFIX=/opt/homebrew/Cellar/librealsense/2.48.0 -DBUILD_WITH_OPENMP=OFF -DENABLE_CCACHE=OFF -DCMAKE_HAVE_THREADS_LIBRARY=1 -DCMAKE_THREAD_LIBS_INIT="-lpthread" -DCMAKE_USE_WIN32_THREADS_INIT=0 -DTHREADS_PREFER_PTHREAD_FLAG=ON -DCMAKE_USE_PTHREADS_INIT=1 -DOPENSSL_ROOT_DIR=/opt/homebrew/opt/openssl -DCMAKE_CXX_FLAGS="-w" -DCMAKE_C_FLAGS="-w" -G Xcode`

**then open in Xcode and try to compile get this error**

`
Showing Recent Messages
Ld /Users/gennaroschiano/Documents/GitHub/librealsense/build/librealsense2.build/Debug/realsense2.build/Objects-normal/x86_64/Binary/librealsense2.2.49.0.dylib normal x86_64 (in target 'realsense2' from project 'librealsense2')
    cd /Users/gennaroschiano/Documents/GitHub/librealsense
    /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang++ -target x86_64-apple-macos11.3 -dynamiclib -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX11.3.sdk -L/Users/gennaroschiano/Documents/GitHub/librealsense/build/Debug -F/Users/gennaroschiano/Documents/GitHub/librealsense/build/Debug -filelist /Users/gennaroschiano/Documents/GitHub/librealsense/build/librealsense2.build/Debug/realsense2.build/Objects-normal/x86_64/realsense2.2.49.0.LinkFileList -install_name /librealsense2.2.49.0.dylib -Xlinker -object_path_lto -Xlinker /Users/gennaroschiano/Documents/GitHub/librealsense/build/librealsense2.build/Debug/realsense2.build/Objects-normal/x86_64/realsense2.2.49.0_lto.o -Xlinker -no_deduplicate -dynamiclib -Wl,-headerpad_max_install_names -install_name @rpath/librealsense2.2.49.dylib -framework CoreFoundation -framework IOKit /Users/gennaroschiano/Documents/GitHub/librealsense/build/common/fw/Debug/libfw.a /Users/gennaroschiano/Documents/GitHub/librealsense/build/third-party/realsense-file/Debug/librealsense-file.a -lpthread /opt/homebrew/lib/libusb-1.0.dylib -Xlinker -no_adhoc_codesign -compatibility_version 2.49.0 -current_version 2.49.0 -Xlinker -dependency_info -Xlinker /Users/gennaroschiano/Documents/GitHub/librealsense/build/librealsense2.build/Debug/realsense2.build/Objects-normal/x86_64/realsense2.2.49.0_dependency_info.dat -o /Users/gennaroschiano/Documents/GitHub/librealsense/build/librealsense2.build/Debug/realsense2.build/Objects-normal/x86_64/Binary/librealsense2.2.49.0.dylib

ld: warning: ignoring file /opt/homebrew/lib/libusb-1.0.dylib, building for macOS-x86_64 but attempting to link with file built for macOS-arm64
Undefined symbols for architecture x86_64:
  "_libusb_alloc_transfer", referenced from:
      librealsense::platform::usb_request_libusb::usb_request_libusb(libusb_device_handle*, std::__1::shared_ptr<librealsense::platform::usb_endpoint>) in request-libusb.o
  "_libusb_bulk_transfer", referenced from:
      librealsense::platform::usb_messenger_libusb::bulk_transfer(std::__1::shared_ptr<librealsense::platform::usb_endpoint> const&, unsigned char*, unsigned int, unsigned int&, unsigned int) in messenger-libusb.o
  "_libusb_cancel_transfer", referenced from:
      librealsense::platform::usb_messenger_libusb::cancel_request(std::__1::shared_ptr<librealsense::platform::usb_request> const&) in messenger-libusb.o
      librealsense::platform::usb_request_libusb::~usb_request_libusb() in request-libusb.o
  "_libusb_claim_interface", referenced from:
      librealsense::platform::handle_libusb::claim_interface(unsigned char) in device-libusb.o
  "_libusb_clear_halt", referenced from:
      librealsense::platform::usb_messenger_libusb::reset_endpoint(std::__1::shared_ptr<librealsense::platform::usb_endpoint> const&, unsigned int) in messenger-libusb.o
  "_libusb_close", referenced from:
      librealsense::platform::handle_libusb::~handle_libusb() in device-libusb.o
  "_libusb_control_transfer", referenced from:
      librealsense::platform::usb_messenger_libusb::control_transfer(int, int, int, int, unsigned char*, unsigned int, unsigned int&, unsigned int) in messenger-libusb.o
  "_libusb_detach_kernel_driver", referenced from:
      librealsense::platform::handle_libusb::claim_interface(unsigned char) in device-libusb.o
  "_libusb_exit", referenced from:
      librealsense::platform::usb_context::~usb_context() in context-libusb.o
  "_libusb_free_config_descriptor", referenced from:
      librealsense::platform::usb_device_libusb::usb_device_libusb(libusb_device*, libusb_device_descriptor const&, librealsense::platform::usb_device_info const&, std::__1::shared_ptr<librealsense::platform::usb_context>) in device-libusb.o
      librealsense::platform::get_subdevices(libusb_device*, libusb_device_descriptor) in enumerator-libusb.o
  "_libusb_free_device_list", referenced from:
      librealsense::platform::usb_context::~usb_context() in context-libusb.o
  "_libusb_free_transfer", referenced from:
      librealsense::platform::usb_request_libusb::usb_request_libusb(libusb_device_handle*, std::__1::shared_ptr<librealsense::platform::usb_endpoint>)::$_0::operator()(libusb_transfer*) const in request-libusb.o
  "_libusb_get_bus_number", referenced from:
      librealsense::platform::get_device_path(libusb_device*) in enumerator-libusb.o
  "_libusb_get_config_descriptor", referenced from:
      librealsense::platform::usb_device_libusb::usb_device_libusb(libusb_device*, libusb_device_descriptor const&, librealsense::platform::usb_device_info const&, std::__1::shared_ptr<librealsense::platform::us [... cut ...]
    /bin/sh -c /Users/gennaroschiano/Documents/GitHub/librealsense/build/librealsense2.build/Debug/libcurl.build/Script-611BFFFE38F83ED1EBDEAE07.sh

WriteAuxiliaryFile /Users/gennaroschiano/Documents/GitHub/librealsense/build/librealsense2.build/Debug/libcurl.build/Script-D316BE8B38B0E3DE179CA6A2.sh (in target 'libcurl' from project 'librealsense2')
    cd /Users/gennaroschiano/Documents/GitHub/librealsense
    write-file /Users/gennaroschiano/Documents/GitHub/librealsense/build/librealsense2.build/Debug/libcurl.build/Script-D316BE8B38B0E3DE179CA6A2.sh

`

shadowgroundz, 2021-09-09:
You got error when run cmake? Not make?

gschian0, 2021-09-09:
I get the error running make now... or when trying to build using Xcode.

shadowgroundz, 2021-09-09:
Try to add this option
`-DCMAKE_APPLE_SILICON_PROCESSOR=arm64`

Anyway, what your target programming language?

gschian0, 2021-09-09:
ultimately I was planning on using C.  I want to create a program that stitches multiple realsense cameras in real time and be able to send that texture into realtime 3d software.  So my plan was using C to stitch together multiple feeds that I can then lookup from my 3d software to create the 3d effects in real time like unreal engine I'm getting started on this on pc and it's moving slow but moving haha.  Ultimately I like how the shaders look with the default setup on the intel realsense for some performances I want to do and composite the real sense images into another 3D scene in real time... I just want to use it on my mac to do some live performances.

shadowgroundz, 2021-09-09:
If we succeed build in Apple M1, we can use it with any programming language?

shadowgroundz, 2021-09-09:
> Hi @shadowgroundz  As you suggested, https://github.com/IntelRealSense/librealsense/issues/8090 is usually the best place to seek advice about installing librealsense on an Apple Silicon M1 model of Mac, and the fellow M1 users are always pleased to offer advice in addition to the solutions already posted.  There are also links to other discussions about M1 installation within https://github.com/IntelRealSense/librealsense/issues/8090 and most problems that are encountered have already had possible solutions suggested in these cases.
> 
> As your Intel support contact on this case, I am also happy to help if possible (though I do not have a Mac at my location to test with).
> 
> Once the RealSense SDK is successfully installed on your M1 Mac then you should be able to use it with any language that the SDK is able to interface with and which works on your Mac.  A directory of compatibility wrappers can be found at the link below.
> 
> https://github.com/IntelRealSense/librealsense/tree/master/wrappers   

I have one example that success build on xcode.I'm not familiar with XCode. When finished build, I try to go to `wrappers` on root => `nodeJS` and try to `npm install `and enter to `examples` folder and run `node ./nodejs-capture.js` and show `modules not found`

I'll give exact information about this error soon. Because I not front my laptop. But, did I do correct step?

MartyG-RealSense, 2021-09-09:
@shadowgroundz  A RealSense user successfully built the Node.js wrapper with **npm install** on a MacOS Catalina machine in May 2021 )probably therefore an older Intel-based Mac model rather than an M1).  Their case is at https://github.com/IntelRealSense/librealsense/issues/9068

shadowgroundz, 2021-09-10:
> @shadowgroundz A RealSense user successfully built the Node.js wrapper with **npm install** on a MacOS Catalina machine in May 2021 (probably therefore an older Intel-based Mac model rather than an M1). Their case is at #9068

Okay I got the pattern from #8090 and follow this instruction https://dev.intelrealsense.com/docs/macos-installation-for-intel-realsense-sdk

1. Set env variable
```
export OPENSSL_ROOT_DIR=/opt/homebrew/Cellar/openssl@1.1/1.1.1l
export OPENSSL_LIBRARIES=/opt/homebrew/Cellar/openssl@1.1/1.1.1l/lib
export OPENSSL_CRYPTO_LIBRARY=/opt/homebrew/Cellar/openssl@1.1/1.1.1l/lib/libcrypto.a
export OPENSSL_INCLUDE_DIR=/opt/homebrew/Cellar/openssl@1.1/1.1.1l/include
```
2.  `git clone https://github.com/IntelRealSense/librealsense`
3. `cd librealsense`
4. Edit file `src/proc/color-formats-converter.cpp` and change `#if defined (ANDROID) || (defined (__linux__) && !defined (__x86_64__)) `to `#if defined (ANDROID) || (defined (__linux__) && !defined (__x86_64__)) || defined(__aarch64__)` same with  https://github.com/IntelRealSense/librealsense/issues/8090#issuecomment-826164584
5. Edit file `CMake/external_libcurl.cmake` same as https://github.com/IntelRealSense/librealsense/issues/8090#issuecomment-836892782
6. `cmake .. -DBUILD_EXAMPLES=true -DBUILD_WITH_OPENMP=false -DCMAKE_APPLE_SILICON_PROCESSOR=arm64 -DHWM_OVER_XU=false -G Xcode` . I add `-DCMAKE_APPLE_SILICON_PROCESSOR=arm64` because if not I got error on `libusb`
7. Open `librealsense2.xcodeproj` then go to `Product` -> `Clean Build Folder` then `Product` -> `Build`

And voila. Build success. But, I don't know how to test. I'm trying to do this
Go to root folder -> wrappers/nodejs and `npm install`. But I got error `clang: error: no such file or directory: 'XXXXXXXX/wrappers/nodejs/../../build/Release/librealsense2.dylib'`

In other `CMAKE` configurations, there is librealsense2.dylib in `build` folder. So I make a new folder Release and copy `librealsense2.dylib`. Any suggestion?

shadowgroundz, 2021-09-10:
> @shadowgroundz A RealSense user successfully built the Node.js wrapper with **npm install** on a MacOS Catalina machine in May 2021 (probably therefore an older Intel-based Mac model rather than an M1). Their case is at #9068

And anyway it looks different error with mine

shadowgroundz, 2021-09-10:
I success running realsense-viewer on xcode. But I don't know how to use with nodejs, Becasu have an error when trying to `npm install`

[image]

gschian0, 2021-09-10:
node I understand very well.. I'm going to try your suggested build pattern now.   What is the node error?

shadowgroundz, 2021-09-10:
> node I understand very well.. I'm going to try your suggested build pattern now. What is the node error?

Thanks to try help. I got some error when `npm install` : `clang: error: no such file or directory: 'XXXXXXXX/wrappers/nodejs/../../build/Release/librealsense2.dylib'`

gschian0, 2021-09-10:
You have to build the bindings in the build command for node.. try to add that.

gschian0, 2021-09-10:
hmm I feel like running those path commands messed me up... now open ssl isn't linking.

shadowgroundz, 2021-09-10:
> hmm I feel like running those path commands messed me up... now open ssl isn't linking.

Hmm adjust with your own

shadowgroundz, 2021-09-10:
> You have to build the bindings in the build command for node.. try to add that.

I found a new error when `node nodejs-capture.js` in `examples` folder. Step to reproduce : 
1. I build with `cmake .. -DBUILD_EXAMPLES=true -DBUILD_WITH_OPENMP=false -DCMAKE_BUILD_TYPE=release -DCMAKE_APPLE_SILICON_PROCESSOR=arm64 -DHWM_OVER_XU=false -G Xcode` and build with `xcode`
2. After success build with `xcode`, enter to `/wrappers/nodejs` and `npm install`
3. Enter to `examples` folder and `npm install` again
4. Run `node nodejs-capture.js`

```
dyld: lazy symbol binding failed: Symbol not found: _glfwInit
  Referenced from: /Users/vinzlee/Documents/IntelRealSense/testing/librealsenseV2/wrappers/nodejs/examples/node_modules/node-12-glfw-3/build/Release/glfw.node
  Expected in: flat namespace

dyld: Symbol not found: _glfwInit
  Referenced from: /Users/vinzlee/Documents/IntelRealSense/testing/librealsenseV2/wrappers/nodejs/examples/node_modules/node-12-glfw-3/build/Release/glfw.node
  Expected in: flat namespace

zsh: abort      node nodejs-capture.js
```

shadowgroundz, 2021-09-10:
I forgot to mention, at https://dev.intelrealsense.com/docs/macos-installation-for-intel-realsense-sdk I change `glfw3` at step `brew install homebrew/core/glfw3` because M1 not support on M1

shadowgroundz, 2021-09-10:
> I forgot to mention, at https://dev.intelrealsense.com/docs/macos-installation-for-intel-realsense-sdk I change `glfw3` to `glfw` at step `brew install homebrew/core/glfw3` because M1 not support on M1

gschian0, 2021-09-10:
finally got it to compile following your steps.. I added to the file .. `external_libcurl.cmale` after line 35

`-DOPENSSL_ROOT_DIR=/opt/homebrew/Cellar/openssl@1.1/1.1.1l`

[image]

[image]
`

gschian0, 2021-09-10:
It is running pretty laggy to start however... new sleuthing to go on!

gschian0, 2021-09-10:
Thanks @MartyG-RealSense for continuing to follow up with me... I know I will need more help now that I'm finally using the software!

shadowgroundz, 2021-09-10:
> Thanks @MartyG-RealSense for continuing to follow up with me... I know I will need more help now that I'm finally using the software!

Congrats

MartyG-RealSense, 2021-09-10:
@gschian0   It's great to hear that you got it working!

@shadowgroundz   Thanks very much for your advice to @gschian0  :)

shadowgroundz, 2021-09-10:
Okay, after trial and error, finally I got the perfect step for mine (I hope can help others too). And please note my environment, different versions may cause different effects !!

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model | L515 | 
| Firmware Version | 01.05.08.01 | 
| Device | Macbook Pro Apple Silicon (M1)|
| Operating System & Version |  MacOS  11.5.2| 
| SDK Version |2| 
| Target Programming Language |  NodeJS | 
| NodeJS Version| 14.17.3|
---------------------------------

**STEP**

1. Same as instruction at this link https://dev.intelrealsense.com/docs/macos-installation-for-intel-realsense-sdk . Make sure you install `Xcode` via App Store & `HomeBrew` running on native and make sure installed on `/opt/homebrew` not `usr/local` because `usr/local` running on `rosetta`

2. Install homebrew package
```
brew install libusb pkg-config
brew install glfw
brew install cmake
```
Not use `brew install homebrew/core/glfw3` because not compatible in M1. Make sure all packages installed on `/opt/homebrew/lib` . If failed follow an instruction

3. Set env variable
```
export OPENSSL_ROOT_DIR=/opt/homebrew/Cellar/openssl@1.1/1.1.1l
export OPENSSL_LIBRARIES=/opt/homebrew/Cellar/openssl@1.1/1.1.1l/lib
export OPENSSL_CRYPTO_LIBRARY=/opt/homebrew/Cellar/openssl@1.1/1.1.1l/lib/libcrypto.a
export OPENSSL_INCLUDE_DIR=/opt/homebrew/Cellar/openssl@1.1/1.1.1l/include
```
In my case, I use `.zshrc` at `/Users/Username/.zshrc`. So edit the file and `source ~/.zshrc` and make sure use path same with your installation path
4. Clone repository from `git clone https://github.com/IntelRealSense/librealsense`
5. Edit file `src/proc/color-formats-converter.cpp` and change `#if defined (ANDROID) || (defined (__linux__) && !defined (__x86_64__))` to `#if defined (ANDROID) || (defined (__linux__) && !defined (__x86_64__)) || defined(__aarch64__)` at line 21
6. Edit file `CMake/external_libcurl.cmake` and add `-DOPENSSL_ROOT_DIR=/opt/homebrew/opt/openssl` after line 35. Once again make sure to use same path with your current path
7. Enter directory `cd librealsense`
8. Make folder `build` and enter directory with command `mkdir build && cd build`
9. `sudo xcode-select --reset`
10.` cmake .. -DBUILD_EXAMPLES=true -DBUILD_WITH_OPENMP=false -DCMAKE_APPLE_SILICON_PROCESSOR=arm64 -DBUILD_NODEJS_BINDINGS=1 -DHWM_OVER_XU=false -G Xcode`
11. Open `librealsense2.xcodeproj` on folder build with `Xcode` then go to `Product` -> `Clean Build Folder` then `Product` -> `Build`
12. After build finish you can try to running viewer with connect camera via USB then select realsense-viewer and click button play as picture 
[image]

If you succeed running with camera, you finish to Build Native C++ librealsense on Apple Silicon M1
---------------------------------
**Running SDK with NodeJS Wrapper**
---------------------------------
1. After succeed make connection with camera, duplicate folder `build/Debug` that contain 
```
librealsense2.2.49.dylib
librealsense2.dylib
librealsense2.2.49.0.dylib
```
and rename duplicated folder to `Release`
2. `npm install` / `yarn install`
3. Enter examples directory and `npm install` / `yarn install` and  `node nodejs-capture.js`

Hope with detail step can help other and @MartyG-RealSense and other Intel team can improve documentation special for Apple Silicon (M1). Feel free if you have another question

closed 2021-09-10 by shadowgroundz

MartyG-RealSense, 2021-09-10:
Thanks so much @shadowgroundz  for sharing your guide!  Each solution to problems that M1 users share makes it easier for other M1 users to achieve a successful build of librealsense  :)