realsenseai/librealsense #9748: L515: Incorrect depth data on plants/vegetation

mfrobergsobolt, 2021-09-09:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L500 (L515) | 
| Firmware Version                   | 01.05.08.01 | 
| Operating System & Version |   Linux (Ubuntu 18.04) | 
| Kernel Version (Linux Only)    |  5.4.0-81-generic     | 
| Platform                                 | PC |
| SDK Version                            |  v2.45                   | 
| Language                            |  python                        | 
| Segment			|  Phenotyping                          | 

### Issue Description
I have been trying to get height measurements of plants using the L515 LiDAR camera, but the measured height from the acquired point clouds always undershoot the actual height of the plant by 1-1.5 cm. I have conducted various experiments with objects of known height in the same environment (camera position, camera settings, lighting conditions). For all objects the measured height deviated 0-0.5 cm from the actual height, except for plants! Are there certain camera configuration parameters or filtering techniques that I should use to increase the accuracy of the acquired depth data on plants?

Furthermore, I am wondering how the depth value of a pixel in a single frame is calculated. I can imagine that a single pulse from the laser of the camera returns multiple peaks of energy (see image below). Does the depth value correspond to the distance calculated from the highest peak?

[image]

MartyG-RealSense, 2021-09-09:
Hi @mfrobergsobolt   Plant leaves have light-reflective properties, so that may be affecting your depth results.  Highly reflective objects such as mirrors and other smooth surfaces have **specular** reflection.  With specular reflection, the laser light of the L515 may not get reflected back into its receiver for detection.  Observing such surfaces from a large angle (like the tree image above, but more diagonal) instead of straight down may help to improve depth measurement.

[image]

I would also recommend trying the **Short Range** L515 camera preset, which reduces the value of the **Laser Power** and **Receiver Gain** and may work better for a highly reflective surface.  You can set the preset from a drop-down menu in the side-panel of the RealSense Viewer.  If you need to set the Short Range preset in Python though, https://github.com/IntelRealSense/librealsense/issues/8619 provides a script that demonstrates a method for doing so by setting preset number '5'.

[image]

********

In regard to how the L515 model measures depth, Intel have published a detailed guide at the link below.

https://www.intelrealsense.com/optimizing-the-lidar-camera-l515-range/

MartyG-RealSense, 2021-09-16:
Hi @mfrobergsobolt  Do you require further assistance with this case, please?  Thanks!

mfrobergsobolt, 2021-09-16:
Hi @MartyG-RealSense, thanks a lot for your suggestions. I didn't get to do the experiments with the camera this week, but I'm planning to do them next week! I'll get back to you.

MartyG-RealSense, 2021-09-16:
That's no problem at all, @mfrobergsobolt - thanks very much for the update, and good luck with your tests next week  :)

MartyG-RealSense, 2021-09-27:
Hi @mfrobergsobolt  Do you have an update about this case that you can provide, please?  Thanks!

mfrobergsobolt, 2021-09-27:
Hi @MartyG-RealSense , last week I've done various experiments with different camera positions and the camera settings you suggested to get the most accurate height data of plants with the L515 camera. I used the following depth sensor settings for all different camera positions:

```
device = profile.get_device()
depth_sensor = device.query_sensors()[0]
depth_sensor.set_option(rs.option.visual_preset, 5)
depth_sensor.set_option(rs.option.confidence_threshold, 3)
depth_sensor.set_option(rs.option.noise_filtering, 6)
```

I used CloudCompare to measure the height of the plant within the point cloud. I always leveled the point cloud to the platform the container with plants was standing on. This means that the distance along the z-axis between two points represents the  height in a point cloud.

Looking at the height of the plant from the top (~50cm above the plant) with no angle the height data was clearly undershooting the actual height of the plant. In the image below you see that the height in the point cloud is 15.2 - 15.6 cm, but the actual height of the plant is 16.8 cm.

[image]

Changing the angle of the camera didn't do anything. See below the same environment but with the camera at a 45 degrees angle.

[image]

Only when I position the camera directly beside the plant, the height of the plant matches the actual plant height. 

[image]

However, from this position the LiDAR camera is not really measuring height, but more the width of the plant. I tried to illustrate this in the drawing below.

[image]

I don't know what else I can do to improve the acquired data except for acquiring data all around the plant, so both top- and side-view. Or is there something else that I can do?

MartyG-RealSense, 2021-09-27:
Thanks very much for the detailed feedback.  Perhaps you could measure directly in the SDK instead of using CloudCompare by measuring between two points on the pointcloud.  A RealSense user created a Python conversion (in the link below) of the SDK's C++ **rs-measure** example for measuring between two points.

https://github.com/soarwing52/RealsensePython/blob/master/separate%20functions/measure_new.py

closed 2021-10-09 by MartyG-RealSense