realsenseai/librealsense #9750: IR/daylight  interference

shindm, 2021-09-10:
Camera Model                       Intel RealSense L515
Firmware Version                   01.05.08.01
Operating System & Version  Linux Ubuntu 21.04
Kernel Version (Linux Only)    5.11.0-27-generic                                    
Platform                                 PC
SDK Version                           2.49.0 
Language                            python                           
Segment			 Motion Capture/VR

### Issue Description
I have a setup with L515 camera and Vicon Motion Capture IR cameras in a big bright room with big windows.   I often get these vertical columns in the color depth map images (see attached example of depth color map image and corresponding RGB image and depth values csv file).
I am wondering what's causing this interference. The Vicon IR cameras operate at different IR wavelength, could that be still the problem or it is the fact that the room is brightly lit and I should cover the windows?

[image]

[image]

[depth_values2.csv](https://github.com/IntelRealSense/librealsense/files/7140593/depth_values2.csv)

Thank you!

DS

MartyG-RealSense, 2021-09-10:
Hi @shindm   The L515 camera model is sensitive to interference from infrared sources and natural sunlight.  It should ideally be used in controlled indoor lighting conditions without sunlight.

You may not need to cover the windows if you set the camera to use the L515's **Low Ambient Light** camera preset, which can enable the camera to improve the depth image by compensating for the presence of sunlight.  

This setting and others are discussed in Intel's guide to optimizing L515 images in the link below, which should be helpful in enhancing your results.

https://www.intelrealsense.com/optimizing-the-lidar-camera-l515-range/

If you need to capture points on the lower body then I recommend not using black clothing, as dark grey and black colours are difficult for depth cameras to analyze for depth detail because they absorb light.  This is discussed in https://github.com/IntelRealSense/librealsense/issues/6757

MartyG-RealSense, 2021-09-17:
Hi @shindm  Do you require further assistance with this case, please?  Thanks!

shindm, 2021-09-17:
Hi Marty,

Thank you for your recommendations. So, after reading the suggested materials it is my understanding that placing two L515s on opposite sides of a room and pointing at each other may bring a lot of interference and result in bad data. Is that the case? In our motion capture lab, we wanted to place four L515s, one on each side of the room and point to the center. Can we still do it by adjusting parameters of L515 or it is a bad idea?

Thanks again!
DS

MartyG-RealSense, 2021-09-17:
If the cameras are facing each other but spaced far enough apart on each side of the room that their fields of view are not overlapping then interference should be low.

[image]

If the fields of view are overlapping then you can reduce interference by setting up **hardware synchronization** between the cameras.  This is described in the L515 multiple camera guide linked to below.

https://dev.intelrealsense.com/docs/lidar-camera-l515-multi-camera-setup

I cannot offer a prediction about what affect the presence of the Vicon IR cameras may have on the results though.

If you have not purchased the four L515 cameras yet then the **D455** model may work better for your project.  It has a long range and high depth sensing accuracy even at 6 meters from the camera and can operate in strong light environments such as outdoors.  Multiple 400 Series cameras also do not interfere with each other (though they can interfere with non-RealSense IR devices, which may provide a complication in regard to the Vicon).

https://store.intelrealsense.com/buy-intel-realsense-depth-camera-d455.html

shindm, 2021-09-17:
Thanks, Marty! It is very helpful. DS

shindm, 2021-09-23:
No. Thank you!

shindm, 2021-09-23:
No. Thanks again.

MartyG-RealSense, 2021-09-23:
Thanks very much @shindm for the update!  As you do not require further assistance, I will close this case.  Thanks again!

closed 2021-09-23 by MartyG-RealSense