realsenseai/librealsense #9751: Firmware releases D400

Sweet-orange-gethub, 2021-09-10:
Hello!
| Camera Model                       |  D400  | 

My realsen d435 sometimes has the following situations
1.No Frames Received!
2.There is a lot of noise in the depth image
I saw that one of the above problems was solved in a firmware update, but I can't remember clearly.
May I ask the possible reasons for the above problems and which version has been updated to solve the corresponding problems.
The firmware version of the camera I am using is5.11

MartyG-RealSense, 2021-09-10:
Hi @Sweet-orange-gethub   The 5.11.x firmware versions are very old now, dating back to 2019.  Numerous firmware releases since then have added a range of fixes, features and improvements.  So it is certainly recommendable to update the firmware to a recent version.

Having said that, noise in your depth image may not necessarily be related to the firmware.  Would it be possible to provide an RGB image of the scene that the camera is observing please in order to see whether there are elements in the scene that might be causing noise?

Sweet-orange-gethub, 2021-09-10:
Thank you for your reply. The camera is in a relatively stable environment. Generally, it is normal, because there are many cameras. It is not easy for each camera to upgrade the firmware. Based on this situation, what is the necessity for us to upgrade the firmware?
In addition, I've seen an answer before to whether the official has updated the firmware for the above questions, but it's not clear, so I want to ask you,

MartyG-RealSense, 2021-09-10:
It is straightforward to upgrade the firmware through the RealSense Viewer but it does become more complicated to keep the firmware updated when you have many cameras in use, especially if they located at different sites.  Some RealSense users have approached this problem by writing their own firmware update script that checks the camera to see whether it needs updating and then performs the update if it does.

Occasionally there may be a particular camera feature that requires a firmware update for that feature to be used.  In general though, the changes between firmware versions tend to be small.  It may therefore work best for you if you only install firmwares labelled by Intel as **Production**.  These are firmwares that are validated as stable and suited for use in commercial products rather than for testing and development purposes.  They are older than the most recent firmware versions but useful if you need stability and wish to have a long period in-between updating your cameras.

https://dev.intelrealsense.com/docs/recommended-production-camera-configurations

MartyG-RealSense, 2021-09-23:
Case closed due to no further comments received

closed 2021-09-23 by MartyG-RealSense