realsenseai/librealsense #9753: L515 not working in Android

rodrigo-a-lima, 2021-09-10:
Hi,

I'm trying to use my L515 into an Android Lenovo Tab M10 FHD Plus by the Intel RS Camera app. I'm using a 2.0 USB. However, it just shows that the device is connected, but the screen is black. Is there any workaround?

Best,

MartyG-RealSense, 2021-09-10:
Hi @rodrigo-a-lima   The discussion https://github.com/IntelRealSense/librealsense/issues/8069 states that to use the Android **RS Camera** app with L515, it should be installed and run directly as an **apk** obtained from the link (below).

https://egiintel.jfrog.io/artifactory/librealsense/com/intel/realsense/librealsense/

rodrigo-a-lima, 2021-09-10:
Hi Marty, I tried the camera-2.49.0.apk, but it still keeps the black screen asking to connect the RealSense camera

MartyG-RealSense, 2021-09-10:
A RealSense user with an L515 had the same issue with the black screen in https://github.com/IntelRealSense/librealsense/issues/7053 and was advised to connect the camera with USB 3, though they continued to have the problem even when connecting with USB 3.

Which firmware version is your L515 using, please?  L515 initially only supported USB 3 when first released but support for USB 2 was added in a later firmware driver.  However, if you currently have a firmware installed in the camera that is older than the version that introduced USB 2 support then you have to update the firmware on a computer with a USB 3 connection.

rodrigo-a-lima, 2021-09-12:
Hi Marty, thank you very much for your reply. I will check the firmware of my L515 next week. I left it at the university. Fingers crossed that it worked; it will be part of my research. I will keep you updated about it.

MartyG-RealSense, 2021-09-12:
Yes, please do update here when you have checked the firmware in the coming week.  Good luck!

rodrigo-a-lima, 2021-09-13:
Hi Marty! I updated the firmware to version 1.5.8.1, and it is still not working on my Android tablet. 
To updated, I used the standard cable that came with my L515. Was it correct?

MartyG-RealSense, 2021-09-13:
Yes, it is correct to use the cable that is supplied with the camera to perform firmware update.

Are you able to build and run any other RealSense example applications on your Lenovo Tab successfully, such as **Capture**

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/android/examples/capture

There may be a rare occasion where a particular model of phone has been given a chipset by its manufacturer that is incompatible with RealSense, as demonstrated in https://github.com/IntelRealSense/librealsense/issues/8442#issuecomment-785798027

rodrigo-a-lima, 2021-09-13:
Hi Marty, my android is the 10 version.

I'm sorry to say, but I have no idea how to do the capture.

Please, where could I find a tutorial to do this?

rodrigo-a-lima, 2021-09-13:
Hi Marty, I'm sorry to say, but I didn't find any option after step 4 using the link you sent me.
Is there any beginner tutorial for this?

MartyG-RealSense, 2021-09-14:
I researched your request for a beginner tutorial carefully but did not find one.  I did though go through the installation steps in Intel's guide from the beginning to check them.

May I confirm whether you were able to install Android Studio successfully?  I found that it did not work correctly on a networked Windows PC but installed correctly on a PC that was not on a network.

At step 4 of the procedure, I went to the **Open** option of Android Studio and instead of selecting a location on the computer, I pasted the website address that Intel provide in their guide into the address bar at the top of the Open pop-up box.

In the guide it is written as **path to build.gradle**, but it is a link and clicking on it provides the full address below:

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/android/build.gradle

After I pasted this address into the location to open and clicked the **OK** button, the build gradle downloaded into Android Studio and installed automatically.

[image]

rodrigo-a-lima, 2021-09-14:
Yes, I had some issues with the Android Studio. In fact, I need to install it all the time when I'll use it. It's pretty weird.
Thank you very much for your support. It is very important to make things work for my research. I tried to find something too, but there is nothing for beginners as you said...

I did what you said after **step 4**, and all I have in return was this that you could see on the print screens...I even don't know if it is the correct thing that should appear.

[image]

[image]

MartyG-RealSense, 2021-09-15:
That is the same screen that I received in my own Android Studio installation after opening using the supplied build gradle web address.

Logically, the next step would be to move on to Step 5 of the installation guide.  "In the gradle tasks run :librealsense-->build-->assembleRelease:"

[image]

I was unable to find the Gradle Tasks interface in the above illustration.  Apparently in Android Studio 2021 it is disabled by default because it can negatively impact performance, and the procedure to re-enable it is as follows:

1.  Go to **File > Settings**

[image]

2.  Go to the **Experimental** category.  Untick the setting **Do not build Gradle task list during Gradle sync**.  then click the **Apply** and **OK** buttons.

3.  Click the **Sync projects with Gradle file** icon on the toolbar in the top corner of the Android Studio window.

[image]

rodrigo-a-lima, 2021-09-15:
Hy Marty, 

Thank you so much for your explanation; I'd never find out it by myself.

I followed your steps and this is the screen that I got.

[image]

MartyG-RealSense, 2021-09-15:
You're very welcome   :)

Are you able to display a Gradle tasks list like the one in the instructions?  If you are, then are you able to select the **assembleRelease** entry and run it?

[image]

rodrigo-a-lima, 2021-09-15:
Well, I find the option, but it doesn't look very similar to your previous screens...What do you think?

[image]

MartyG-RealSense, 2021-09-15:
Given that the Intel installation guide that we are referring to was published a long time ago, there are bound to be differences between its images and the current 2021 version of Android Studio.  So it is not necessarily something to be concerned about.

Are you able to run **assembleRealSense** ?  The installation guide says that the build process after running it should take a few minutes, and if the build succeeds then the generated AAR will be located in `<librealsense_root_dir>/wrappers/android/librealsense/build/outputs/aar`

rodrigo-a-lima, 2021-09-15:
Unfortunately no... I tried to run it, but this error occurred during the process
Please ignore part of a zoom meeting to the right. I was supposed to be part of it, but I'm more worried to solve this issue with the L515!

[image]

rodrigo-a-lima, 2021-09-15:
Hi Marty, 

I solved it!

[image]

MartyG-RealSense, 2021-09-16:
That's excellent news, @rodrigo-a-lima - thanks so much for the update!   :)

rodrigo-a-lima, 2021-09-16:
Now could you help me with the next step, please? Sorry about this...

MartyG-RealSense, 2021-09-16:
It's no problem at all.  After running **assembleRealSense**, were you able to find a built AAR in the **librealsense/wrappers/android/librealsense/build/outputs/aar** folder?

If you do have an AAR, is the next step that you would like help with to run one of the example apps?

rodrigo-a-lima, 2021-09-16:
You won't believe it, but my tablet had some issue with the charger. I already claimed the warranty, but I will stay some days without it...Or I'll have a refund... Anyway, Marty, do you know a portable device that will de compatible with the L515? I'll need to check other possibilities because I don't know if I could rely on this tablet for the fieldwork.

MartyG-RealSense, 2021-09-16:
I would recommend avoiding devices with Android 10 if possible because of the the problems that the RealSense Android wrapper has with it.  So Android 11 would be the ideal.  

I have not seen librealsense Android wrapper issues that use L515 with named Android 11 models, so unfortunately I do not have recommendations that I can highlight.

In regard to non-L515 reports, a RealSense user successfully used a **Google Pixel 4** with Android 11 and a 400 Series camera in https://github.com/IntelRealSense/librealsense/issues/7691

MartyG-RealSense, 2021-09-22:
Hi @rodrigo-a-lima  Do you require further assistance with this case, please?  Thanks!

rodrigo-a-lima, 2021-09-22:
No Marty, thank you very much for your patience on this! It is truly appreciated.

Best,
Rodrigo
________________________
Rodrigo de Lima / PhD candidate
Labeca – Laboratory of Studies on the Ancient City
University of Sao Paulo
Museum of Archaeology and Ethnology
University of Bristol
Department of Anthropology and Archaeology

Pronouns he, him, his

MartyG-RealSense, 2021-09-22:
You are very welcome.   As you do not require further assistance, I will close the case.  Thanks again!

closed 2021-09-22 by MartyG-RealSense