realsenseai/librealsense #9754: Do I need Kernel patches on openSUSE?

snopytas, 2021-09-11:
Hi, I am using librealsense on openSUSE Tumbleweed with Kernel 5.14. Librealsense is included in the openSUSE repositories. It runs well. But in your installation instructions there is written that librealsense requires kernel patches. Now I have two questions:

1) What is the purpose of these patches? 
2) Are they included in the mainline 5.14 kernel? Or do I need to apply them manually?

MartyG-RealSense, 2021-09-11:
Hi @snopytas  A primary purpose of patching the kernel is to add support for hardware metadata timestamps in the librealsense SDK.   Librealsense-specific functionality is not included in Linux kernels by default and has to be patched in.  Also, the SDK currently supports up to kernel 5.8 officially at the time of writing this.  Newer kernels can work with the SDK but there may be unpredictable effects.  

If librealsense is built from source code with the **RSUSB backend** method then it is not dependent on Linux versions or kernel versions and does not require patching.  This can be a useful approach if using recent kernels that are not officially supported yet, or if there is a conflict between librealsense and the Linux / kernel version being used on that computer.

A RealSense team member provides advice in https://github.com/IntelRealSense/librealsense/issues/6837#issuecomment-660600748 regarding checking whether a patch has been correctly applied.

snopytas, 2021-09-11:
Very interesting, thanks. Does this mean that I get more unprecise timestamps when not using a patched kernel?

MartyG-RealSense, 2021-09-11:
The SDK should use hardware timestamps when support for them is enabled, otherwise it will provide **software timestamps** based on system time.  

The SDK documentation link below states that metadata support is required to access **FRAME_TIMESTAMP**

https://intelrealsense.github.io/librealsense/doxygen/classrs2_1_1frame.html#a25f71d45193f2f4d77960320276b83f1

snopytas, 2021-09-15:
Thanks. Now I was in lab with my openSUSE tumbleweed workstation. The timestamp domain shows global time and I can also set it to hardware time. So I think these patches are applied by default in the mainline kernel or at least in the openSUSE Kernel.

closed 2021-09-15 by snopytas

MartyG-RealSense, 2021-09-15:
Thanks very much @snopytas for the update   :)