realsenseai/librealsense #9755: GL-ERROR with VNC Viewer

didi1150, 2021-09-11:
| Camera Model                       | { D400 } | 
| Operating System & Version |   Raspbian OS  | 
| Platform                                 | Raspberry Pi 3b+ | 

### Issue Description
I am using VNC Viewer to open the realsense app, according to this tutorial 

https://www.youtube.com/watch?v=LBIBUntnxp8&t=166s

But when I run the command realsense-viewer, it opens, and every text in the programm is blurred and pixelated. Why is it this way? I have done everything exactly as in the tutorial.
[image]

And I also get the error: 

GL CALLBACK: ** GL ERROR ** type = 0x824c, severity = 0x9146, message = GL_INVALID_VALUE in glTexImage2D(invalid width=4096 or height=512 or depth=1)

MartyG-RealSense, 2021-09-11:
Hi @didi1150  This distortion effect in the RealSense Viewer is very rare but has been reported to have occurred on Raspberry Pi 3 in the past in https://github.com/IntelRealSense/librealsense/issues/7533

It is thought to be related to a lack of memory in the **texture atlas**.  The above link provides a couple of suggestions for correcting the problem.

The RealSense user in that particular case chose to resolve the problem by upgrading to a Pi 4B instead.

First though, you could check whether the issue on your particular Pi is being caused by the **Font Samples** setting of the Viewer being set to its maximum value of '8' instead of its default value of '2', as described in https://github.com/IntelRealSense/librealsense/issues/7479#issuecomment-702880652

didi1150, 2021-09-11:
I tried setting it to 2, but it still is blurred the same way

MartyG-RealSense, 2021-09-12:
Does the same effect occur if launching the Viewer directly on the Pi instead of through a VNC Viewer connection?

didi1150, 2021-09-12:
hmm lemme check

didi1150, 2021-09-12:
yes it is still bugged

didi1150, 2021-09-12:
Ive edited the model-views.cpp file, but I dont know how to rebuild

MartyG-RealSense, 2021-09-12:
A RealSense user in https://github.com/IntelRealSense/librealsense/issues/7450#issuecomment-706923841 who had this distortion problem on Pi indicates that you can use the **make** instruction to rebuild.

didi1150, 2021-09-12:
No it still doesnt work

MartyG-RealSense, 2021-09-12:
You just do the 'make' instruction.

didi1150, 2021-09-12:
I still dont understand. What do you mean by make instruction?

MartyG-RealSense, 2021-09-12:
**make** is a CMake build instruction.  Are you familiar with how CMake works, please?

didi1150, 2021-09-12:
cmake .. -DBUILD_EXAMPLES=true -DCMAKE_BUILD_TYPE=Release -DFORCE_LIBUVC=true
U mean this?

MartyG-RealSense, 2021-09-12:
Please input the word **make**into the Ubuntu terminal when you are in the build folder.

didi1150, 2021-09-12:
im on raspbian

didi1150, 2021-09-12:
I tried it with make and it still didnt work

didi1150, 2021-09-12:
Im now trying with cmake before that

didi1150, 2021-09-12:
Does my micro SD card have anything to do with my problem?

MartyG-RealSense, 2021-09-12:
Once a Raspberry Pi runs out of memory space then it can use a **swapfile** to use space on the storage as slower 'virtual memory'.  Sometimes, increasing the size of the swapfile may therefore improve performance of an application.

https://www.bitpi.co/2015/02/11/how-to-change-raspberry-pis-swapfile-size-on-rasbian/

didi1150, 2021-09-12:
I increased the sawpfile alreeady

MartyG-RealSense, 2021-09-12:
To recap, the known solutions for this problem with distortion of the Viewer's graphics are one of the actions below:

1.  Change the **Font Samples** setting in the RealSense Viewer

2.  Edit the **model-views.cpp** file and rebuild the Viewer

3.  Edit the configuration file located at ~/.realsense-config.json

4.  Upgrade from a Pi 3 to Pi 4

You have tried changing the Font Samples and editing model-views.cpp.  Have you tried editing the configuration file  located at ~/.realsense-config.json as described in https://github.com/IntelRealSense/librealsense/issues/4558#issuecomment-518635640

didi1150, 2021-09-12:
yes I have tried that too

MartyG-RealSense, 2021-09-12:
The four methods listed above are the only solutions known so far for this Viewer graphics corruption problem.  The only remaining untried option on the list is to upgrade to the Pi 4.

You could also try switching to the Viewer's **fullscreen** mode and back again to see whether it clears the problem.  you can do this by pressing the **F8** keyboard key whilst the Viewer is open to go to fullscreen mode, and press F8 again to go back to normal sized view.

didi1150, 2021-09-12:
No it is still bugged

didi1150, 2021-09-12:
'Edit the configuration file located at ~/.realsense-config.json' 
What can I edit there? I wanna try again

didi1150, 2021-09-12:
Okay ive tried that again, it still doesnt work

didi1150, 2021-09-12:
GL CALLBACK: ** GL ERROR ** type = 0x824c, severity = 0x9146, message = GL_INVALID_VALUE in glTexImage2D(invalid width=4096 or height=512 or depth=1)
 What does this mean?

MartyG-RealSense, 2021-09-12:
I only know of one previous occurrence of that GL error in the RealSense Viewer in https://github.com/IntelRealSense/librealsense/issues/8705 and it did not suggest a solution.

I conducted extensive research about this Viewer distortion but could not find any options other than the four listed above, of which changing to a Pi 4 is the only one that has not been attempted.

Let's instead look at the possibility that there is a problem in the librealsense SDK installation on your Pi.  Which method did you use to install the SDK on Raspbian OS on Your Pi 3B+, please?

didi1150, 2021-09-12:
I followed the tutorial by acrobotic: https://www.youtube.com/watch?v=LBIBUntnxp8&t=166s
which means I first cloned the git repo and then compiled it

MartyG-RealSense, 2021-09-13:
Thanks very much.  The text version of the Acrobotic guide in the link below offers debugging advice.

https://github.com/acrobotic/Ai_Demos_RPi/wiki/Raspberry-Pi-4-and-Intel-RealSense-D435#debugging

The Acrobotic guide was designed though for Pi 4, whilst you are using Pi 3B+.  Intel have their own official installation guide for Pi 3 that is similar in structure to Acrobotic's Pi 4 guide.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_raspbian.md

didi1150, 2021-09-13:
OOps I just noticed I was using ubuntu mate when I followed the intel guide

MartyG-RealSense, 2021-09-13:
Yes, there is a different Raspberry Pi guide that uses Ubuntu MATE.  

https://github.com/IntelRealSense/librealsense/blob/master/doc/RaspberryPi3.md

It was published first in November 2017 (when the 400 Series cameras first launched) and the Raspbian guide was published later in January 2019.

didi1150, 2021-09-14:
Hey I get the error autoreconf not found when doing ./autogen.sh in the install protobuf step

didi1150, 2021-09-14:
okay fixed it by following #4

didi1150, 2021-09-14:
I get another error, the same as described as in here https://www.raspberrypi.org/forums/viewtopic.php?t=288684

didi1150, 2021-09-14:
and I couldn't fix it

MartyG-RealSense, 2021-09-14:
Thanks very much for your patience.  It's great to hear that you were able to install successfully by using libtool.

In regard to the newest error, a RealSense Raspberry Pi user who encountered this problem in the past tried fixing it using the command below:

**apt-get install python3-opencv**

didi1150, 2021-09-14:
This is kind of turning into a small guide itself. I'll report further problems if they occur so other users who might get into the same errors can fix it too

didi1150, 2021-09-14:
oh btw for everyone with the problem: 

CMake Error at /home/pi/cmake-3.11.4/share/cmake-3.11/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  Could NOT find OpenSSL, try to set the path to OpenSSL root folder in the
  system variable OPENSSL_ROOT_DIR (missing: OPENSSL_CRYPTO_LIBRARY
  OPENSSL_INCLUDE_DIR)
Call Stack (most recent call first):
  /home/pi/cmake-3.11.4/share/cmake-3.11/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
  /home/pi/cmake-3.11.4/share/cmake-3.11/Modules/FindOpenSSL.cmake:390 (find_package_handle_standard_args)
  CMake/external_libcurl.cmake:56 (find_package)
  CMake/global_config.cmake:93 (include)
  CMakeLists.txt:44 (global_set_flags)

just type "sudo apt-get install libssl-dev"

MartyG-RealSense, 2021-09-14:
Thanks very much for your very helpful solution sharing   :)

didi1150, 2021-09-14:
okay I just had a LOT of errors while using make on librealsense

didi1150, 2021-09-14:
In file included from /home/pi/librealsense/third-party/realsense-file/boost/boost/config.hpp:30,
                 from /home/pi/librealsense/third-party/realsense-file/boost/boost/enable_if.hpp:17,
                 from /home/pi/librealsense/third-party/realsense-file/rosbag/roscpp_traits/include/ros/message_traits.h:36,
                 from /home/pi/librealsense/third-party/realsense-file/rosbag/rosbag_storage/include/rosbag/bag.h:49,
                 from /home/pi/librealsense/third-party/realsense-file/rosbag/rosbag_storage/src/bag.cpp:28:
/home/pi/librealsense/third-party/realsense-file/boost/boost/config/user.hpp:1:1: error: expected unqualified-id before numeric constant
 1$      � .�k� �     �
                        �� �1
 ^~
/home/pi/librealsense/third-party/realsense-file/boost/boost/config/user.hpp:1:22: warning: extra ‘;’ [-Wpedantic]
 1$      � .�k� �     �
                        �� �1
                      ^
/home/pi/librealsense/third-party/realsense-file/boost/boost/config/user.hpp:1:42: error: expected unqualified-id before numeric constant
 1$      � .�k� �     �
                        �� �1
                                          ^
In file included from /home/pi/librealsense/third-party/realsense-file/boost/boost/config.hpp:30,
                 from /home/pi/librealsense/third-party/realsense-file/boost/boost/enable_if.hpp:17,
                 from /home/pi/librealsense/third-party/realsense-file/rosbag/roscpp_traits/include/ros/message_traits.h:36,
                 from /home/pi/librealsense/third-party/realsense-file/rosbag/rosbag_storage/include/rosbag/bag.h:49,
                 from /home/pi/librealsense/third-party/realsense-file/rosbag/rosbag_storage/src/bag.cpp:28:
/home/pi/librealsense/third-party/realsense-file/boost/boost/config/user.hpp:4:9: warning: extra ‘;’ [-Wpedantic]
  �. 7� ��  �� �. �. 8� ��  �� �. �. 9� M  �� z. �. :� �{  �� Q. �. ;� �{  �� �. �. <� �|  ��  �. =� �{  �� S. �. >� �{  �� Y. �. ?� �|  �� Z. �. @� �|  �� X� �. A�  |  �� ^. �. B� |  �� _. �. C� �  �� .     D�  �k� � �� �] �� }. �. E� �] �� ~. �. F� �  �� �. �. G� ��  �� �. �. H�  % �� _� �. I� H  �� �# �. J� G  �� f. �. K� �|  �� m �. L� d% �� ��     M� �k �� K' �. N� �  �� � �. O� �\ �� {. �. P� 
                                                                                                                                                                                �  �� �. �. Q�  �  �� �. �. R� <|  �� �.     S� ��k� �     ��k� �     �  �� �� �. T� �% �� �� �. U� K ��     �. V� F�  �� �� �. W� il �� �% �. X� �\ �� �. �. Y� �  �� H. �. Z�  �  �� �- �. [� �  �� � �. \� 
                                                                                                                                                            �  �� �. �. ]�  �  �� �.     ^� �k� �     ��  �� �. �. _� �g �� �. �. `� �\ �� �. �. a� 
                  �  �� �. �. b� �  �� �. �. c� �{  �� �.     d� F�k� �     ��  �� 
         ^
/home/pi/librealsense/third-party/realsense-file/boost/boost/config/user.hpp:4:29: error: expected unqualified-id before ‘-’ token
  �. 7� ��  �� �. �. 8� ��  �� �. �. 9� M  �� z. �. :� �{  �� Q. �. ;� �{  �� �. �. <� �|  ��  �. =� �{  �� S. �. >� �{  �� Y. �. ?� �|  �� Z. �. @� �|  �� X� �. A�  |  �� ^. �. B� |  �� _. �. C� �  �� .     D�  �k� � �� �] �� }. �. E� �] �� ~. �. F� �  �� �. �. G� ��  �� �. �. H�  % �� _� �. I� H  �� �# �. J� G  �� f. �. K� �|  �� m �. L� d% �� ��     M� �k �� K' �. N� �  �� � �. O� �\ �� {. �. P� 
                                                                                                                                                                                �  �� �. �. Q�  �  �� �. �. R� <|  �� �.     S� ��k� �     ��k� �     �  �� �� �. T� �% �� �� �. U� K ��     �. V� F�  �� �� �. W� il �� �% �. X� �\ �� �. �. Y� �  �� H. �. Z�  �  �� �- �. [� �  �� � �. \� 
                                                                                                                                                            �  �� �. �. ]�  �  �� �.     ^� �k� �     ��  �� �. �. _� �g �� �. �. `� �\ �� �. �. a� 
                  �  �� �. �. b� �  �� �. �. c� �{  �� �.     d� F�k� �     ��  �� 
                             ^
In file included from /home/pi/librealsense/third-party/realsense-file/boost/boost/throw_exception.hpp:42,
                 from /home/pi/librealsense/third-party/realsense-file/boost/boost/function/detail/prologue.hpp:15,
                 from /home/pi/librealsense/third-party/realsense-file/boost/boost/function.hpp:24,
                 from /home/pi/librealsense/third-party/realsense-file/rosbag/roscpp_traits/include/ros/message_event.h:42,
                 from /home/pi/librealsense/third-party/realsense-file/rosbag/rosbag_storage/include/rosbag/bag.h:50,
                 from /home/pi/librealsense/third-party/realsense-file/rosbag/rosbag_storage/src/bag.cpp:28:
/home/pi/librealsense/third-party/realsense-file/boost/boost/exception/exception.hpp:137:40: error: expected declaration before end of line
 #  pragma GCC visibility push (default)
                                        ^
make[2]: *** [third-party/realsense-file/CMakeFiles/realsense-file.dir/build.make:115: third-party/realsense-file/CMakeFiles/realsense-file.dir/rosbag/rosbag_storage/src/bag.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:192: third-party/realsense-file/CMakeFiles/realsense-file.dir/all] Error 2
make: *** [Makefile:130: all] Error 2

MartyG-RealSense, 2021-09-15:
I have only seen the **expected unqualified-id before numeric constant** error in one previous case in https://github.com/IntelRealSense/librealsense/issues/2897 where a RealSense team member suggested using the CMake build flag **-DBUILD_WITH_TM2=false** as a workaround.

didi1150, 2021-09-15:
thank you the solution worked

didi1150, 2021-09-15:
Nope after following these instructions it is still buggy textured

didi1150, 2021-09-15:
Unfortunately, I think I still have to buy a raspberry pi 4. Which version of the raspberry pi 4 should I buy? The 4GB Ram one, or the 8 GB RAM one?

MartyG-RealSense, 2021-09-15:
The 4 GB model should be sufficient for librealsense.

didi1150, 2021-09-24:
why are there black areas/mist all over the place? And the cam is also veryyyy slow. I was expecting maybe 30 fps but it is 1 fps

MartyG-RealSense, 2021-09-25:
Would it be possible to provide an image of the black areas and mist that you are seeing, please?

didi1150, 2021-09-25:
[image]

didi1150, 2021-09-25:
[image]

didi1150, 2021-09-25:
first one is 3d, second one is 2d

didi1150, 2021-09-25:
25/09 12:21:13,554 WARNING [2866795584] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 25/09 12:21:24,489 WARNING [2754602048] (sensor.cpp:337) Frame received with streaming inactive,Depth0, Arrived,0.000000 1632565284476.807617
 25/09 12:22:46,861 WARNING [2866795584] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 25/09 12:23:07,057 WARNING [2866795584] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 25/09 12:23:19,825 WARNING [2866795584] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 
 
 
 And also these errors

MartyG-RealSense, 2021-09-25:
Is the camera looking at objects that are dark grey or black?   If so, a general physics principle for depth cameras (not just RealSense) is that dark grey and black colors absorb light.  This makes it hard for the camera to read depth detail from such surfaces unless they have a strong light-source on them.  Black surfaces may therefore appear to be rendered on the image but are actually plain black empty areas with no depth detail.  The darker the shade, the harder it is for light to be reflected.

An example is depth-sensing a black cable.  It may appear on the image to be rendered as a black cable but is actually a cable-shaped empty area (an effect very similar to your second depth image above).

didi1150, 2021-09-25:
ok....

didi1150, 2021-09-25:
how do I add scripts or code to the camera? any good beginner tutorials you would recommend? Because I can only find intermediate tutorials

MartyG-RealSense, 2021-09-25:
If you are going to be programming in C++ but are not using Visual Studio then the Getting Started guide at the link below may be helpful.

https://github.com/zivsha/librealsense/tree/getting_started_example/examples/getting-started

If you are using Python then there is the SDK's **Tutorial 1** example Python program.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/python-tutorial-1-depth.py

MartyG-RealSense, 2021-10-01:
Hi @didi1150  Do you require further assistance with this case, please?  Thanks!

didi1150, 2021-10-02:
My cam is still showing black mist, even if I have have a light shining onto an object, it is still being displayed as black mist if it is not white

didi1150, 2021-10-02:
For example if im holding my hand towards the cam, everything around it is turned black

didi1150, 2021-10-02:
And all these tutorials dont explaine one thing: How do I setup the coding environment? And do I need to code on my computer or the raspberry pi? And if I'm writing everything on my pc, how do I export the code to the raspberry? Also none of these tutorials seem to be made for visual studio

> If you are going to be programming in C++ but are not using Visual Studio then the Getting Started guide at the link below may be helpful. [...]

This doesn't explain how to write code yourself.

[14 more replies]

closed 2021-10-16 by MartyG-RealSense