realsenseai/librealsense #9760: Building pyrealsense2 does not include librealsense

oliverdain, 2021-09-12:
Camera Model: D415 (but also irrelevant)
FW Version: irrelevant
Operating System: Linux (Debian Bullseye) and OSX (Big Sur)
Platform: OSX and Debian Linux
Language: cmake

### Issue Description

Intel doesn't provide a pre-build `pyrealsense2` for OSX so I compiled my own. However, if I `pip install` the `.whl` file I built it is not statically linked against `librealsense2` so users also have to install that. This is true on both OSX and Linux. However, I followed the instructions to compile everything and then ran `python3.7 setup.py bdist_wheel` to create the wheel file but somehow my version is different from the one built by Intel. For example, for the one I built if I run `ldd` on the `pyrealsense2.cpython-37m-x86_64-linux-gnu.so` that is included in the wheel I get:

```
        linux-vdso.so.1 (0x00007fffb6dec000)
        librealsense2.so.2.49 => not found
        libstdc++.so.6 => /lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f9767761000)
        libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f976761d000)
        libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f9767603000)
        libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f976743e000)
        /lib64/ld-linux-x86-64.so.2 (0x00007f9767d68000)
```

but if I download the official Linux one and run `ldd` I get:

```
        linux-vdso.so.1 (0x00007ffebd9f1000)
        libusb-1.0.so.0 => /lib/x86_64-linux-gnu/libusb-1.0.so.0 (0x00007fb2f3ca0000)
        libstdc++.so.6 => /lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007fb2f3ad3000)
        libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007fb2f398f000)
        libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007fb2f3975000)
        libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007fb2f3953000)
        libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fb2f378e000)
        /lib64/ld-linux-x86-64.so.2 (0x00007fb2f53bc000)
        libudev.so.1 => /lib/x86_64-linux-gnu/libudev.so.1 (0x00007fb2f3764000)

```
so the official one does not need to be dynamically linked against `librealsense2.so.2.49`. I suspect the instructions just need to be updated.

To be clear, here's the steps I followed:

1. Run `cmake` with `-DBUILD_PYTHON_BINDINGS:bool=true` and `'-DPython_ADDITIONAL_VERSIONS=3.7'`.
2. Run `make -j4 pyrealsense2 pybackend2`
3. Copy `pybackend2.cpython-37m-x86_64-linux-gnu.so` and `pyrealsense2.cpython-37m-x86_64-linux-gnu.so` into `wrappers/python/pyrealsense2`
4. Run the `find_librs_version.py` script
5. Run `python3.7 setup.py bdist_wheel`

I imagine I need some extra cmake flag to have my `pyrealsense2.cpython-37m-x86_64-linux-gnu.so` statically linked against `librealsense2` but I don't see an obvious flag and it's not documented.

Note that the main reason I'm doing this is that there isn't an OSX version so I need to compile that. I only repeated the steps on Linux so I could compare to the official `pyrealsense2` wheel provided by Intel.

oliverdain, 2021-09-12:
Also note that copying `librealsense2` to `wrappers/python/pyrealsense2` causes it to be in the wheel file but it's still not found as ldd doesn't know where to look for it.

MartyG-RealSense, 2021-09-12:
Hi @oliverdain    A RealSense user recently created an automated process at https://github.com/IntelRealSense/librealsense/issues/9687 for installing pyrealsense2 on MacOS that can create pre-built wheel files.  Comparing it to your own method may provide useful insights.

oliverdain, 2021-09-13:
Thanks for the pointer @MartyG-RealSense . I took a look at his code and he's bundling `librealsense` and then using `delocate` after the wheel is built to update the `rpath`. That works but seems really hacky and I suspect it's not how you build the official wheels you release (particularly since `delocate` is OSX only). I was hoping there's a cleaner way to do it.

MartyG-RealSense, 2021-09-14:
The Python wrapper documentation states that the PyPi (pip install) wheel files for x86 / x64 are generated from the Python wrapper folder with the command **python setup.py bdist_wheel**

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python

[image]

oliverdain, 2021-09-14:
Thanks @MartyG-RealSense . As noted that is what I'm doing. And I can get a working wheel file, but one that relies on `librealsense2.49` having been installed system wide which means a simple `pip install` doesn't give you a usable system. Furthermore, Intel's own `.whl` files do _not_ have a dependency on an external `librealsense` so it seems it's possible to create a stand-alone `pyrealsense2` but following the official instructions, including the `setup.py bdist_wheel` step you recommend. The full procedure I followed (which is essentially what's in the official instructions plus a step or two those instructions are missing) are in the original bug report.

MartyG-RealSense, 2021-09-15:
I researched your question carefully but it is admittedly one that is outside of my Python installation knowledge.  If you would like to explore deeper into the subject of wheel file creation, there is a highly technical discussion at https://github.com/IntelRealSense/librealsense/issues/8229

Within that discussion, a RealSense user suggested downloading an existing wheel file and then editing its binary with a hex editor tool.

https://github.com/IntelRealSense/librealsense/issues/8229#issuecomment-767115385

The downloads page for the official wheels is here:

https://pypi.org/project/pyrealsense2/#files

oliverdain, 2021-09-15:
Hi @MartyG-RealSense . Thanks for the update. I'm afraid that link doesn't help: it's about how to modify the metadata in a wheel to declare that it's compatible with Python 3.9 which isn't my situation at all.

Clearly _somebody_ at Intel has a process for building a wheel file such that it does not depend on an externally install `librealsense2`: that's what you release to PyPi. I'd expect that you also have that series of steps running automatically on your CI/CD. Could the person who knows how to do it maybe chime in with the magic missing step?

MartyG-RealSense, 2021-09-15:
@Nir-Az  Would it be possible to offer advice to @oliverdain regarding their question above please?  Thanks!

Nir-Az, 2021-09-16:
Hi,
If you wish to build pyrealsense to be self contained you need to add the following CMake flag to the CMake configure step.

`-DBUILD_SHARED_LIBS=false`

This way you get a statically linked binary file.
Let us know if it did the trick 😉

oliverdain, 2021-09-17:
That did the trick @Nir-Az . Thank you!

I suggest updating the documentation; it was definitely not obvious to me how to do this.

closed 2021-09-17 by oliverdain

MartyG-RealSense, 2021-09-17:
@Nir-Az   Thanks very much for your always excellent help on the subject of Python wheels!

@oliverdain   Great to hear that the advice of @Nir-Az was successful for you  :)

Nir-Az, 2021-09-19:
@oliverdain  Thanks for alerting about the missing documentation.
The python wrapper readme file has been updated! #9781