realsenseai/librealsense #9766: how to record video rgb frame and depth frame at the same time?

dohyeonYoon, 2021-09-14:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | v2.47.0 | 
| Operating System & Version |  Win 10 | 
| Kernel Version (Linux Only)    |                                          | 
| Platform                                 | PC  |
| SDK Version                            |  2.47.0 }                          | 
| Language                            |  python                          | 
| Segment			| others          | 

### Issue Description
hello 
I'm trying to record the rgb video and depth video at the same time, using L515 camera. 
is it possible??  

thank you

MartyG-RealSense, 2021-09-14:
Hi @dohyeonYoon   The discussion https://github.com/IntelRealSense/librealsense/issues/8528 has a Python script for saving depth and color as .avi format video files if your goal is to record conventional video files.  It also discusses recording camera data into bag files.

dohyeonYoon, 2021-09-15:
I saved it in the avi file format as shown in the picture below.

 After that, I captured this video and got several png files.

In this case, is there a way to get distance information for each pixel from the png file?

dohyeonYoon, 2021-09-15:
in addition, It has been confirmed that the viewer can get the bag file with the camera recording function.

 Is there a way to get distance information for each pixel in the bag file?

MartyG-RealSense, 2021-09-15:
PNG files are not a reliable way to get depth information, as most of the depth detail is lost when saving to PNG format.  https://github.com/IntelRealSense/librealsense/issues/4427#issuecomment-511201083

In regard to bag files, you should be able to change any RealSense script that uses live camera data to use a pre-recorded bag file as its data source instead by adding a **cfg.enable_device_from_file** instruction above your pipe start line.  https://github.com/IntelRealSense/librealsense/issues/3121#issuecomment-718628425 has a Python example of implementig this instruction.

dohyeonYoon, 2021-09-16:
Thanks for the reply!!
I noticed that depth information was lost when converting to PNG files.
Then I need to get the bag file and convert the bag file to a raw file, is there a way to get RGB frame and depth frame as a bag file in window 10 with just python code without a viewer?

MartyG-RealSense, 2021-09-16:
You could convert a bag file to .raw without using Python by using the SDK's **rs-convert** tool.  In the Windows SDK you can find a pre-built executable version at **C: > Program Files (x86) > Intel RealSense SDK 2.0 > Tools**

https://github.com/IntelRealSense/librealsense/tree/master/tools/convert

[image]

In regard to recording a bag file containing depth and color data in Python, I hope https://github.com/IntelRealSense/librealsense/issues/3671#issuecomment-492559383 will be a helpful reference.

MartyG-RealSense, 2021-09-22:
Hi @dohyeonYoon  Do you require further assistance with this case, please?  Thanks!

dohyeonYoon, 2021-09-23:
thanks for the kind explanation. I have one question left
I'm trying to  "pip install pyrealsense2" on jetson nano, but it doesn't work.
I was referring to this article #6964  
It didn't work properly due to an error in make.

How can I install pyrealsense2 module on jetson nano?

os: jetpack 4.5.1 for jetson nano

MartyG-RealSense, 2021-09-23:
My understanding is that this script is designed to save a single frame to bag.  And so whilst it is a useful Python programming reference, using this script as it is would not provide a continuous recording of camera data.

https://github.com/soarwing52/RealsensePython#data-capture

I located another script example for Python where depth and color are saved to a bag file.

https://github.com/IntelRealSense/librealsense/issues/2882

dohyeonYoon, 2021-09-23:
What I want is to store persistent color and depth frames in a bag file format. using python code.

MartyG-RealSense, 2021-09-23:
Whilst there are plenty of examples of Python scripts for reading a bag file, there are few examples of recording a bag in Python and the links already mentioned in this discussion are everything that I was able to find.  So you may have to piece together a bag recording script from those.

dohyeonYoon, 2021-09-23:
oh I see.
It helped me a lot.
thank you so much.

closed 2021-10-05 by MartyG-RealSense