realsenseai/librealsense #9769: Issue on python-tutorial-1-depth.py

GitNzino, 2021-09-14:
Hi to everyone,

I had some issue on launching the tutorial examples in python. In particular the script that gave me problem is `python-tutorial-1-depth.py `.

I have the API of librealesense and pyplibrealsense upgraded and i have not modified any part of the pyhon code.
The output on the terminal is the following:

[image]

MartyG-RealSense, 2021-09-14:
Hi @GitNzino  That is how that particular example is meant to look.  It represents the depth as **ASCII art** (an image made from simple characters).  

[image]

If you are seeking a depth image output that is more like that of the RealSense Viewer tool then I would recommend the **opencv_viewer_example.py** Python example.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/opencv_viewer_example.py

[image]

GitNzino, 2021-09-14:
Thank you!
I have confused the two examples, my fault!
Sorry for my question

closed 2021-09-14 by GitNzino