realsenseai/librealsense #9771: Building source for matalb

nchu-cha, 2021-09-14:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 01.05.08.01 | 
| Operating System & Version | Win 10 | 
| Kernel Version (Linux Only)    |                                         | 
| Platform                                 | PC |
| SDK Version                           | v2.49.0 | 
| Language                               | matlab | 
| Segment			        | others | 

### Issue Description
hello
I'm trying to build the source for matlab, so I follow the `readme.md` in `librealsense/wrappers/matlab/`, In the process of building the source, there are some questions I want to ask for help.

1.  When I run the `cmake .. -D =ON` command, it shows `could not find apriling` what effect will it cause?
[image]

2.  I saw a video that was executed on Linux, and used `cmake .. -L` command. I had tried this command and it shows `GLFW3.3 not found, using internal version`. does it matter if I didn't install GLFW3.3?
[image]

3.  The second step in `readme.md` is build the `librealsense_mex` target, so I copy matlab file in the wrappers, and key `make librealsense_mex` , then I get an error : 
`fatal error: mex.h: No such file or directory` 
[image]
There is no file called mex.h, is it means that I need to write the mex.h file by myself? If yes, does anyone know how to write a mex.h? If no, where can I get the mex.h file?

thank you

nchu-cha, 2021-09-14:
I already install SDK, do you mean the MATALB wrapper refers to the MATALB folder in the picture? 
[image]

If MATALB wrapper can't support L515, how can I use MATALB to capture the RGB image and DEPTH image or the  parameters of the camera?

I used to capture the RGB image and DEPTH image in Realsense Viewer (in the picture show, I will click the button of the camera icon), so I had a `Color.png` for RGB image and a `Color.png` for depth image, a `RAW file` of depth information. Then used MATALB code to read all the file, but I found that RGB image and DEPTH image data are not identical. So I think maybe I need to install SDK to get some camera's parameters to conversion. Can you give me some suggestion?
[image]

thank you

MartyG-RealSense, 2021-09-15:
Thanks very much @nchu-cha  for your confirmation that your L515 was able to work with the RealSense MATLAB wrapper.  :)

You should be able to move the +realsense folder to a location of your choice and then define a MATLAB search path to it, as described in https://github.com/IntelRealSense/librealsense/issues/2400#issuecomment-422047168

If the pipe start instruction's brackets are empty then the RealSense SDK should automatically use the default configuration of the particular RealSense camera model that is attached.  On the L515 model, the default configuration would be 640x480 depth at 30 FPS (30 is the only FPS available for depth and infrared on L515) and 1280x720 color at 30 FPS.

If you need both depth and color to be the same then you could set them both to **640x480** with a custom **cfg** configuration.  The L515 should have firmware **1.5.8.1** or newer installed in order to use 640x480 color. 

[image]

https://github.com/IntelRealSense/librealsense/issues/3735 has an example of a cfg stream configuration in the MATLAB wrapper.

nchu-cha, 2021-09-16:
thanks for your apply, I'm trying the cfg stream configuration.

closed 2021-09-16 by nchu-cha